isaac-sim/IsaacLab #805: How to use RayCasterCamera

jih189, 2024-08-08:
I want to use ray caster camera as depth camera to watch a drawer and use the pointcloud as the observation to train the model to open drawer.
`
 self.scene.ray_caster_camera = RayCasterCameraCfg(
            prim_path="{ENV_REGEX_NS}/Robot/head_tilt_link/head_camera",
            mesh_prim_paths=["{ENV_REGEX_NS}/Cabinet"],
            # mesh_prim_paths=["/World/GroundPlane"],
            update_period=0.1,
            offset=RayCasterCameraCfg.OffsetCfg(pos=(0.055, 0.02, 0.0225), rot=(0,0,1,0)),
            data_types=["distance_to_image_plane", "normals", "distance_to_camera"],
            debug_vis=True,
            max_distance=10.0,
            pattern_cfg=patterns.PinholeCameraPatternCfg(
                focal_length=24.0,
                horizontal_aperture=20.955,
                height=480,
                width=640,
            ),
        )
`
However, it returns the error
`
AttributeError: 'RayCasterCamera' object has no attribute 'drift'
`

If I change the mesh_prim_paths back to "/World/GroundPlan", then it will works.

jsmith-bdai, 2024-08-09:
Can you please use the bug report template: https://github.com/isaac-sim/IsaacLab/issues/new?assignees=&labels=&projects=&template=bug.md&title=%5BBug+Report%5D+Bug+title

Please include the full stack trace in your bug report, as this seems like a case where the actual error is further up in the stack trace and the `drift` exception is just a result of a crashed initialization. Thanks!

closed 2024-08-09 by jsmith-bdai

dmtrung14, 2024-10-28:
Is this bug resolved?