isaac-sim/IsaacLab #806: [Question] Contact force for the soft bodies and gripper

Entongsu, 2024-08-08:
Hi,

I wonder if it is possible to read the contact force between the soft bodies and the gripper. I have added the contact sensor to the gripper and found that the contact force between the soft bodies and the gripper is always zero. Thank you.

cjoattesollo, 2024-08-09:
Hi,

Did you set the parameter 'activate_contact_sensors' as True?

You can check https://isaac-sim.github.io/IsaacLab/source/api/lab/omni.isaac.lab.sim.spawners.html#omni.isaac.lab.sim.spawners.from_files.UrdfFileCfg.activate_contact_sensors

Mayankm96, 2024-08-13:
Thank you for bringing this up. 

Contact forces between a soft and rigid body are not currently available. We have requested this feature from the PhysX team.

kumarlokesh2194, 2024-12-02:
Is there any update on this?

AntoineRichard, 2026-08-27:
Thanks for raising this, @Entongsu — and sorry the later follow-up here went without a reply.

To be straight about the status: this isn't a design decision, it's a capability we don't have yet. Soft-body/rigid-body contact forces aren't surfaced by the solvers Isaac Lab builds on, so the contact sensor has nothing to report — and that's still true on every backend today, including Newton, whose contact sensor covers rigid bodies and shapes only.

Closing as not planned rather than resolved, since there's no workaround to offer. If deformable contact reporting matters for your work, a feature request describing the use case is genuinely useful for prioritising it — and please reopen this if you'd rather keep it tracked here.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard