isaac-sim/IsaacLab #809: [Question] Why I can get semantic or instance image for shape cube, but not for franka pandar arm?

mickeyouyou, 2024-08-09:
Get rgb and semantic of cameras in some shape scene, it's ok. But get RGB/semantic of cameras in Franka Panda scene, semantic result is black? 

cube scene
[image]
[image]

franka scene
[image]
[image]

jsmith-bdai, 2024-08-09:
Can you provide more details on your set up for using Semantic segmentation please?

You need to fill out the `semantic_tags` to set up tags associated with the franka environment - see an example here: https://github.com/isaac-sim/IsaacLab/blob/6451d235785780167f80c14bb9d635d20aff9b29/source/standalone/tutorials/04_sensors/run_usd_camera.py#L142 for how we do this in the tutorial.

mickeyouyou, 2024-08-12:
Thanks  for your reply so much.

my scene config is here :
```python
class LiCabinetSceneCfg(ObjectTableSceneCfg):
    def __post_init__(self):
        # Set franka as robot
        print(f"FRANKA_PANDA_HIGH_PD_CFG {FRANKA_PANDA_HIGH_PD_CFG}")
        self.robot = FRANKA_PANDA_HIGH_PD_CFG.replace(prim_path="{ENV_REGEX_NS}/Robot", semantic_tags=[("class", "franka")],)
        
        # Set Cube as object
        self.object = RigidObjectCfg(
            prim_path="{ENV_REGEX_NS}/Object",
            init_state=RigidObjectCfg.InitialStateCfg(pos=[0.5, 0, 0.055], rot=[1, 0, 0, 0]),
            spawn=UsdFileCfg(
                usd_path=f"{ISAAC_NUCLEUS_DIR}/Props/Blocks/DexCube/dex_cube_instanceable.usd",
                scale=(0.8, 0.8, 0.8),
                rigid_props=RigidBodyPropertiesCfg(
                    solver_position_iteration_count=16,
                    solver_velocity_iteration_count=1,
                    max_angular_velocity=1000.0,
                    max_linear_velocity=1000.0,
                    max_depenetration_velocity=5.0,
                    disable_gravity=False,
                ),
            ),
        )
```

As you said before, `ShapeCfg`, `ConeCfg`, `CuboidCfg`, `CylinderCfg` have attribute `semantic_tags`, BUT for Franka, this is a type of `ArticulationCfg` which doesn't have attribute `semantic_tags`?

mickeyouyou, 2024-08-16:
thks for your solution. it'ok .

all class extend `SpawnerCfg` should have this  field.

[image]

closed 2024-08-16 by mickeyouyou

Entongsu, 2025-02-11:
Hi, I wonder how can you get the segmentation of each link? I just get the segmentation of the whole robot.