isaac-sim/IsaacLab #810: [Question] Question title

Rowing0914, 2024-08-09:
I just finished installing isaac-sim and was going to try out a simple script. But it failed. can anyone help me with this?

```bash
# cmd
./isaaclab.sh -p source/standalone/tutorials/00_sim/create_empty.py --headless

# Error msg
[INFO] Using python from: /opt/conda/bin/python                                                                                                                
[WARN][AppLauncher]: There are no arguments attached to the ArgumentParser object. If you have your own arguments, please load your own arguments before calling the `AppLauncher.add_app_launcher_args` method. This allows the method to check the validity of the arguments and perform checks for argument names.
[INFO][AppLauncher]: Loading experience file: /root/workspace/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/../../../../../../apps/isaaclab.python.headless.kit
[Warning] [omni.isaac.kit.simulation_app] Modules: ['omni.kit_app'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Starting kit application with the following args:  ['/opt/conda/lib/python3.10/site-packages/isaacsim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/root/workspace/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/../../../../../../apps/isaaclab.python.headless.kit', '--/app/tokens/exe-path=/opt/conda/lib/python3.10/site-packages/omni', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True', '--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--/app/fastShutdown=True', '--ext-folder', '/opt/conda/lib/python3.10/site-packages/isaacsim/exts', '--ext-folder', '/opt/conda/lib/python3.10/site-packages/isaacsim/apps', '--/renderer/activeGpu=0', '--/physics/cudaDevice=0', '--portable', '--no-window', '--/app/window/hideUi=True', '--allow-root']
Passing the following args to the base kit application:  ['--headless']
Loading user config located at: '/opt/conda/lib/python3.10/site-packages/omni/data/Kit/Isaac-Sim/4.1/user.config.json'
[Info] [carb] Logging to file: /opt/conda/lib/python3.10/site-packages/omni/logs/Kit/Isaac-Sim/4.1/kit_20240809_093849.log
2024-08-09 09:38:49 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[0.020s] Failed to solve some dependencies locally, syncing with extension registry...
[0.021s] [ext: omni.kit.async_engine-0.0.0] startup
[0.024s] [ext: omni.client-1.1.0] startup
[0.115s] [ext: omni.kit.registry.nucleus-0.0.0] startup
2024-08-09 09:38:53 [4,043ms] [Error] [omni.ext.plugin] Failed to resolve extension dependencies. Failure hints:
    [isaaclab.python.headless-1.1.0] dependency: 'omni.physx.fabric' = { version='^' } can't be satisfied. Available versions:
     (none found)
 Synced registries:
     - kit/default         : found 356 packages 
     - kit/sdk             : found 284 packages 
     - kit/community       : found 362 packages 

2024-08-09 09:38:53 [4,043ms] [Error] [omni.kit.app.plugin] Exiting app because of dependency solver failure...
Traceback (most recent call last):
  File "/root/workspace/IsaacLab/source/standalone/tutorials/00_sim/create_empty.py", line 29, in <module>
    app_launcher = AppLauncher(args_cli)
  File "/root/workspace/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/app_launcher.py", line 113, in __init__
    self._create_app()
  File "/root/workspace/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/app_launcher.py", line 569, in _create_app
    self._app = SimulationApp(self._sim_app_config, experience=self._sim_experience_file)
  File "/opt/conda/lib/python3.10/site-packages/isaacsim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py", line 213, in __init__
    from .utils import create_new_stage, open_stage, set_livesync_stage
  File "/opt/conda/lib/python3.10/site-packages/isaacsim/exts/omni.isaac.kit/omni/isaac/kit/utils.py", line 17, in <module>
    from omni.kit.usd import layers
ModuleNotFoundError: No module named 'omni.kit.usd'
```

Toni-SM, 2024-08-12:
@Rowing0914 

Have you installed the Isaac Sim - Python packages cached extension dependencies (`extscache`) as described in https://docs.omniverse.nvidia.com/isaacsim/latest/installation/install_python.html#installation-using-pip point 3

Rowing0914, 2024-08-15:
@Toni-SM Thank you for your support. I had forgotten that, and yeah, now it works!

closed 2024-08-15 by Rowing0914