isaac-sim/IsaacLab #814: Updating the camera view using set_world_pose_from_view

balaprenithreddy, 2024-08-09:
Hello,

I implemented a USD camera sensor using the sensor listed in omni.isaac.lab.sensors and attached it to a cube. I wanted the camera to always look at the target, so I used the function set_world_pose_from_view from the camera class to look at the target in the get_observation function. When I tested my env with a random agent, sometimes I see only the target but most of the times I see my cube as well in the camera. I am also updating the eye position using the cube root_pos_w and am setting the offset to position my camera above the cube.

Currently, I am confused and not sure what i missed or am doing something wrong. 

Any help is appreciated.

balaprenithreddy, 2024-08-13:
Any help?

pascal-roth, 2024-08-17:
Hi, 

if I understand you correctly, then you are defining the offsets in the `CameraCfg` and there you also set the prim_path as the one of the cubes. Afterward, you use the `set_world_pose_from_view` function. If you look into the code, then you realize that the offsets are not applied anymore when the view is defined in that way. I assume that your camera will be originally offset from the cube, but, after this operation, set inside the cube. 

I hope this helps to understand your issue.

balaprenithreddy, 2024-08-20:
Hi,

I am applying the offsets again to the current position so that my camera is outside of my cube. The main issue i am facing is while training the cube position is being changed as intented and the offsets are again being applied to the cube so that camera is set outside the cube but the eye postion is not updating. From the camera output I could only see my previous position where the cube was earlier. The eye position is not moving (or its not being captured in the camera output ) to the new position of my cube.

RandomOakForest, 2025-01-05:
Thank you for your interest in Isaac Lab. Let us know if you are still facing this issue and haven't resolved your problem. We will aim to address it in our _discussions_ section.