isaac-sim/IsaacLab #819: [Bug Report] Acceleration and Linear velocities of `Physx` off for small masses and with contact solver

pascal-roth, 2024-08-13:
### Describe the bug

The measured accelerations and velocities given by the Physx `RigidBodyAPI` do not seem reasonable in two cases:

- when a rigid body with a small mass is used. 
- when a sphere is kept at a constant velocity (set every physx step)

The error occurred during the development of the IMU sensor #619 

### Steps to reproduce

Added a test with a RigidBodyView on a robot's `base` and the `imu_link` and did the hand computation of the respective transformation. To reproduce run the test `test_offset_calculation` [here](https://github.com/isaac-sim/IsaacLab/blob/33e22a1669c5d8df44eef31ba3a21d8d66970484/source/extensions/omni.isaac.lab/test/sensors/test_imu.py#L258) .

Added a test with a cube and sphere on a GroundPlane which sets their velocities at every time step. While the cube velocities and accelerations stay approximately constant, the accelerations of the cube first decrease and then start to increase again. To reproduce run the test `test_constant_velocity` [here](https://github.com/isaac-sim/IsaacLab/blob/33e22a1669c5d8df44eef31ba3a21d8d66970484/source/extensions/omni.isaac.lab/test/sensors/test_imu.py#L134).

### System Info

- Commit: 1b8e2c0efbb61e4a5bb77e14bc3bd04fad12b633
- Isaac Sim Version: 4.1.0-rc.7+4.1.14801.71533b68.gl
- OS: Ubuntu 20.04
- GPU: RTX 4090
- CUDA: 12.2
- GPU Driver: 535.183.01 

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

- [ ] Expected behavior in the test is achieved

pascal-roth, 2024-08-13:
@Mayankm96  for viz

Mayankm96, 2024-08-13:
@Dhoeller19, this is currently blocking the IMU MR. @preist-nvidia, any advice here?

preist-nvidia, 2024-08-15:
Pierre will have a look and let you know.

preist-nvidia, 2024-08-30:
@pascal-roth - we found a bug with the acceleration compute in PhysX when the velocity setters are used on CPU and GPU. Internally we finite difference the velocities over the time step for rigid bodies and the pre-sim velocity buffer was not updated when using the getters, so the values were incorrect for the first step after the set.

The small mass case for which @Mayankm96 helped with a repro we are still investigating.

pascal-roth, 2024-08-30:
@preist-nvidia thanks for the update

closed 2024-10-18 by Dhoeller19