isaac-sim/IsaacLab #821: [Bug Report]The camera's quaternion rotation settings are misaligned with their sim counterparts

hahans, 2024-08-14:
The camera's quaternion rotation settings are misaligned with their sim counterparts 

[image]

- Isaac Sim Version: 4.1
- OS: Ubuntu 22.04
- GPU: RTX3090

Mayankm96, 2024-08-15:
This isn't a bug, rather it has to do with frame conventions in Omniverse and code.

The transforms for cameras in the Omniverse GUI are written in the "OpenGL" convention, i.e., it assumes Y-up. However, since most robotics applications don't follow this convention, we allow users to set and get camera poses in different conventions:

* "opengl" - forward axis: -Z - up axis: +Y - Offset is applied in the OpenGL (Usd.Camera) convention.
* "ros" - forward axis: +Z - up axis: -Y - Offset is applied in the ROS convention.
* "world" - forward axis: +X - up axis: +Z - Offset is applied in the World Frame convention.

If you change your settings to "opengl", the transforms should match.

hahans, 2024-08-15:
> This isn't a bug, rather it has to do with frame conventions in Omniverse and code.
>
> The transforms for cameras in the Omniverse GUI are written in the "OpenGL" convention, i.e., it assumes Y-up. [...]

Oh！Thanks for your comments, helped me a lot.

closed 2024-08-15 by hahans

Dafodilrat, 2025-11-12:
I’m currently running Isaac Sim through the GUI. Is there a way to set this parameter directly within the GUI? If not, how can I configure it so that the parameter is automatically applied when I press Play in the GUI?