isaac-sim/IsaacLab #828: [Bug Report] filter_prim_paths_expr in contact sensor not working

supersglzc, 2024-08-14:
### Describe the bug

I recently updated my IsaacSim to 4.1.0 and pulled the newest IsaacLab. However, when I run `add_sensors_on_robot.py`, it gives the following warning: 

```
[Warning] [omni.usd] Warning (secondary thread): in GetInstanceIndices at line 2190 of /buildAgent/work/ac88d7d902b57417/USD/pxr/usdImaging/usdImaging/pointInstancerAdapter.cpp -- ProtoIndex 1 out of bounds (prototypes size = 1) for (/Visuals/ContactSensor, /Visuals/ContactSensor.proto1_mesh_id0)

[Warning] [omni.usd] Warning (secondary thread): in GetInstanceIndices at line 3256 of /buildAgent/work/ac88d7d902b57417/USD/pxr/usdImaging/usdImaging/delegate.cpp -- Empty InstanceIndices (/Visuals/ContactSensor, /Visuals/ContactSensor.proto1_mesh_id0)
```
I also tested `filter_prim_paths_expr`, but seems like the filtering does not work as expected. I modified the contact sensor cfg to 
```
contact_forces = ContactSensorCfg(
        prim_path="{ENV_REGEX_NS}/Robot/LF_FOOT", update_period=0.0, history_length=6, debug_vis=True, track_air_time=True, filter_prim_paths_expr=["{ENV_REGEX_NS}/Robot/RF_FOOT"]
    )
```
This only uses one prim and one filter prim. I suppose this cfg means that we only want to detect the contact between LF_Foot and RF_Foot. However, the sensor still detects the contact between the LF_Foot and the ground plane. 

I am wondering if I misunderstand something or this is actually a bug.

Mayankm96, 2024-08-15:
Are you using the right attribute?

> https://isaac-sim.github.io/IsaacLab/source/api/lab/omni.isaac.lab.sensors.html#omni.isaac.lab.sensors.ContactSensorData.force_matrix_w

supersglzc, 2024-08-15:
I used the `current_air_time` to detect if there is a contact. I found that this was computed by `net_forces_w` so that's why it didn't work. Now I changed to manually compute the norm using `force_matrix_w`. 

I think now the sensor body can be multiples when working with `filter_prim_paths_expr`, not necessarily being one-to-many as documentation said. Maybe the documentation should be updated?

Also, do you have any idea why the warning pops up? The warning is not there if I don't use `filter_prim_paths_expr`.

supersglzc, 2024-08-23:
Follow-up: I occasionally encountered this error on some usd objects. Basically the `net_force `is not zero but the `force_matrix` is always zero. I converted the usd from urdfs. Thanks! @Mayankm96 
`[Warning] [omni.physx.tensors.plugin] GPU contact filter for collider '/World/envs/env_3853/Object_0' is not supported`

SuomiKP31, 2024-12-17:
Have you found out the issues? I'm in a similar situation where I need to obtain the contact info between two bodies within the same env (basic self collision detection), but I can't seem to find a clue.
Also a sidenote, contact sensors don't work in CPU mode right now, see: #1390

closed 2024-12-22 by supersglzc