isaac-sim/IsaacLab #831: [Bug Report] Can't launch IsaacSim via './isaaclab -s'

WYYAHYT, 2024-08-15:
### Describe the bug

I installed IsaacSim 4.1.0.0 via pip, and cloned the latest repo of IsaacLab, but failed to launch IsaacSim by excuting `./isaaclab.sh -s`.   

I know that the latest version of IsaacLab has solved the bug that can't find the path of IsaacSim installed by pip, but it seems that there is still a small bug, cause no `isaac-sim.sh` in the installation path.

**And I find a binarry file "isaacsim" in ${virtual_env}/bin, it seems to be the right file? But still not a  file ends with `.sh`.**

### Steps to reproduce
```bash
pip install isaacsim==4.1.0.0 --extra-index-url https://pypi.nvidia.com
pip install isaacsim-extscache-physics==4.1.0.0 isaacsim-extscache-kit==4.1.0.0 isaacsim-extscache-kit-sdk==4.1.0.0 --extra-index-url https://pypi.nvidia.com
git clone https://github.com/isaac-sim/IsaacLab.git
cd IsaacLab
./isaaclab.sh -s
```
output:
```
[ERROR] No Isaac Sim executable found at path: /home/xx/anaconda3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/isaac-sim.sh
```

### System Info

Describe the characteristic of your environment:

- Commit: [1b8e2c0e]
- Isaac Sim Version: [4.1.0-rc.7+4.1.14801.71533b68.gl]
- OS: [Ubuntu 22.04]
- GPU: [RTX 3090]
- CUDA: [12.2]
- GPU Driver: [535.171.04]

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

WYYAHYT, 2024-08-15:
I tried to `cp ${virtual_env}/bin/isaacsim  ${virtual_env}/lib/python3.10/site-packages/isaacsim/`, and then rename `isaacsim` as `isaac-sim`, then modify the line 90 in `isaaclab.sh`, from `local isaacsim_exe=${isaac_path}/isaac-sim.sh`, to `local isaacsim_exe=${isaac_path}/isaac-sim`

Then I tried several commands like:
```
./isaaclab.sh -s
``` 

```
./isaaclab.sh -s omni.isaac.sim.python.kit
```

```
./isaaclab.sh -s omni.isaac.sim.kit
```

And it works.

Mayankm96, 2024-08-16:
Ah I see. Thanks for finding this issue and suggesting a fix. We'll adapt the script to find the binaries from the right place.