isaac-sim/IsaacLab #834: [Question] How to implement pick and place funciton of the Franka robot arm in isaac lab?

hahans, 2024-08-15:
### Question

How to implement pick and place funciton of the Franka robot arm in isaac lab?In isaac sim, there are examples of pick and place,but how to do it again through the API in isaac lab.

Mayankm96, 2024-08-15:
We do have examples of state machines in Warp that do batch execution of skills while the environment is using a Differential-IK controller.

> https://github.com/isaac-sim/IsaacLab/blob/main/source/standalone/environments/state_machine/lift_cube_sm.py

Does this help?

hahans, 2024-08-16:
> We do have examples of state machines in Warp that do batch execution of skills while the environment is using a Differential-IK controller.
> 
> > https://github.com/isaac-sim/IsaacLab/blob/main/source/standalone/environments/state_machine/lift_cube_sm.py
> 
> Does this help?

closed 2024-08-16 by hahans