isaac-sim/IsaacLab #836: [Question] How to get timestamp of the data? Is it directly related to physics dt?

dhruvkm2402, 2024-08-16:
### Question
 Hello,
I have an RL based task and I want to collect data at 10 Hz.
My question is that how do I get timestamp of the data like in ROS? In my case, physics dt is 1/200 and decimation is 20. If my understanding is correct, the actions are being applied at 10 Hz. Similarly, If I'm collecting the data after the pre-physics step, Is it collecting the data at 10 Hz or at every simulation step?

Mayankm96, 2024-08-16:
You can get the episode length from the buffers and multiply that with `dt` to get the current "time-stamp" of the episode.

Does this answer your question?

dhruvkm2402, 2024-08-16:
Oh yes. That answers my question. 
Thank you for your help.

closed 2024-08-16 by dhruvkm2402