isaac-sim/IsaacLab #837: [Question] Always fail in registering my own designed environment

hahans, 2024-08-16:
Always fail in registering my own designed environment.I copied the isaac_tasks file structure to write setup.py,__init__.py and other files, and then used pip install-e, but when I used random_agent.py to run my own designed environment, there was still an error.
[image]

Mayankm96, 2024-08-16:
Where did you add your environment? In the `lab_tasks` extension or your own separate extension?

If it is the latter, you need to call import so that the tasks from that python module gets registered to Gym.

> https://github.com/isaac-sim/IsaacLab/blob/main/source/standalone/environments/random_agent.py#L35

hahans, 2024-08-16:
> Where did you add your environment? In the `lab_tasks` extension or your own separate extension?
>
> If it is the latter, you need to call import so that the tasks from that python module gets [...]
it was added in my own separate extension, and I also call the import when I run random_agent.py
 
[image]
But it 
didn't work

hahans, 2024-08-16:
I fixed my issue.A low-level error in the writing of the Cfg code in the environment caused the registration failure of the environment

closed 2024-08-16 by hahans