isaac-sim/IsaacLab #839: [Question] Inquiry About Mobile Robot Reinforcement Learning Support in Isaac Lab

junminkyung, 2024-08-16:
Dear Isaac Lab Team,

I hope this message finds you well.

I am currently working on a project involving reinforcement learning for mobile robots, and I am interested in using Isaac Lab for this purpose. I have a few questions regarding the support for mobile robots:

1. Are there any plans to update Isaac Lab to better support reinforcement learning for mobile robots? If so, could you provide an estimated timeline for these updates?

2. In the meantime, are there any existing examples or resources within Isaac Lab that you would recommend for getting started with reinforcement learning on mobile robots?

Thank you very much for your time and assistance. I look forward to your response.

KyleM73, 2024-08-20:
I'm not sure if this has changed in a more recent release of PhysX but previously there was no cylinder prim object for the GPU pipeline, which defaulted to an n-gon (I believe it was 18 but could be configured up to 64). As a result directly simulating wheels for RL was not physically realistic. Instead, the supported workflow was to simulate the robot as the end effector of a virtual arm parallel to the ground, and then use the action space for that virtual arm (pose deviation, velocity, acceleration, whatever you prefer) to control the physical robot with some low level pose/velocity/etc. controller. The idea was that for most indoor environment "locomotion" is not the difficult task that must be learned with RL, but rather (potentially perceptive) navigation, as well as navigation with medium-to-long horizon manipulation. I'm not sure if an official example of this workflow exists, and of course if the supported workflow has changed please someone correct me.

Mayankm96, 2024-08-21:
The PhysX team is working on proper wheel simulation. Once that's ready, we'll add some concrete examples on wheel-based mobile robots. Stay tuned :)