isaac-sim/IsaacLab #844: [Question] List of all available objects or USD?

yichaoshen-MS, 2024-08-20:
### Question

How can I find the all available objects or USD?

Mayankm96, 2024-08-21:
Do you mean on the Nucleus?

yichaoshen-MS, 2024-08-21:
> Do you mean on the Nucleus?

Maybe yes, I found most of the objects (or USD) are gotten from http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.1/Isaac, however, I cannot get the full list of all available objects or USD.

RandomOakForest, 2025-01-05:
The list of assets changes from version to version of Isaac Sim. Currently, you may find the list of categories of objects and more information on the assets of the current release [in this doc](https://docs.omniverse.nvidia.com/isaacsim/latest/features/environment_setup/assets/usd_assets_overview.html). Let us know if you need further help. Thank you.

closed 2025-01-05 by RandomOakForest