isaac-sim/IsaacLab #845: [Question] How to change the backgrounds each sub-env?

yichaoshen-MS, 2024-08-20:
### Question

I would like to know the how to change the backgrounds each sub-env? like the demos shows
[image]

mickeyouyou, 2024-08-26:
use different Environment Assets(https://docs.omniverse.nvidia.com/isaacsim/latest/features/environment_setup/assets/usd_assets_environments.html#environment-assets)

yichaoshen-MS, 2024-08-26:
> use different Environment Assets(https://docs.omniverse.nvidia.com/isaacsim/latest/features/environment_setup/assets/usd_assets_environments.html#environment-assets)

Thank you very much for your thoughtful and detailed response. In addition, could I ask where I can find the list of available assets(objects, like the flags, Red wine glass in the picture),

RandomOakForest, 2025-01-05:
Please refer to [this issue](https://github.com/isaac-sim/IsaacLab/issues/844), and let us know if you need further help. Thank you for your interest in Isaac Lab.

closed 2025-01-05 by RandomOakForest