isaac-sim/IsaacLab #848: [Question] Inference during simulation

SonYeongGwang, 2024-08-21:
Hi, recently, I found that when I tried to inference with my model (CNN) during simulation, it makes my simulation stuck.

Here's my code. Most of the parts are same with Isaac Lab tutorial (source/standalone/tutorials/01_assets/run_rigid_object.py). I just added

```
class fooNet(nn.Module):
    def __init__(self):
        super(fooNet, self).__init__()
        self.conv1 = nn.Conv2d(3, 6, 12)

    def forward(self, x):
        x = F.relu(self.conv1(x))

        return x
```
and

```
        if count % 250 == 0:
            # reset counters
            sim_time = 0.0
            count = 0
            # reset root state
            root_state = cone_object.data.default_root_state.clone()
            # sample a random position on a cylinder around the origins
            root_state[:, :3] += origins
            root_state[:, :3] += math_utils.sample_cylinder(
                radius=0.1, h_range=(0.25, 0.5), size=cone_object.num_instances, device=cone_object.device
            )
            # write root state to simulation
            cone_object.write_root_state_to_sim(root_state)
            # reset buffers
            cone_object.reset()
            print("----------------------------------------")
            print("[INFO]: Resetting object state...")

            input_t = torch.rand((1, 3, 480, 640)).to('cuda')
            output = model(input_t)
            print(output)
```

This cause the simulation stuck in somewhere (without any error message, it just stopped)

However, if I change the device of the model (cuda -> cpu) and sustain device for the input (cuda) it works.

What is the reason for this and how to handle this situation?

Thanks.

Mayankm96, 2024-08-21:
I will have to try it out to see the exact issue, but as a guess: can you wrap all operations (particularly sim-related) with `torch.inference_mode()`? Maybe some gradients are getting accumulated and causing memory issues.

SonYeongGwang, 2024-08-21:
Thanks for answer, I tried `torch.inference_mode()` of my code but it still not working..

I've also monitored vram usage but there's still a free space.

As a detailed explanation on my problem, the code stuck exactly at the moment where actual calculation takes place using conv2d method. nn.ReLU(), nn.MaxPool2d() methods works find.

Mayankm96, 2024-08-21:
I changed the model to the following and it works:

```python
class FooNet(torch.nn.Module):
    """A simple network."""

    def __init__(self, device="cuda"):
        super(FooNet, self).__init__()
        self.conv1 = torch.nn.Conv2d(3, 6, 12, device=device)

    def forward(self, x):
        x = F.relu(self.conv1(x))
        return x
```

SonYeongGwang, 2024-08-21:
Unfortunately, still same...

I attached the whole code.

```
import argparse

from omni.isaac.lab.app import AppLauncher

# add argparse arguments
parser = argparse.ArgumentParser(description="Tutorial on spawning and interacting with a rigid object.")
# append AppLauncher cli args
AppLauncher.add_app_launcher_args(parser)
# parse the arguments
args_cli = parser.parse_args()

# launch omniverse app
app_launcher = AppLauncher(args_cli)
simulation_app = app_launcher.app

"""Rest everything follows."""

import torch
import torch.nn as nn
import torch.nn.functional as F

import omni.isaac.core.utils.prims as prim_utils

import omni.isaac.lab.sim as sim_utils
import omni.isaac.lab.utils.math as math_utils
from omni.isaac.lab.assets import RigidObject, RigidObjectCfg
from omni.isaac.lab.sim import SimulationContext

class fooNet(nn.Module):
    def __init__(self, device="cuda"):
        super(fooNet, self).__init__()
        self.conv1 = nn.Conv2d(3, 6, 12, device=device)
        self.pool = nn.MaxPool2d(2, 2)
        self.relu = nn.ReLU()

    def forward(self, x):
        x = self.conv1(x)

        return x

def design_scene():
    """Designs the scene."""
    # Ground-plane
    cfg = sim_utils.GroundPlaneCfg()
    cfg.func("/World/defaultGroundPlane", cfg)
    # Lights
    cfg = sim_utils.DomeLightCfg(intensity=2000.0, color=(0.8, 0.8, 0.8))
    cfg.func("/World/Light", cfg)

    # Create separate groups called "Origin1", "Origin2", "Origin3"
    # Each group will have a robot in it
    origins = [[0.25, 0.25, 0.0], [-0.25, 0.25, 0.0], [0.25, -0.25, 0.0], [-0.25, -0.25, 0.0]]
    for i, origin in enumerate(origins):
        prim_utils.create_prim(f"/World/Origin{i}", "Xform", translation=origin)

    # Rigid Object
    cone_cfg = RigidObjectCfg(
        prim_path="/World/Origin.*/Cone",
        spawn=sim_utils.ConeCfg(
            radius=0.1,
            height=0.2,
            rigid_props=sim_utils.RigidBodyPropertiesCfg(),
            mass_props=sim_utils.MassPropertiesCfg(mass=1.0),
            collision_props=sim_utils.CollisionPropertiesCfg(),
            visual_material=sim_utils.PreviewSurfaceCfg(diffuse_color=(0.0, 1.0, 0.0), metallic=0.2),
        ),
        init_state=RigidObjectCfg.InitialStateCfg(),
    )
    cone_object = RigidObject(cfg=cone_cfg)

    # return the scene information
    scene_entities = {"cone": cone_object}
    return scene_entities, origins

def run_simulator(sim: sim_utils.SimulationContext, entities: dict[str, RigidObject], origins: torch.Tensor):
    """Runs the simulation loop."""
    # Extract scene entities
    # note: we only do this here for readability. In general, it is better to access the entities directly from
    #   the dictionary. This dictionary is replaced by the InteractiveScene class in the next tutorial.
    cone_object = entities["cone"]
    # Define simulation stepping
    sim_dt = sim.get_physics_dt()
    sim_time = 0.0
    count = 0

    device = torch.device('cuda')
    # print("device:", device)

    model = fooNet(device)
    # model.to(device)

    # Simulate physics
    while simulation_app.is_running():
        # reset
        if count % 250 == 0:
            # reset counters
            sim_time = 0.0
            count = 0
            # reset root state
            root_state = cone_object.data.default_root_state.clone()
            # sample a random position on a cylinder around the origins
            root_state[:, :3] += origins
            root_state[:, :3] += math_utils.sample_cylinder(
                radius=0.1, h_range=(0.25, 0.5), size=cone_object.num_instances, device=cone_object.device
            )
            # write root state to simulation
            cone_object.write_root_state_to_sim(root_state)
            # reset buffers
            cone_object.reset()
            print("----------------------------------------")
            print("[INFO]: Resetting object state...")

            input_t = torch.rand((1, 3, 480, 640)).to(device)
            output = model(input_t)
            print(output)

        # apply sim data
        cone_object.write_data_to_sim()
        # perform step
        sim.step()
        # update sim-time
        sim_time += sim_dt
        count += 1
        # update buffers
        cone_object.update(sim_dt)
        # print the root position
        if count % 50 == 0:
            print(f"Root position (in world): {cone_object.data.root_state_w[:, :3]}")

def main():
    """Main function."""
    # Load kit helper
    sim_cfg = sim_utils.SimulationCfg()
    sim = SimulationContext(sim_cfg)
    # Set main camera
    sim.set_camera_view(eye=[1.5, 0.0, 1.0], target=[0.0, 0.0, 0.0])
    # Design scene
    scene_entities, scene_origins = design_scene()
    scene_origins = torch.tensor(scene_origins, device=sim.device)
    # Play the simulator
    sim.reset()
    # Now we are ready!
    print("[INFO]: Setup complete...")
    # Run the simulator
    run_simulator(sim, scene_entities, scene_origins)

if __name__ == "__main__":
    # run the main function
    main()
    # close sim app
    simulation_app.close()
``` 

I can see Isaac Sim window pops up and the simulation steps until it meet the line where the convolution operation starts.

I always stucked like the image below

[image]

Mayankm96, 2024-08-21:
On my PC it works with your script as well. Can you share more details on your system configuration and torch version?

SonYeongGwang, 2024-08-21:
I ran the script with
- Ryzen 7 5700
- RTX 3070
- Isaac sim 4.1.0
- conda environment configured following the manual (https://isaac-sim.github.io/IsaacLab/source/setup/installation/binaries_installation.html)

I've activated my conda env and all the tutorials and examples works well.

Thanks!

Mayankm96, 2024-08-21:
Great to hear! :)

closed 2024-08-21 by Mayankm96