isaac-sim/IsaacLab #849: [Bug Report] Change the simulator parameter during the simulation

Entongsu, 2024-08-21:
Hi,

I have tried to change the simulator parameters during the simulation, such as simulation_hexahedral_resolution and contact_offset... But I have gotten the warning Changing parameter during simulation is not supported". I am wondering whether I can change this parameter during the simulation. Thank you.

Mayankm96, 2024-08-24:
You can only change these parameters through the tensor API. I doubt you can do this for the `simulation_hexahedral_resolution` since this parameter affects the baking of the assets for FEM that only happens once when the simulation starts playing (i.e. sim.reset() or sim.play()) is called.

We disable some of the writing/listening to UI to speed up read-write operations (also called enabling Fabric). If you want more UI features, you can run the scripts with `--disable_fabric` command, or set the argument `use_fabric` to False in the [`SimulationCfg`](https://isaac-sim.github.io/IsaacLab/source/api/lab/omni.isaac.lab.sim.html#omni.isaac.lab.sim.SimulationCfg.use_fabric).

Entongsu, 2024-08-25:
Thank you very much for your information. Could you provide some hints about how to change these parameters using the tensor API? Thank you.

RandomOakForest, 2025-01-05:
Thank you for posting this. It seems an Isaac Sim topic. If you still need help with this issue, please open a [topic in the Isaac Sim forums](https://forums.developer.nvidia.com/c/omniverse/simulation/69).

closed 2025-01-05 by RandomOakForest