isaac-sim/IsaacLab #852: [Bug Report] RTX Driver verification failed ISAAC 4.0 Nvidia driver  560

paramoza999, 2024-08-22:
isaaclab.bat -p source/standalone/demos/quadrupeds.py
[INFO] Using python from: C:\ProgramData\anaconda3\envs\isaaclab\python
[WARN][AppLauncher]: There are no arguments attached to the ArgumentParser object. If you have your own arguments, please load your own arguments before calling the `AppLauncher.add_app_launcher_args` method. This allows the method to check the validity of the arguments and perform checks for argument names.
[INFO][AppLauncher]: Loading experience file: C:\Users\ozapa\IsaacLab\source\extensions\omni.isaac.lab\omni\isaac\lab\app\..\..\..\..\..\..\apps\isaaclab.python.kit
Starting kit application with the following args:  ['C:\\Users\\ozapa\\IsaacLab\\_isaac_sim\\exts\\omni.isaac.kit\\omni\\isaac\\kit\\simulation_app.py', 'C:\\Users\\ozapa\\IsaacLab\\source\\extensions\\omni.isaac.lab\\omni\\isaac\\lab\\app\\..\\..\\..\\..\\..\\..\\apps\\isaaclab.python.kit', '--/app/tokens/exe-path=C:\\Users\\ozapa\\IsaacLab\\_isaac_sim\\kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True', '--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--/app/fastShutdown=True', '--ext-folder', 'C:\\Users\\ozapa\\IsaacLab\\_isaac_sim/exts', '--ext-folder', 'C:\\Users\\ozapa\\IsaacLab\\_isaac_sim/apps', '--/renderer/activeGpu=0', '--/physics/cudaDevice=0', '--portable', '--/app/window/hideUi=False']
Passing the following args to the base kit application:  []
Loading user config located at: 'c:/users/ozapa/isaaclab/_isaac_sim/kit/data/Kit/Isaac-Sim/4.1/user.config.json'
[Info] [carb] Logging to file: c:/users/ozapa/isaaclab/_isaac_sim/kit/logs/Kit/Isaac-Sim/4.1/kit_20240821_174104.log
2024-08-22 00:41:04 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[0.137s] [ext: omni.kit.async_engine-0.0.0] startup
[0.517s] [ext: omni.stats-0.0.0] startup
[0.520s] [ext: omni.client-1.1.0] startup
[0.539s] [ext: omni.datastore-0.0.0] startup
[0.540s] [ext: omni.blobkey-1.1.0] startup
[0.540s] [ext: omni.ujitso.default-1.0.0] startup
[0.542s] [ext: omni.rtx.shadercache.vulkan-1.0.0] startup
[0.545s] [ext: omni.assets.plugins-0.0.0] startup
[0.548s] [ext: omni.gpu_foundation-0.0.0] startup
[0.569s] [ext: carb.windowing.plugins-1.0.0] startup
[0.672s] [ext: omni.kit.renderer.init-0.0.0] startup
[0.673s] [ext: omni.kit.pipapi-0.0.0] startup
[0.675s] [ext: omni.kit.pip_archive-0.0.0] startup
[0.676s] [ext: omni.pip.compute-1.3.1] startup
[0.678s] [ext: omni.pip.cloud-1.1.3] startup
[0.682s] [ext: omni.isaac.core_archive-2.3.0] startup
[0.683s] [ext: omni.mtlx-0.1.0] startup
[0.684s] [ext: omni.usd.config-1.0.3] startup
[0.698s] [ext: omni.gpucompute.plugins-0.0.0] startup
[0.702s] [ext: omni.usd.libs-1.0.0] startup
[0.845s] [ext: omni.isaac.ml_archive-2.0.1] startup
[0.846s] [ext: omni.kit.loop-isaac-1.2.0] startup
[0.847s] [ext: omni.kit.test-0.0.0] startup
[0.925s] [ext: omni.kit.telemetry-0.5.0] startup
[0.932s] [ext: omni.appwindow-1.1.8] startup
[0.953s] [ext: omni.kit.renderer.core-1.0.1] startup
[0.959s] [ext: omni.kit.renderer.capture-0.0.0] startup
[0.965s] [ext: omni.kit.renderer.imgui-1.0.1] startup
[0.976s] [ext: omni.ui-2.23.11] startup
[0.987s] [ext: omni.kit.mainwindow-1.0.3] startup
[0.989s] [ext: carb.audio-0.1.0] startup
[0.991s] [ext: omni.uiaudio-1.0.0] startup
[0.993s] [ext: omni.kit.uiapp-0.0.0] startup
[0.993s] [ext: omni.usd.schema.semantics-0.0.0] startup
[1.072s] [ext: omni.usd.schema.isaac-2.0.2] startup
[1.082s] [ext: omni.usd.schema.scene.visualization-2.0.2] startup
[1.085s] [ext: omni.usd.schema.audio-0.0.0] startup
[1.091s] [ext: omni.usd.schema.forcefield-106.0.16] startup
[1.112s] [ext: omni.usd.schema.physx-106.0.16] startup
[1.134s] [ext: omni.usd.schema.omnigraph-1.0.0] startup
[1.142s] [ext: omni.usd.schema.geospatial-0.0.0] startup
[1.147s] [ext: omni.usd.schema.anim-0.0.0] startup
[1.162s] [ext: omni.kit.window.popup_dialog-2.0.24] startup
2024-08-22 00:41:05 [1,115ms] [Error] [carb.graphics-vulkan.plugin] Flags 0x6 must be the same for both device and instance.
[1.167s] [ext: omni.usd_resolver-1.0.0] startup
[1.173s] [ext: omni.kit.widget.nucleus_connector-1.1.5] startup
[1.176s] [ext: omni.timeline-1.0.10] startup
[1.180s] [ext: omni.usd.core-1.2.11] startup
[1.182s] [ext: omni.kit.usd_undo-0.1.8] startup
[1.184s] [ext: omni.hydra.scene_delegate-0.3.3] startup
[1.186s] [ext: omni.activity.core-1.0.1] startup
[1.189s] [ext: omni.resourcemonitor-105.0.1] startup
[1.193s] [ext: omni.graph.exec-0.9.3] startup
[1.194s] [ext: omni.kit.actions.core-1.0.0] startup
[1.198s] [ext: usdrt.scenegraph-7.4.8] startup
[1.252s] [ext: omni.kit.audiodeviceenum-1.0.1] startup
[1.255s] [ext: omni.kit.exec.core-0.13.0] startup
[1.260s] [ext: omni.kit.commands-1.4.9] startup

|---------------------------------------------------------------------------------------------|
| Driver Version: 560.81        | Graphics API: Vulkan
[... 214 lines cut ...]
[5.088s] [ext: omni.kit.property.light-1.0.8] startup
[5.090s] [ext: omni.kit.window.commands-0.2.5] startup
[5.093s] [ext: omni.kit.widget.timeline-105.0.1] startup
[5.097s] [ext: omni.kit.menu.stage-1.2.5] startup
[5.099s] [ext: omni.kit.property.bundle-1.2.11] startup
[5.101s] [ext: omni.kit.window.status_bar-0.1.6] startup
[5.105s] [ext: omni.kit.window.title-1.1.3] startup
[5.106s] [ext: omni.replicator.isaac-1.15.0] startup
[5.129s] [ext: omni.replicator.replicator_yaml-2.0.5] startup
[5.157s] [ext: omni.rtx.settings.core-0.6.0] startup
[5.167s] [ext: omni.kit.window.console-0.2.12] startup
[5.174s] [ext: semantics.schema.editor-0.3.4] startup
[5.180s] [ext: semantics.schema.property-1.0.2] startup
[5.185s] [ext: omni.kit.window.script_editor-1.7.6] startup
[5.190s] [ext: omni.kit.profiler.window-2.2.1] startup
2024-08-22 00:41:09 [5,147ms] [Warning] [omni.kit.profiler.window] remove _SpanInstance.__lt__ and use insort 'key' arg instead
[5.200s] [ext: omni.isaac.utils-1.0.1] startup
[5.205s] [ext: omni.isaac.cloner-0.8.1] startup
[5.209s] [ext: omni.kit.window.stats-0.1.6] startup
[5.211s] [ext: isaaclab.python-1.1.0] startup
[5.212s] Simulation App Starting
2024-08-22 00:41:09 [5,354ms] [Error] [gpu.foundation.plugin] Raytracing is not supported on GPU: 0. Hardware, driver or Operating system upgrade is required.
2024-08-22 00:41:09 [5,354ms] [Error] [carb.scenerenderer-rtx.plugin] rtx driver verification failed.
2024-08-22 00:41:09 [5,354ms] [Warning] [omni.hydra.rtx] HydraEngine rtx failed creating scene renderer.
[5.473s] app ready

Ailiee, 2024-09-10:
I have the same question. May I ask how you resolved it later? Did you reinstall the drivers and CUDA?

Sathvik-cy, 2024-09-18:
I have the same issue. Any update ?

robotNo201, 2024-09-19:
same issue

robotNo201, 2024-09-19:
window11, rtx4090

robotNo201, 2024-09-19:
Please don't update more functionality if even this cannot be fixed.

J3ykob, 2024-10-02:
Same issue here:(

kellyguo11, 2024-10-02:
Can you please verify that `nvidia-smi` works correctly on the machine and try with the recommended driver versions found [here](https://docs.omniverse.nvidia.com/isaacsim/latest/common/technical-requirements.html)?

J3ykob, 2024-10-02:
> Can you please verify that `nvidia-smi` works correctly on the machine and try with the recommended driver versions found [here](https://docs.omniverse.nvidia.com/isaacsim/latest/common/technical-requirements.html)?

nvidia-smi
Wed Oct  2 23:42:32 2024
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 561.09                 Driver Version: 561.09         CUDA Version: 12.6     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                  Driver-Model | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 4070 ...  WDDM  |   00000000:01:00.0  On |                  N/A |
| N/A   66C    P8              8W /   30W |     499MiB /   8188MiB |      1%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+

+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A     12120    C+G   C:\Windows\explorer.exe                     N/A      |
|    0   N/A  N/A     16920    C+G   ...werToys\PowerToys.PowerLauncher.exe      N/A      |
|    0   N/A  N/A     21176    C+G   ...gicPodsService\MagicPodsService.exe      N/A      |
+-----------------------------------------------------------------------------------------+

---

nvcc --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2024 NVIDIA Corporation
Built on Tue_Feb_27_16:28:36_Pacific_Standard_Time_2024
Cuda compilation tools, release 12.4, V12.4.99
Build cuda_12.4.r12.4/compiler.33961263_0

Jellal-17, 2024-10-03:
> Can you please verify that `nvidia-smi` works correctly on the machine and try with the recommended driver versions found [here](https://docs.omniverse.nvidia.com/isaacsim/latest/common/technical-requirements.html)?

nvidia-smi
Thu Oct  3 10:13:47 2024
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 537.58                 Driver Version: 537.58       CUDA Version: 12.2     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                     TCC/WDDM  | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce RTX 4070 ...  WDDM  | 00000000:01:00.0 Off |                  N/A |
| N/A   50C    P0              18W / 130W |      0MiB /  8188MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+

+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|  No running processes found                                                           |
+---------------------------------------------------------------------------------------+

asrvn, 2024-10-11:
Same issue here. Win 11, RTX 4070

RandomOakForest, 2025-01-05:
Thank you for posting this. Could you confirm you still have this issue using Isaac Sim 4.2 and Isaac Lab 1.4.0? We will then aim to reproduce this on our side. Thank you.

closed 2025-01-14 by RandomOakForest

Nick-bit233, 2025-01-17:
Same issue here :( 
I have tried every examples in /source/standalone, it just keeps happen.
However, using "isaaclab.bat -s" to start isaacsim runs pretty well.
What I am using is listed here:
OS: Win11
Isaac Sim: 4.2.0.2 (installed by pip)
Isaac Lab: 1.4.0
Graphics: RTX 4070 Laptop
Driver Version: 560.94
CUDA Version: 12.6

constant5, 2025-01-31:
Same issue.  I had a working install yesterday but it seems a driver update broke it.  Things I have tried:
1) Downgrade Nvidia Driver to 537.58
2) Downgrade to CUDA 12.1
3) Deleted %APPDATA%/local/ov
4) Deleted and reinstalled IssacSim venv 

My GPU and current driver setup setup is:
```
$ nvidia-smi
Thu Jan 30 16:59:59 2025
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 537.58                 Driver Version: 537.58       CUDA Version: 12.2     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                     TCC/WDDM  | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce RTX 4060 ...  WDDM  | 00000000:01:00.0 Off |                  N/A |
| N/A   62C    P3              11W /  35W |      0MiB /  8188MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+

+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|  No running processes found                                                           |
+---------------------------------------------------------------------------------------+
```

I reinstalled IssacSim and tried to run 'issacsim'
```
[[4.257s] [ext: omni.appwindow-1.1.9] startup
2025-01-31 00:18:47 [4,147ms] [Warning] [gpu.foundation.plugin] Skipping unsupported non-NVIDIA GPU: Intel(R) Iris(R) Xe Graphics
2025-01-31 00:18:47 [4,147ms] [Warning] [gpu.foundation.plugin] Skipping unsupported non-NVIDIA GPU: Intel(R) Iris(R) Xe Graphics
2025-01-31 00:18:47 [4,147ms] [Warning] [gpu.foundation.plugin] Skipping unsupported non-NVIDIA GPU: Intel(R) Iris(R) Xe Graphics
2025-01-31 00:18:47 [4,148ms] [Warning] [gpu.foundation.plugin] Skipping unsupported non-NVIDIA GPU: Intel(R) Iris(R) Xe Graphics
2025-01-31 00:18:47 [4,148ms] [Error] [gpu.foundation.plugin] Multiple Installable Client Drivers (ICDs) are found for the same GPU on the system. This causes the same GPU to be reported multiple times by more than one driver and leads to instability or crash.
For Windows, a clean driver reinstallation may be required. It might also happen after connecting with some certain remote desktop applications.
[4.284s] [ext: omni.kit.renderer.core-1.0.2] startup

|---------------------------------------------------------------------------------------------|
| Driver Version: 537.58        | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 4060 Laptop.. |        |     | 7973    MB | 10de      | b8f57529.. |
|     |                                  |        |     |            | 28a0      | ed609946.. |
|     |                                  |        |     |            | 1         |            |
|---------------------------------------------------------------------------------------------|
| 1   | Intel(R) Iris(R) Xe Graphics     |        |     | 32603   MB | 8086      | ccc80100.. |
|     |                                  |        |     |            | a7a0      | 8680a0a7.. |
|     |                                  |        |     |            | N/A       |            |
|---------------------------------------------------------------------------------------------|
| 2   | Intel(R) Iris(R) Xe Graphics     |        |     | 32603   MB | 8086      | a8e2f10a.. |
|     |                                  |        |     |            | a7a0      | 8680a0a7.. |
|     |                                  |        |     |            | N/A       |            |
|=============================================================================================|
| OS: Windows 11 Pro, Version: 10.0 (23H2), Build: 22631, Kernel: 10.0.22621.4746
| Processor: 13th Gen Intel(R) Core(TM) i9-13900H
| Cores: 14 | Logical Cores: 20
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 65207 | Free Memory: 26591
| Total Page/Swap (MB): 69303 | Free Page/Swap: 21521
|---------------------------------------------------------------------------------------------|
[4.311s] [ext: omni.kit.renderer.capture-0.0.0] startup
[4.338s] [ext: omni.kit.renderer.imgui-1.0.2] startup
[... 22 lines cut ...]
[... cut ...]
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 65207 | Free Memory: 26545
| Total Page/Swap (MB): 69303 | Free Page/Swap: 21485
|---------------------------------------------------------------------------------------------|
2025-01-31 00:18:48 [4,486ms] [Error] [omni.gpu_foundation_factory.plugin] Failed to create any GPU devices, including an attempt with compatibility mode.
```
It seems that the omni.gpu_foundation_factory.plugin cannot identify the GPU. 

And I get this popup. 

[image]

MikeTheSnowman, 2025-03-04:
I too am encountering this issue and would like some advice and guidance on what can be done.
@RandomOakForest , I noticed that the prevailing advice you've been giving to people has been to "downgrade" drivers to the recommended version, which is currently 537.58. These drivers were released in October 2023, and are now relatively old now at this point. Because of that, I do not believe that downgrading to drivers that old should be an acceptable course of action here.

On my Windows 11 laptop with an RTX 4090, the full command I used is:
`.\isaac-sim.bat --/renderer/multiGpu/enabled=false`

Unfortunately, there are still three remaining issues that I'm needing assistance with:
```log
2025-03-04 13:07:05 [1,929ms] [Error] [carb.graphics-vulkan.plugin] Flags 0x6 must be the same for both device and instance.
2025-03-04 13:07:13 [10,621ms] [Error] [gpu.foundation.plugin] Raytracing is not supported on GPU: 0. Hardware, driver or Operating system upgrade is required.
2025-03-04 13:07:13 [10,621ms] [Error] [carb.scenerenderer-rtx.plugin] rtx driver verification failed.
```

Are there any other possible options that I can try using to address the remaining three errors that I just mentioned? Thanks!

MiangChen, 2025-03-12:
I meet same problem.
```Failed to create any GPU devices, including an attempt with compatibility mode.```
but the ubuntu is newly installed, and I only install the nvidia driver for once