isaac-sim/IsaacLab #853: [Question] Guidance Needed for Creating a Navigation Task in a New Gym Environment

Kenn3o3, 2024-08-22:
### Objective

I am attempting to develop a navigation task inspired by `omni.isaac.tasks > manager based > navigation`, with a similar training approach. The task involves a robot navigating to specified x, y coordinates and orientation. However, during inference, I want to integrate real-time camera outputs. For instance, a robot (anymal_c) will use camera images to set a goal pose (x, y, orientation). Upon reaching the initial goal, it will determine subsequent goals dynamically based on new camera images.

#### Current Progress

- **Training Setup:** I've mirrored the training setup of the original task, where the robot navigates to randomly sampled displacements and orientations.
- **Camera Integration:** I've managed to attach a camera to the robot within the `MyScene(interactivescene)` class and render its output as a tensor in the while loop simulation in `play.py`. However, this output is not yet treated as an observation.

#### Questions
I am uncertain about the necessary modifications or additions to the file structure to properly register this new gym environment for navigation tasks. Despite reviewing all related tutorials three times and reading the API documents over several months, I am still unclear about it.

1. What files should I create or modify to register a new gym environment tailored for this navigation task?
2. How can I ensure that the camera's output is integrated as an observation within the environment?

Could anyone provide guidance or point towards relevant examples or documentation that could assist in setting up this environment? 
I am stuck with simply understanding this library and all the APIs for more than a month. I have been working on this as a single person without any guidance. Any help would be greatly appreciated.

Thank you very much for your time and help.

Kenn3o3, 2024-08-29:
May I know if there will be any update?

Mayankm96, 2024-10-09:
1. Please follow the guide here: https://isaac-sim.github.io/IsaacLab/source/overview/developer-guide/template.html
2. We have an example now for using images. Please check here: https://isaac-sim.github.io/IsaacLab/source/overview/environments.html