isaac-sim/IsaacLab #873: [Question] Implementation status of the operational space and joint impedance controllers

ozhanozen, 2024-08-23:
Hello everyone, 

I need to use an operational space controller for some manipulation tasks in the context of RL/sim2real. I have noticed that an implementation of such a controller already exists in the [controllers folder](https://github.com/isaac-sim/IsaacLab/blob/main/source/extensions/omni.isaac.lab/omni/isaac/lab/controllers/operational_space.py), along with some other task-space controllers. However, I do not see these in the API documentation, and the last changes on these files were quite some time ago. 

Are these controllers (e.g., operational space and joint impedance controllers) fully functional for RL at the moment, or do they have any limitations that need to be solved?

Mayankm96, 2024-08-24:
I wrote these implementations some time ago, but due to other commitments, I wasn't able to create proper unit tests or examples for them. Although I'm confident in their mathematical accuracy, I can only guarantee their current functionality with thorough testing.

I'm not sure if the team has the bandwidth to look into this right now, but @jtigue-bdai, this might interest you.

jtigue-bdai, 2024-08-24:
Yeah we can assign this one to me.

Just to clarify my understanding of the ask:
- add unit tests for joint space controllers
- add unit tests for task space controllers
- update documentation with potential examples and limitations

ozhanozen, 2024-08-25:
Thanks a lot @Mayankm96 and @jtigue-bdai !

Please let me know if there is something I could support you with, e.g., documentation or examples. I would be glad to contribute.

jtigue-bdai, 2024-08-26:
@ozhanozen, if you are interested (especially if you are wanting use it in the near term), I have started a branch and draft pr [PR](https://github.com/isaac-sim/IsaacLab/pull/879) that adds operational space controller to the documentation and has a skeleton of a test file. I think the best way to get a sense of limitations and how to use is to build out some test cases for the controller. So feel free to build out some tests. You can push to this branch and we can work together on it. 

Now there are a bunch of parameters for that controller so feel free to create any tests outside of what I have defined (especially if they are towards your use case). Once we have a few test cases for this controller we can convert some of them to examples. 

Also if you see anything that could use some extra explanation in the API feel free to make changes or suggestions on the PR.

ozhanozen, 2024-08-27:
Thanks @jtigue-bdai, I will work on the PR branch then.

commit fe976d76 (2024-12-16): Adds `OperationSpaceController` to docs and tests and implement corresponding action/action_cfg classes (#913)

# Description

This PR adds the `OperationalSpaceController` to the docs and provides
some tests for its parametric features. Moreover, it implements the
corresponding `OperationalSpaceControllerAction` and
`OperationalSpaceControllerActionCfg` classes so they can be used with
manager-based environments.

Fixes #873 

## Type of change

- Bug fix (non-breaking change which fixes an issue)
- New feature (non-breaking change which adds functionality)
- This change requires a documentat [... cut]