isaac-sim/IsaacLab #874: [Bug Report] Contact Sensor not working on cpu

choijoshua, 2024-08-23:
When I run my environment on cpu, the get_net_contact_sensor() function on my rigidprimview, it doesn't seem to give me any value, nor does the get_contact_force_matrix() function. Are they only supposed to work on gpu?

alessandroassirelli98, 2024-10-29:
I am also experiencing the same issue

RandomOakForest, 2025-01-05:
Our apologies we couldn't follow up in a timely manner. Could you confirm if you still have this issue using Isaac Sim 4.2 and Isaac Lab 1.4.0? Thank you.

gbrlb, 2025-01-24:
@RandomOakForest Yes, I'm experiencing this issue as well with Isaac Sim 4.2 and Isaac Lab 1.4.0. Please let me know if any updates or solutions are available.

mpgussert, 2025-02-18:
hello gbrlb!

I need some more details in order to help you :) What function are you calling specifically?  do you mean the `get_net_contact_forces` member of `contact_physx_view`? 

What is your environment and how are you "running it on the CPU?  what workflow are you using?

gbrlb, 2025-02-19:
Hi @mpgussert!
I'm using illegal_contact in [terminations.py](https://github.com/isaac-sim/IsaacLab/blob/6a415df23ba73854fdafd5914d5c95f30036300f/source/isaaclab/isaaclab/envs/mdp/terminations.py#L150) and launching the simulation with `./isaaclab.sh` It works fine with `--device=cuda`, but on `--device=cpu`, it always returns 0.
- Using Manager-Based Workflow with locomotion tasks.
- Running on IsaacLab 2.0 and IsaacSim 4.5.0
Thanks!

mpgussert, 2025-02-20:
Can you please provide me with a standalone script that I can run which demonstrates the issue?  I don't have a manager based MDP environment on hand that uses this termination in order to test, and I don't know if the one I am constructing has the same assumptions as yours. Sorry!

Something that can be run like `./isaaclab.sh -p test_script.py --device=cuda` and `./isaaclab.sh -p test_script.py --device=cpu` would be ideal

This way I have something explicit I can dig into and share :)

Thank you for your patience!

kellyguo11, 2025-05-24:
Could you see if this commit fixes the issue? https://github.com/isaac-sim/IsaacLab/commit/cb5c7cab3573f03e124f340259af83ee4f6cda60

ooctipus, 2025-06-26:
Hi gbrlb,

with
isaac-lab 2.1
isaac-sim 4.5

runing
`./isaaclab.sh -p scripts/reinforcement_learning/rsl_rl/train.py --task Isaac-Velocity-Flat-Anymal-C-v0 --seed 50 --device cpu --num_envs 16`

I can observe that 
```
def illegal_contact(env: ManagerBasedRLEnv, threshold: float, sensor_cfg: SceneEntityCfg) -> torch.Tensor:
    """Terminate when the contact force on the sensor exceeds the force threshold."""
    # extract the used quantities (to enable type-hinting)
    contact_sensor: ContactSensor = env.scene.sensors[sensor_cfg.name]
    net_contact_forces = contact_sensor.data.net_forces_w_history
    # check if any contact force exceeds the threshold
    return torch.any(
        torch.max(torch.norm(net_contact_forces[:, :, sensor_cfg.body_ids], dim=-1), dim=1)[0] > threshold, dim=1
    )
```
behaves correctly with cpu and does return non-zero value when base contacts the floor.

Can you run the script above and see if it is working properly?
If it doesn't work properly, please let me know you system configurations, I can see if I can reproduce
If it does work properly, then could you please share the minimal reproducible example you ran into and I can try run it in my end

closed 2025-06-26 by ooctipus