isaac-sim/IsaacLab #875: [Bug Report] Video Recording Wrapper Broken

arhanjain, 2024-08-23:
### Describe the bug

When wrapping the IsaacLab environment with the `gym.wrappers.RecordVideo`, I'm no longer able to get it to output videos successfully.

### Steps to reproduce

```
import argparse
from omni.isaac.lab.app import AppLauncher

parser = argparse.ArgumentParser(description="test")
parser.add_argument(
    "--disable_fabric",
    action="store_true",
    default=False, help="Disable fabric and use USD I/O operations.",) 
parser.add_argument(
    "--num_envs", type=int, default=1, help="Number of environments to simulate."
)
parser.add_argument("--task", type=str, default=None, help="Name of the task.")

AppLauncher.add_app_launcher_args(parser)
args_cli = parser.parse_args()

app_launcher = AppLauncher(args_cli)
simulation_app = app_launcher.app

########################################

import torch
import gymnasium as gym
from omni.isaac.lab_tasks.utils import parse_env_cfg

env_cfg = parse_env_cfg(
    args_cli.task,
    device= args_cli.device,
    num_envs=args_cli.num_envs,
    use_fabric=not args_cli.disable_fabric,
)

# adjust camera resolution and pose
env_cfg.viewer.resolution = (640, 480)
env_cfg.viewer.eye = (1.0, 1.0, 1.0)
env_cfg.viewer.lookat = (0.0, 0.0, 0.0)
# create isaac-env instance
# set render mode to rgb_array to obtain images on render calls
env = gym.make("Isaac-Lift-Cube-Franka-v0", cfg=env_cfg, render_mode="rgb_array")
# wrap for video recording
video_kwargs = {
    "video_folder": "videos/train",
    "step_trigger": lambda step: step % 1500 == 0,
    "video_length": 200,
}
env = gym.wrappers.RecordVideo(env, **video_kwargs)

env.reset()
for i in range(200):
    action = env.action_space.sample()
    action = torch.tensor(action).float()
    obs, reward, done, trunc, info = env.step(action)

env.close()
```

Run the script above with `python test2.py --headless --enable_cameras --task Isaac-Lift-Cube-Franka-v0`

Error:
```
Moviepy - Building video /home/arhan/projects/clean-up-the-kitchen/videos/train/rl-video-step-0.mp4.
Moviepy - Writing video /home/arhan/projects/clean-up-the-kitchen/videos/train/rl-video-step-0.mp4

Traceback (most recent call last):
  File "/home/arhan/projects/clean-up-the-kitchen/test2.py", line 53, in <module>
    obs, reward, done, trunc, info = env.step(action)
  File "/home/arhan/micromamba/envs/isaaclab-pip/lib/python3.10/site-packages/gymnasium/wrappers/record_video.py", line 187, in step
    self.close_video_recorder()
  File "/home/arhan/micromamba/envs/isaaclab-pip/lib/python3.10/site-packages/gymnasium/wrappers/record_video.py", line 204, in close_video_recorder
    self.video_recorder.close()
  File "/home/arhan/micromamba/envs/isaaclab-pip/lib/python3.10/site-packages/gymnasium/wrappers/monitoring/video_recorder.py", line 157, in close
    clip.write_videofile(self.path, logger=moviepy_logger)
  File "/home/arhan/micromamba/envs/isaaclab-pip/lib/python3.10/site-packages/decorator.py", line 232, in fun
    return caller(func, *(extras + args), **kw)
  File "/home/arhan/micromamba/envs/isaaclab-pip/lib/python3.10/site-packages/moviepy/decorators.py", line 54, in requires_duration
    return f(clip, *a, **k)
  File "/home/arhan/micromamba/envs/isaaclab-pip/lib/python3.10/site-packages/decorator.py", line 232, in fun
    return caller(func, *(extras + args), **kw)
  File "/home/arhan/micromamba/envs/isaaclab-pip/lib/python3.10/site-packages/moviepy/decorators.py", line 135, in use_clip_fps_by_default
    return f(clip, *new_a, **new_kw)
  File "/home/arhan/micromamba/envs/isaaclab-pip/lib/python3.10/site-packages/decorator.py", line 232, in fun
    return caller(func, *(extras + args), **kw)
  File "/home/arhan/micromamba/envs/isaaclab-pip/lib/python3.10/site-packages/moviepy/decorators.py", line 22, in convert_masks_to_RGB
    return f(clip, *a, **k)
  File "/home/arhan/micromamba/envs/isaaclab-pip/lib/python3.10/site-packages/moviepy/video/VideoClip.py", line 300, in write_videofile
    ffmpeg_write_video(self, filename, fps, codec,
  File "/home/arhan/micromamba/envs/isaaclab-pip/lib/python3.10/site-packages/moviepy/video/io/ffmpeg_writer.py", line 213, in ffmpeg_write_video
    with FFMPEG_VideoWriter(filename, clip.size, fps, codec = codec,
  File "/home/arhan/micromamba/envs/isaaclab-pip/lib/python3.10/site-packages/moviepy/video/io/ffmpeg_writer.py", line 88, in __init__
    '-r', '%.02f' % fps,
TypeError: must be real number, not NoneType
```

### System Info

- Commit: 2af9018dc4ca623dff7d5033d84c5d231ce7030b
- Isaac Sim Version: 4.1.0
- OS: Ubuntu 22.04.2
- GPU: NVIDIA GeForce RTX 3090 Ti
- CUDA: 12.4
- GPU Driver: 550.54.14

### Checklist

- [X ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

arhanjain, 2024-09-01:
This seems to be a dependency issue with MoviePy. Not directly relevant to isaaclab!

closed 2024-09-01 by arhanjain

aravindev, 2024-12-12:
I had the same issue and here's some info that could be useful for future travellers:
The vanilla conda installation I was using had an older version of `moviepy`:
```
➜ pip list | grep movie
moviepy                    1.0.3
```

Upgrading the package to `2.1.1` fixed the issue.