isaac-sim/IsaacLab #878: [Bug Report] 'omni.kit.window.title' not found when running in --headless mode with pip installation

EricVoll, 2024-08-26:
### Describe the bug

I installed IsaacSim 4.1.0 with IsaacLab (latest main) using the pip installation following the documentation: https://isaac-sim.github.io/IsaacLab/source/setup/installation/pip_installation.html#
The "validate installation" steps all work as expected.

When running any script with the AppLauncher in headless mode, the following happens:

```python
from omni.isaac.lab.app import AppLauncher
AppLauncher(
        launcher_args={
            "headless": True,
        }
)

> [...]
[2.199s] [ext: isaaclab.python.headless-1.1.0] startup
[2.200s] Simulation App Starting
[2.323s] app ready
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/er/ws/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/app_launcher.py", line 113, in __init__
    self._create_app()
  File "/home/er/ws/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/app_launcher.py", line 576, in _create_app
    self._app = SimulationApp(self._sim_app_config, experience=self._sim_experience_file)
  File "/home/er/ws/IsaacLab/_isaac_sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py", line 250, in __init__
    from omni.kit.window.title import get_main_window_title
ModuleNotFoundError: No module named 'omni.kit.window.title'
```

### Steps to reproduce

1. Install IsaacSim via pip
2. Run the script above

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/er/ws/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/app_launcher.py", line 113, in __init__
    self._create_app()
  File "/home/er/ws/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/app_launcher.py", line 576, in _create_app
    self._app = SimulationApp(self._sim_app_config, experience=self._sim_experience_file)
  File "/home/er/ws/IsaacLab/_isaac_sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py", line 250, in __init__
    from omni.kit.window.title import get_main_window_title
ModuleNotFoundError: No module named 'omni.kit.window.title'
```

-->

### System Info

Describe the characteristic of your environment:

- Commit: [df184e1](https://github.com/isaac-sim/IsaacLab/commit/df184e1c076f0a9f6fc09d349cd1d7fa16214752)
- Isaac Sim Version: 4.1.0 (pip)
- OS: Ubuntu 24
- GPU: RTX 4060 Ti
- CUDA: 12.5
- GPU Driver: 555.42.06 

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

EricVoll, 2024-08-26:
Just noticed that I have the old `_isaac_sim` folder still lying around from before installing it via pip (had it installed through the launcher prev.) I'll check that quickly.

MinstrelsyXia, 2024-08-27:
Hello! I appreciate the effort you've made to quickly tackle the problem. 
I've met the same problem. Could you please state clearer what you've done? I've searched for `_isaac_sim` folder but can't find any. 
Isaac Sim Version: 4.0.0
Isaac Lab Installation:  [pip tutorial](https://isaac-sim.github.io/IsaacLab/source/setup/installation/pip_installation.html)

EricVoll, 2024-08-27:
Hi @MinstrelsyXia 
I originally had IsaacLab setup using the binary install path through the omniverse launcher. In that tutorial you create a symlink `_isaac_sim` to the installation path of IsaacSim. I forgot this symlink there, and it somehow messed up my python environment, even though I deleted the old conda environment and recreated it. 

After deleting the symlink (and probably re-running the installation - not sure anymore) everything worked. I guess if you want to be super sure, run `git clean` in the IsaacLab repo and reset to main, or do a fresh clone.

elle-miller, 2024-08-28:
I am encountering the same error when I did a fresh clone today of Isaac Lab and reinstalled with Isaac Sim 4.0.0 binaries, even after deleting and remaking the symlink.

Can be reproduced with:
```
git@github.com:isaac-sim/IsaacLab.git
cd IsaacLab
ln -s /home/$your_path/.local/share/ov/pkg/isaac-sim-4.0.0 _isaac_sim
./isaaclab.sh --conda isaac
conda activate Isaac
./isaaclab.sh --install

# errors with "omni.kit.window.title"
./isaaclab.sh -p source/standalone/workflows/skrl/train.py --task Isaac-Reach-Franka-v0 --headless
```
Is Isaac Sim 4.0.0 stilll compatible with newest Isaac Lab? 

If I follow the Isaac Sim pip installation method, it works. (note I had to manually run `pip install scipy` because the dependency resolver didn't install it)

EricVoll, 2024-08-28:
The problem is fixed for me, but should we re-open the issue if others are still experiencing it? @elle-miller

elle-miller, 2024-08-28:
Yes, let's reopen - issue is resolved for installing via pip but not from 4.0.0 binaries (at least in my setup). Does anyone else experience this?

**System info:**
Commit: [df184e1](https://github.com/isaac-sim/IsaacLab/commit/df184e1c076f0a9f6fc09d349cd1d7fa16214752)
Isaac Sim Version: 4.0.0 (binaries)
OS: Ubuntu 22.04
GPU: RTX 2080
CUDA: 12.2
GPU Driver: 535.183.01

EricVoll, 2024-08-28:
I guess you tried to nuke the whole conda setup, IsaacLab repo etc. and do everything from scratch?

wildweasel, 2024-08-30:
I get the exact same issue when running headless from inside a singularity container w/ 4.1.0.   There should only be one isaac sim installation w/ the container, correct?

System info:
Commit: [7452386](https://github.com/isaac-sim/IsaacLab/commit/74523868ab71ceb8259a71bc9e50422d802ca9ad)
Isaac Sim Version: 4.1.0 (docker/singularity)
OS: Ubuntu 22.04
GPU: A40
CUDA: 11.8
GPU Driver: 520.61.05

kandeng, 2024-08-31:
Yes, I got the same error when running --headless  @elle-miller 

>
> No module named 'omni.kit.window.title'
>

Isaac Sim Version: 4.0.0 (binaries)
OS: Ubuntu 22.04
GPU: RTX 3070 Ti
CUDA: 12.4
GPU Driver: 550.90.07

freakontrol, 2024-09-01:
In my case (pip install method) was not necessary to reinstall IsaacLab or to recreate the conda environment or git clean, I just reinstalled all isaacsim packages one by one with pip.

EricVoll, 2024-09-02:
For different reasons I had to try use the newest main branch commit with the old isaacsim 4.0.0, where this issue arised again. 

After commenting out the lines L253 to L258 in the SimulationApp class it works. 
.../IsaacLab/_isaac_sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py

RandomOakForest, 2025-01-05:
Our apologies we couldn't follow up in a timely manner. Please reopen the issue if you still see this using Isaac Sim 4.2 and Isaac Lab 1.4.0. Thank you.

closed 2025-01-05 by RandomOakForest