isaac-sim/IsaacLab #887: [Question] Best Strategy for Real Robot Codebase Reuse

dHonerkamp, 2024-08-28:
### Question

Hi everyone, I'm very excited to see the framework mature and get easier and easier to use.

I'm trying to port my code to the isaaclab format. At the same time, I want to reuse important parts of the code for the real world robot environments.

But to use even the config objects (e.g. `from omni.isaac.lab.assets.articulation.articulation_cfg importArticulationCfg`), python tries to import `omni.isaac.core`. This fails unless I start the simulator (`app_launcher = AppLauncher(args_cli); simulation_app = app_launcher.app`). But for obvious reasons I do not want to start the simulator when working with the real robot.

Do you have any tips on what the best method is to import or use these parts of the code without starting the simulator? Or any tips on how you structure your real world code without duplicating all configs etc.?

Mayankm96, 2024-08-29:
Unfortunately, the short answer is: it isn't possible :(

As much as we have asked the Omniverse team to allow the lazy loading of imports, I'm not sure if this is something they are actively working on. This is probably because it may require a large amount of rewrites to their "hot-reloading" concept, which is designed for the extensions workflow primarily.

I'll pass it on to the team again to see if we can raise our request again.

Alternately, we can try to make our own lazy-import wrappers for Omniverse. It might add some overhead but could be doable. I think [OmniGibson](https://github.com/StanfordVL/OmniGibson/blob/main/omnigibson/lazy.py) does something like that already.

@jsmith-bdai for viz.

dHonerkamp, 2024-08-30:
I see. Would definitely make things a lot easier if that were possible. But I'll check out the lazy loader or revert back to yaml configs then

Thanks a lot for the swift reply.

placeforyiming, 2024-09-11:
> ### Question
> Hi everyone, I'm very excited to see the framework mature and get easier and easier to use.
>
> I'm trying to port my code to the isaaclab format. At the same time, I want to reuse [...]

What I did is using two threads: one works with Isaac only and the other one works with the pure algorithm. When the real robot comes, I replace the Isaac thread with a script in charge of ros message communicating.

closed 2025-01-05 by RandomOakForest