isaac-sim/IsaacLab #896: [Question] Question title How to get the joint 6-D force/torque data in isaaclab?

3505473356, 2024-08-28:
### Question
In Isaac Sim, we can get the joint force using `Articulation.get_measured_joint_forces()`, but how to get the joint 6-D force/torque data in isaaclab, there is no attribute in articulation.data

Mayankm96, 2024-08-29:
You can get these quantities from the PhysX view directly

> https://isaac-sim.github.io/IsaacLab/source/api/lab/omni.isaac.lab.assets.html#omni.isaac.lab.assets.Articulation.root_physx_view

ergo-zyh, 2024-10-09:
> You can get these quantities from the PhysX view directly
> 
> > https://isaac-sim.github.io/IsaacLab/source/api/lab/omni.isaac.lab.assets.html#omni.isaac.lab.assets.Articulation.root_physx_view

I can't find some useful in this page. Can you point out the exact solution?

BRoln7, 2024-11-10:
I met the same problem too

bhe1004, 2024-12-26:
I met the same problem

RandomOakForest, 2025-01-15:
Thank you for following up on this. Having an Isaac Lab wrapper to access methods on the PhysX extension like this is in our to-do list. We have tagged this issue as an enhancement for the team to follow up.

AntoineRichard, 2026-08-27:
Thanks @calmdw for asking — the Isaac Lab wrapper you wanted now exists.

Isaac Lab ships a first-class `JointWrenchSensor` (`isaaclab.sensors.joint_wrench`) that exposes 6-D joint reaction forces and torques through the normal sensor/config workflow, so no reaching into `root_physx_view.get_measured_joint_forces()`. It landed for Newton in https://github.com/isaac-sim/IsaacLab/pull/5412, PhysX in https://github.com/isaac-sim/IsaacLab/pull/5458, and OvPhysX in https://github.com/isaac-sim/IsaacLab/pull/5706. Please reopen if a quantity you need still isn't exposed.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard