isaac-sim/IsaacLab #899: [Question] Unique id for same objects in different environments

Entongsu, 2024-08-29:
Hi,

I have found the segmentation mask of the same object in a different environment has a different color representation,which makes it hard to segment different objects in different environments. Because the same object has different IDs in different environments. Could you provide some hints on how to fix this problem? thank you,

Mayankm96, 2024-08-29:
Are you setting the parameter for semantic tags?

> https://isaac-sim.github.io/IsaacLab/source/api/lab/omni.isaac.lab.sim.spawners.html#omni.isaac.lab.sim.spawners.SpawnerCfg.semantic_tags

Entongsu, 2024-09-01:
I have set the semantic tags but it still does not work.

Mayankm96, 2024-09-05:
Can you share a simple repro for us to try?

closed 2025-01-05 by RandomOakForest