isaac-sim/IsaacLab #903: [Bug Report] Bug title Isaac-Open-Drawer-Franka-v0 task is not running

hanlinniu, 2024-08-29:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

When I run  : ./isaaclab.sh -p source/standalone/workflows/rsl_rl/train.py --task Isaac-Open-Drawer-Franka-v0 --headless 

I have the following error: 
 File "/home/xxx/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/scene/interactive_scene.py", line 462, in _add_entities_from_cfg
    target_frame.prim_path = target_frame.prim_path.format(ENV_REGEX_NS=self.env_regex_ns)
AttributeError: 'dict' object has no attribute 'prim_path'. 

The same error also exists for [Isaac-Lift-Cube-Franka-v0] task as well. The other tasks like the locomotion and [Isaac-Reach-Franka-v0] have no problem.

pascal-roth, 2024-08-30:
Hi, 
I just tried to run `./isaaclab.sh -p source/standalone/workflows/rsl_rl/train.py --task Isaac-Open-Drawer-Franka-v0 --headless` on my side, and it works with the latest main commit. Could you please double-check that you are on the latest commit?

Mayankm96, 2024-08-30:
We recommend updating to latest main. This should be addressed in https://github.com/isaac-sim/IsaacLab/pull/847.

closed 2024-08-30 by Mayankm96