isaac-sim/IsaacLab #904: [Bug Report] Major bug report and resolution for random seed setting and non-deterministic behavior during training

hojae-io, 2024-08-30:
### Describe the bug

It has been observed that the training results such as reward curve are not same even if you manually set random seed (for instance seed=42).
Several issue has been submitted:
https://github.com/isaac-sim/IsaacLab/issues/489
https://github.com/isaac-sim/IsaacLab/issues/275

### Steps to reproduce

Try running some code like : IsaacLab/source/standalone/workflows/rsl_rl/train.py
Or check the above issue to reproduce the issue.

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: 2024.4.1
- OS: Ubuntu 22.04
- GPU: Geforce 3090
- CUDA: 11.2
- GPU Driver: 550

### Resolution

Here's my resolution, and now I don't have any non-deterministic / stochastic behavior, and the reward curve "exactly" overlap if I train the same code multiple times.

The problem is coming from "setting the seed after the environment is created"
[image]

You can see the seed is being set at line 118 which is after the env is created at line 90

But now, if you set the seed before(line 92 - 102) the env is created (like the image below), all the behavior becomes deterministic.
I don't know why "the time at which you set the seed" is important and it "could" cause non-deterministic behavior. 

It would be nice if you could provide explanation and reflect this bug into your next pull request.

[image]

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

pascal-roth, 2024-08-30:
Hi, 
Thanks for looking into this issue. I agree; it is strange that the moment when the seed is set makes a difference when the seed is set. Could you make a PR with the necessary changes and, ideally, a test to prove that the fix is effective?

Mayankm96, 2024-08-30:
Thanks @hojae-io for digging deeper into this issue. This is a really great find and I may have an explanation.

From what I suspect, at initialization, we have some random events happening. For instance, the terrain generation (a lot of random sampling there), initialization of PhysX solver and internal buffers (not sure). It would make sense that setting the seed before ensures that the randomness from these sources are limited when the seed is fixed.

Definitely makes sense to fix this issue. Instead of modifying the play/train scripts, we should set the seed as the first operation when the [class is constructed](https://github.com/isaac-sim/IsaacLab/blob/main/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_env.py#L77). We can add `seed` into the configuration of the environment.

amrmousa144, 2024-09-03:
I'm having the same issue and looking forward to a contribution to fix it ASAP!

Mayankm96, 2024-09-06:
Based on the suggestion here, I have made the fixes in #940. At least from the unit test, where I do some fixed number of env steps, the obtained obs and rewards are the same. Though this checks this within the same process.

I ran the training for anymal locomotion and the results look way worse right now.

```bash
./isaaclab.sh -p source/standalone/workflows/rsl_rl/train.py --task Isaac-Velocity-Rough-Anymal-C-v0 --headless --run_name seed_fix
``` 

The runs with `seed_fix` appended are the ones from this branch, and the others are on the main.

[image]

closed 2024-09-10 by Mayankm96

LouKordos, 2025-05-20:
@Mayankm96 I have the same issue but only when running with flat terrain. I confirmed that the rough terrain generated is identical across each run by seeding it but the rewards and states are still slightly different even with the adjustments you showed in your explanation. For testing, at the top of the env file, I use a hardcoded seed and set np.random, random, and torch to this specified seed, as well as the terrain generator seed and the overall env seed in post init. With a flat ground plane, training is perfectly deterministic, but when changing to:

```py
from copy import deepcopy
seeded_rough_cfg = deepcopy(ROUGH_TERRAINS_CFG)
seeded_rough_cfg.seed = HARDCODED_SEED
seeded_rough_cfg.use_cache = True

terrain = TerrainImporterCfg(
        prim_path="/World/ground",
        terrain_type="generator",
        terrain_generator=seeded_rough_cfg,
        max_init_terrain_level=5,
        collision_group=-1,
        physics_material=sim_utils.RigidBodyMaterialCfg(
            friction_combine_mode="multiply",
            restitution_combine_mode="multiply",
            static_friction=1.0,
            dynamic_friction=1.0,
        ),
        visual_material=sim_utils.MdlFileCfg(
            mdl_path="{NVIDIA_NUCLEUS_DIR}/Materials/Base/Architecture/Shingles_01.mdl",
            project_uvw=True,
        ),
        debug_vis=False,
    )
```

two consecutive runs differ. I was wondering if it's maybe due to slight differences in contact forces / contact calculations in physx in general? Are those supposed to be perfectly deterministic even with uneven terrain?

I have this in my env post init, but to no avail.
```
self.sim.physx.enable_enhanced_determinism = True
self.sim.random_seed = HARDCODED_SEED
self.seed = HARDCODED_SEED
```

yufengsjtu, 2025-05-26:
> [@Mayankm96](https://github.com/Mayankm96) I have the same issue but only when running with flat terrain. I confirmed that the rough terrain generated is identical across each run by seeding it but the rewards and states are still slightly different even with the adjustments you showed in your explanation. For testing, at the top of the env file, I use a hardcoded seed and set np.random, random, and torch to this specified seed, as well as the terrain generator seed and the overall env seed in post init. With a flat ground plane, training is perfectly deterministic, but when changing to:
> 
> from copy import deepcopy
> seeded_rough_cfg = deepcopy(ROUGH_TERRAINS_CFG)
> seeded_rough_cfg.seed = HARDCODED_SEED
> seeded_rough_cfg.use_cache = True
> 
> terrain = TerrainImporterCfg(
>         prim_path="/World/ground",
>         terrain_type="generator",
>         terrain_generator=seeded_rough_cfg,
>         max_init_terrain_level=5,
>         collision_group=-1,
>         physics_material=sim_utils.RigidBodyMaterialCfg(
>             friction_combine_mode="multiply",
>             restitution_combine_mode="multiply",
>             static_friction=1.0,
>             dynamic_friction=1.0,
>         ),
>         visual_material=sim_utils.MdlFileCfg(
>             mdl_path="{NVIDIA_NUCLEUS_DIR}/Materials/Base/Architecture/Shingles_01.mdl",
>             project_uvw=True,
>         ),
>         debug_vis=False,
>     )
> 
> two consecutive runs differ. I was wondering if it's maybe due to slight differences in contact forces / contact calculations in physx in general? Are those supposed to be perfectly deterministic even with uneven terrain?
> 
> I have this in my env post init, but to no avail.
> 
> ```
> self.sim.physx.enable_enhanced_determinism = True
> self.sim.random_seed = HARDCODED_SEED
> self.seed = HARDCODED_SEED
> ```
> 
> For reference, this is my current code: [train.py](https://github.com/LouKordos/constraints-as-terminations/blob/main/scripts/clean_rl/train.py) and [env config](https://github.com/LouKordos/constraints-as-terminations/blob/main/exts/cat_envs/cat_envs/tasks/locomotion/velocity/config/solo12/cat_go2_rough_terrain_env_cfg.py)

Hi, i also encountered this problem, have you solved it? 
I tried to train my robot on my customized terrain with the env seed fixed, but the rewards and observations are slightly different at first few iterations, and as the training progresses, the curve gradually becomes more inconsistent.

PR #940 (2024-09-10): Adds setting of environment seed at initialization
# Description

Various randomization and procedural generation operations happen at initialization. However, as noted in one of the issues, the seed setting happens after all these operations are performed. This means that the creation of the environment is not completely deterministic. This MR resolves this issue by adding a `seed` configuration to the environment.

Fixes #904

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Screenshots

The before and after results over **three runs** with the default seed (seed:  [... cut]
commit ac71354c (2024-09-10): Adds setting of environment seed at initialization (#940)

# Description

Various randomization and procedural generation operations happen at
initialization. However, as noted in one of the issues, the seed setting
happens after all these operations are performed. This means that the
creation of the environment is not completely deterministic. This MR
resolves this issue by adding a `seed` configuration to the environment.

Fixes #904

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Screenshots

The before and after results over **three runs** with the default seed [... cut]