isaac-sim/IsaacLab #912: [Question] Right way to query self-collisions

yeshgodseagility, 2024-08-31:
### Question

I'm trying to detect certain self-collisions for my robot bodies. Based on the documentation for [ContactSensor](https://isaac-sim.github.io/IsaacLab/source/api/lab/omni.isaac.lab.sensors.html#omni.isaac.lab.sensors.ContactSensor) it seems like it should be possible to create a contact sensor for detecting self collisions between two robot bodies by using
```
ContactSensorCfg(
    prim_path="{ENV_REGEX_NS}/Robot/body_A",
    filter_prim_paths_expr=["{ENV_REGEX_NS}/Robot/body_B"]
)
```

When I try doing this I'm also getting contact reported for when a body is in contact with my ground plane. Is this expected?

Mayankm96, 2024-08-31:
Are you using the right attribute to get the data?

> https://isaac-sim.github.io/IsaacLab/source/api/lab/omni.isaac.lab.sensors.html#omni.isaac.lab.sensors.ContactSensorData.force_matrix_w

yeshgodseagility, 2024-09-01:
@Mayankm96 Ah looks like I didn't read closely enough, was using `net_forces_w` instead of `force_matrix_w`

closed 2024-09-01 by yeshgodseagility