isaac-sim/IsaacLab #914: [Bug Report] Commit `8ed87a00074619d0405c7c3a205df8c0c23662fc` breaks 'Interacting with a rigid object' tutorial: 'AttributeError: 'RigidBodyView' object has no attribute 'get_accelerations''

Aemperior, 2024-08-31:
### Describe the bug

When running the tutorial [interacting with a rigid object](https://isaac-sim.github.io/IsaacLab/source/tutorials/01_assets/run_rigid_object.html) you will get the following error message: 
`AttributeError: 'RigidBodyView' object has no attribute 'get_accelerations'`
in the file: 
`source/extensions/omni.isaac.lab/omni/isaac/lab/assets/rigid_object/rigid_object_data.py`

This is due to the call `self._body_acc_w.data = self._root_physx_view.get_accelerations().unsqueeze(1)` which causes the error. 

This error is introduced in [commit 8ed87a00074619d0405c7c3a205df8c0c23662fc](https://github.com/isaac-sim/IsaacLab/commit/8ed87a00074619d0405c7c3a205df8c0c23662fc). 

The previous commit before it, [commit c3e4b0d0f4f61bb468a0e7f712867eaf1b65e92f](https://github.com/isaac-sim/IsaacLab/commit/c3e4b0d0f4f61bb468a0e7f712867eaf1b65e92f) does not encounter this error when running the tutorial. 

### Steps to reproduce

Provided your python.sh is located at **/isaac-sim/python.sh**, in the **IsaacLab** directory, run: 
`/isaac-sim/python.sh source/standalone/tutorials/01_assets/run_rigid_object.py --livestream=2`

### System Info

- Isaac Sim Version: 4.0.0-rc.21+4.0.13872.3e3cb0c9.gl

### Checklist

- [x ] I have checked that there is no similar issue in the repo (**required**)
- [x ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

- [ ] Tutorial can be run without error in the newest version of main?

pascal-roth, 2024-09-01:
Hi, 

Thanks for pointing out this issue. `get_accelerations` is only available with Isaac Sim Version 4.1. We recommend always using the latest version to avoid running into these issue

closed 2024-09-01 by pascal-roth

Aemperior, 2024-09-02:
Ok thank you. My driver version (Driver Version: 535.183.01   CUDA Version: 12.2) does not fit the recommended one for isaac-sim 4.1 and I don't think I can upgrade since others are also using this remote workstation. 

If I download the newest isaac-sim container [(step 6: docker pull nvcr.io/nvidia/isaac-sim:4.1.0) ](https://docs.omniverse.nvidia.com/isaacsim/latest/installation/install_container.html#isaac-sim-app-install-container/) will this somehow break my old version? Will it be safe for me to just switch back and forth between 4.0.0 and 4.1.0 in my launch script? 

```
docker run --name isaac-sim --entrypoint bash -it --runtime=nvidia --gpus all -e "ACCEPT_EULA=Y" --rm --network=host \
    -e "PRIVACY_CONSENT=Y" \
    -v ~/docker/isaac-sim/cache/kit:/isaac-sim/kit/cache:rw \
    -v ~/docker/isaac-sim/cache/ov:/root/.cache/ov:rw \
    -v ~/docker/isaac-sim/cache/pip:/root/.cache/pip:rw \
    -v ~/docker/isaac-sim/cache/glcache:/root/.cache/nvidia/GLCache:rw \
    -v ~/docker/isaac-sim/cache/computecache:/root/.nv/ComputeCache:rw \
    -v ~/docker/isaac-sim/logs:/root/.nvidia-omniverse/logs:rw \
    -v ~/docker/isaac-sim/data:/root/.local/share/ov/data:rw \
    -v ~/docker/isaac-sim/documents:/root/Documents:rw \
    nvcr.io/nvidia/isaac-sim:4.1.0
```

Aemperior, 2024-09-03:
I did `docker pull nvcr.io/nvidia/isaac-sim:4.1.0` and both 4.0.0 and 4.1.0 work with the launch script below so far. It also seems to work with Driver Version: 535.183.01 just fine so far. 

```
docker run --name isaac-sim --entrypoint bash -it --runtime=nvidia --gpus all -e "ACCEPT_EULA=Y" --rm --network=host \
    -e "PRIVACY_CONSENT=Y" \
    -v ~/docker/isaac-sim/cache/kit:/isaac-sim/kit/cache:rw \
    -v ~/docker/isaac-sim/cache/ov:/root/.cache/ov:rw \
    -v ~/docker/isaac-sim/cache/pip:/root/.cache/pip:rw \
    -v ~/docker/isaac-sim/cache/glcache:/root/.cache/nvidia/GLCache:rw \
    -v ~/docker/isaac-sim/cache/computecache:/root/.nv/ComputeCache:rw \
    -v ~/docker/isaac-sim/logs:/root/.nvidia-omniverse/logs:rw \
    -v ~/docker/isaac-sim/data:/root/.local/share/ov/data:rw \
    -v ~/docker/isaac-sim/documents:/root/Documents:rw \
    nvcr.io/nvidia/isaac-sim:4.1.0
```