isaac-sim/IsaacLab #915: [Question] Training performance comparsion between IsaacLab and IsaacGym

hojae-io, 2024-09-01:
Hi all,

I wondering if anyone closely compare the performance of the policy trained in IsaacLab and IsaacGym.

I have a decent locomotion policy working well in IsaacGym and try to move the codebase to IsaacLab.

I observe that the training performance in IsaacLab is worse than in IsaacGym.

Although you might claim that I may have introduced a bug while transferring the code, I checked multiple times all the configs and code order etc.. (i.e. PPO settings, reward weights, PhysX settings etc..)

I agree that there might still be some bugs that I didn't identify, but for the sake of questions, let's assume the code logic are exactly the same. 

Have anyone observed that their policies in IsaacLab is worse than IsaacGym? 

I have some training plots regarding this matter: purple is IsaacGym, and green is IsaacLab.

[image]

[image]

It seems in IsaacLab the mean reward is lower and value function loss is much higher/noisy than in IsaacGym.

Based on my visual inspection, in IsaacLab the robot tends to reach its torque limit quite frequently and terminate early, resulting in unstable policy and noisy value function.

Does anyone observe similar issue and provide some possible explanation and resolution? (except the bug I might accidently introduced)

One of my hypothesis is the IsaacLab and IsaacGym use different PhysX version and their dynamics (or contact solver) has changed.
I'm wondering if there are any hidden bugs or mistakes in IsaacLab side, or it is just finetuning issues.

Thank you!

Mayankm96, 2024-09-05:
When I last compared it for ANYmal locomotion, things looked pretty similar. However, this was some time ago, and we don't have regular integration tests for this at the moment.

The value function jumping is the most concerning to me. Are you using one of the environments that we provide, or is this your custom environment?

hojae-io, 2024-09-05:
Hi, I'm using custom environment for humanoid. (Not the unitree robots)

But the custom environment settings are same for both in IsaacGym and IsaacLab.

So, I'm concerned.

Is there any major modification that I need to be aware of?
So basically, a few questions:
1) Is the physics simulation from PhysX 4.0 (IsaacGym) and PhysX 5.0 (IsaacLab) same?
2) Other people have questions, for instance, root_lin_vel_w is not the base frame's velocity rather than CoM's velocity. 
3) Do PPO settings or reward weights have to be changed as we change the simulator?

A few evidence show that the training performance in IsaacLab is worse than in IsaacGym. What could possibly make this difference?

I don't know if it comes from just fine-tuning issues based on simulator change, or if IsaacLab has some inherent bug or some fundamental changes that need to be dealt with.

Thank you.

hojae-io, 2024-09-19:
This turns out to be the 'push_robot' problem:

```    
push_robot = EventTerm(
        func=mdp.push_by_setting_velocity,
        mode="interval",
        interval_range_s=(2.5, 2.5),
        params={"velocity_range": {"x": (-0.5, 0.5), "y": (-0.5, 0.5)}},
    )
```

If I change this code into (by setting interval_range to be a range)

```    
push_robot = EventTerm(
        func=mdp.push_by_setting_velocity,
        mode="interval",
        interval_range_s=(2.0, 3.0),
        params={"velocity_range": {"x": (-0.5, 0.5), "y": (-0.5, 0.5)}},
    )
```

then the value function loss is much less.

Still don't know why this happens or this is crucial ..

hojae-io, 2024-09-23:
Hello, 

I have some serious(?) concern about the push_by_setting_velocity function or possible any 'interval' event if they are functioning correctly.

As I showed concerns in this post that the training performance in IsaacLab seems to be worse than IsaacGym, I found the following interesting phenomenon. 

It seems that the function that push robot in IsaacLab seems to have more degrading effect for training performance than IsaacGym.

Mayankm96, 2024-09-23:
@hojae-io 

Before we investigate, could you specify the Isaac Lab commit and Isaac Sim version you are using again? It is easier for us to debug if it is the latest release of both (Isaac Lab 1.2 and Isaac Sim 4.2).

hojae-io, 2024-09-23:
Hi, the versions I'm using are:
Isaac-Sim 4.1 / IsaacLab 1.1.0

But I believe the code stack for `push_by_setting_velocity` or 'interval event' hasn't been changed.

Thanks!

RandomOakForest, 2025-01-06:
Thank you for following up. The team fixed `push_by_setting_velocity` recently. In general, there shouldn't be drastic differences between Isaac Gym and Isaac Lab.

closed 2025-01-06 by RandomOakForest