isaac-sim/IsaacLab #916: [Bug Report] Cannot open isaac lab's demo

HzfFrank, 2024-09-01:
### Describe the bug

After I updated my cuda, I reinstalled isaac sim and lab, however, the demo's in isaac lab could not be opened any longer.

### Steps to reproduce

When I ran the command `python source\standalone\demos\quadcopter.py` in IsaacLab folder. It shows

```
[8.613s] [ext: omni.sensors.nv.common-1.0.1] startup
[8.717s] [ext: omni.replicator.core-1.11.8] startup
2024-09-01 03:13:29 [8,652ms] [Warning] [omni.replicator.core.scripts.annotators] Annotator PostProcessDispatch is already registered, overwriting annotator template
[9.028s] [ext: omni.kit.menu.create-1.0.12] startup
[9.036s] [ext: omni.sensors.nv.materials-1.0.0] startup
[9.068s] [ext: omni.isaac.range_sensor-3.0.1] startup
[9.139s] [ext: omni.isaac.core_nodes-1.15.4] startup
[9.180s] [ext: omni.sensors.nv.wpm-1.0.0] startup
[9.192s] [ext: omni.kit.actions.window-1.1.1] startup
[9.236s] [ext: omni.kit.selection-0.1.4] startup
[9.244s] [ext: omni.sensors.nv.radar-1.0.1] startup
[9.273s] [ext: omni.sensors.nv.lidar-1.0.1] startup
2024-09-01 03:13:30 [9,172ms] [Error] [carb] [Plugin: omni.sensors.nv.lidar.ext.plugin] Dependency: [omni::sensors::lidar::IGenericModelOutputIOFactory v0.1] failed to be resolved.
[9.300s] [ext: omni.kit.menu.file-1.1.9] startup
[9.316s] [ext: omni.kit.menu.edit-1.1.19] startup
[9.334s] [ext: omni.isaac.sensor-11.3.0] startup
2024-09-01 03:13:30 [9,264ms] [Error] [carb.scripting-python.plugin] UnicodeDecodeError: 'gbk' codec can't decode byte 0xa0 in position 1060: illegal multibyte sequence

At:
  D:\application\Anaconda\envs\isaaclab\lib\json\__init__.py(293): load
  c:/users/frank/isaaclab/_isaac_sim/exts/omni.isaac.sensor/omni/isaac/sensor/scripts/menu.py(209): __init__
  c:/users/frank/isaaclab/_isaac_sim/exts/omni.isaac.sensor/omni/isaac/sensor/scripts/extension.py(32): on_startup
  C:\Users\frank\IsaacLab\_isaac_sim/kit/kernel/py\omni\ext\_impl\_internal.py(164): _startup_ext
  C:\Users\frank\IsaacLab\_isaac_sim/kit/kernel/py\omni\ext\_impl\_internal.py(224): startup
  C:\Users\frank\IsaacLab\_isaac_sim/kit/kernel/py\omni\ext\_impl\_internal.py(328): startup_extension
  PythonExtension.cpp::startup()(2): <module>
  C:\Users\frank\IsaacLab\_isaac_sim/exts/omni.isaac.kit\omni\isaac\kit\simulation_app.py(350): _start_app
  C:\Users\frank\IsaacLab\_isaac_sim/exts/omni.isaac.kit\omni\isaac\kit\simulation_app.py(197): __init__
  C:\Users\frank\IsaacLab\source\extensions\omni.isaac.lab\omni\isaac\lab\app\app_launcher.py(576): _create_app
  C:\Users\frank\IsaacLab\source\extensions\omni.isaac.lab\omni\isaac\lab\app\app_launcher.py(113): __init__
  C:\Users\frank\IsaacLab\source\standalone\demos\quadcopter.py(26): <module>

2024-09-01 03:13:30 [9,266ms] [Error] [omni.ext.plugin] [ext: omni.isaac.sensor-11.3.0] Failed to startup python extension.
[9.434s] [ext: omni.kit.hotkeys.window-1.4.5] startup
[9.471s] [ext: omni.kit.property.isaac-0.2.3] startup
```
Part of the wrong `UnicodeDecodeError: 'gbk' codec can't decode byte 0xa0 in position 1060: illegal multibyte sequence` seems to be a comman problem as [here](https://stackoverflow.com/questions/3218014/unicodeencodeerror-gbk-codec-cant-encode-character-illegal-multibyte-sequen) mentions. But I don't know why it has connection between isaac lab and utf-8.

After long time waiting, it finally shows
```
2024-09-01 03:23:28 [607,960ms] [Warning] [omni.usd] Warning: in _ReportErrors at line 2890 of W:\ac88d7d902b57417\USD\pxr\usd\usd\stage.cpp -- In </World/defaultGroundPlane>: Could not open asset @http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.1/Isaac/Environments/Grid/default_environment.usd@ for reference introduced by @anon:00000185EE6FC780:World0.usd@</World/defaultGroundPlane>. (recomposing stage on stage @anon:00000185EE6FC780:World0.usd@ <00000185EAE7E0B0>)

Traceback (most recent call last):
  File "C:\Users\frank\IsaacLab\source\standalone\demos\quadcopter.py", line 111, in <module>
    main()
  File "C:\Users\frank\IsaacLab\source\standalone\demos\quadcopter.py", line 52, in main
    cfg.func("/World/defaultGroundPlane", cfg)
  File "C:\Users\frank\IsaacLab\source\extensions\omni.isaac.lab\omni\isaac\lab\sim\spawners\from_files\from_files.py", line 145, in spawn_ground_plane
    collision_prim_path = prim_utils.get_prim_path(
  File "c:/users/frank/isaaclab/_isaac_sim/exts/omni.isaac.core/omni/isaac/core/utils/prims.py", line 631, in get_prim_path
    return prim.GetPath()
AttributeError: 'NoneType' object has no attribute 'GetPath'
2024-09-01 03:23:28 [608,017ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-09-01 03:23:28 [608,019ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-09-01 03:23:28 [608,022ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-09-01 03:23:29 [608,023ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-09-01 03:23:29 [608,025ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-09-01 03:23:29 [608,025ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-09-01 03:23:29 [608,026ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-09-01 03:23:29 [608,026ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-09-01 03:23:29 [608,027ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
[... 335 lines cut ...]
  File "c:/users/frank/isaaclab/_isaac_sim/exts/omni.isaac.core/omni/isaac/core/utils/prims.py", line 631, in get_prim_path
    return prim.GetPath()
AttributeError: 'NoneType' object has no attribute 'GetPath'
2024-09-01 03:23:28 [608,017ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-09-01 03:23:28 [608,019ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-09-01 03:23:28 [608,022ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-09-01 03:23:29 [608,023ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-09-01 03:23:29 [608,025ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-09-01 03:23:29 [608,025ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-09-01 03:23:29 [608,026ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-09-01 03:23:29 [608,026ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-09-01 03:23:29 [608,027ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-09-01 03:23:29 [608,027ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-09-01 03:23:29 [608,029ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-09-01 03:23:29 [608,029ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Core' for removal
2024-09-01 03:23:29 [608,183ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
```

### System Info

- Isaac Sim Version: 4.0.0 and 4.1.0
- OS: Windows 10
- GPU: RTX 3050 Laptop
- GPU Driver: 560.81
- Cuda: 12.6

HzfFrank, 2024-09-01:
Does isaac lab support cuda 12.6?

change-every, 2024-09-05:
I met the same question. 
And when I running the code:
> isaacsim omni.isaac.sim.python.kit

An error occurred: 
> [0.035s] Failed to solve some dependencies locally, syncing with extension registry...
[0.038s] [ext: omni.kit.async_engine-0.0.0] startup
[0.040s] [ext: omni.client-1.1.0] startup
[0.074s] [ext: omni.kit.registry.nucleus-0.0.0] startup
2024-09-05 12:33:14 [4,163ms] [Error] [omni.ext.plugin] Failed to resolve extension dependencies. Failure hints:
    [omni.isaac.sim.python-4.1.0] dependency: 'omni.isaac.cortex' = { version='^' } can't be satisfied. Available versions:
     (none found)
 Synced registries:
     - kit/default         : found 399 packages 
     - kit/sdk             : found 309 packages 
     - kit/community       : found 362 packages 
2024-09-05 12:33:14 [4,163ms] [Error] [omni.kit.app.plugin] Exiting app because of dependency solver failure...

HzfFrank, 2024-09-05:
After changed my VPN, I can get the usd file from nucleus correctly, it has been solved

closed 2024-09-05 by HzfFrank

Cesiro121, 2024-11-27:
Hello, I have the following problem, can you help me?

(isaaclab) D:\Nvidia_Omniverse\isaaclab3\IsaacLab>isaaclab.bat -p source\standalone\tutorials\00_sim\create_empty.py
[INFO] Using python from: D:\Nvidia_Omniverse\Anaconda\envs\isaaclab\python.exe
[WARN][AppLauncher]: There are no arguments attached to the ArgumentParser object. If you have your own arguments, please load your own arguments before calling the AppLauncher.add_app_launcher_args method. This allows the method to check the validity of the arguments and perform checks for argument names.
[INFO][AppLauncher]: Loading experience file: D:\Nvidia_Omniverse\isaaclab3\IsaacLab\source\apps\isaaclab.python.kit
[Warning] [omni.isaac.kit.simulation_app] Modules: ['omni.kit_app'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Loading user config located at: 'd:/nvidia_omniverse/anaconda/envs/isaaclab/lib/site-packages/omni/data/Kit/Isaac-Sim/4.2/user.config.json'
[Info] [carb] Logging to file: d:/nvidia_omniverse/anaconda/envs/isaaclab/lib/site-packages/omni/logs/Kit/Isaac-Sim/4.2/kit_20241127_111551.log
2024-11-27 09:15:51 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
2024-11-27 09:15:53 [1,969ms] [Error] [omni.ext.plugin] Failed to resolve extension dependencies. Failure hints:
[isaaclab.python-1.3.0] dependency: 'omni.physx.bundle' = { version='^' } can't be satisfied. Available versions:
(none found)
Synced registries:

kit/default : found 388 packages
kit/sdk : found 317 packages
kit/community : found 384 packages
2024-11-27 09:15:53 [1,970ms] [Error] [omni.kit.app.plugin] Exiting app because of dependency solver failure...
Traceback (most recent call last):
File "D:\Nvidia_Omniverse\isaaclab3\IsaacLab\source\standalone\tutorials\00_sim\create_empty.py", line 29, in
app_launcher = AppLauncher(args_cli)
File "D:\Nvidia_Omniverse\isaaclab3\IsaacLab\source\extensions\omni.isaac.lab\omni\isaac\lab\app\app_launcher.py", line 113, in init
self._create_app()
File "D:\Nvidia_Omniverse\isaaclab3\IsaacLab\source\extensions\omni.isaac.lab\omni\isaac\lab\app\app_launcher.py", line 607, in _create_app
self._app = SimulationApp(self._sim_app_config, experience=self._sim_experience_file)
File "D:\Nvidia_Omniverse\Anaconda\envs\isaaclab\Lib\site-packages\isaacsim\exts\omni.isaac.kit\omni\isaac\kit\simulation_app.py", line 214, in init
from .utils import create_new_stage, open_stage, set_livesync_stage
File "D:\Nvidia_Omniverse\Anaconda\envs\isaaclab\Lib\site-packages\isaacsim\exts\omni.isaac.kit\omni\isaac\kit\utils.py", line 17, in
from omni.kit.usd import layers
ModuleNotFoundError: No module named 'omni.kit.usd'
2024-11-27 09:15:53 [1,973ms] [Warning] [carb] Recursive unloadAllPlugins() detected!

(isaaclab) D:\Nvidia_Omniverse\isaaclab3\IsaacLab>

Thank you!