isaac-sim/IsaacLab #918: [Question] AttributeError: 'Articulation' object has no attribute '_root_physx_view'

nugglet, 2024-09-02:
I am not sure if this is a bug or not but I am using IsaacLab with python. I am trying to use the Articulation object API as described in the documentation to get articulation data etc

The code:
```
self.robot = Articulation(self.cfg.robot_cfg)
print(type(self.robot))

```
Throws error:
'''
File "/home/user/Documents/xxx.py", line 233, in _setup_scene
    print(self.robot.num_joints)
  File "/home/user/.local/share/ov/pkg/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/articulation/articulation.py", line 117, in num_joints
    return self.root_physx_view.shared_metatype.dof_count
  File "/home/user/.local/share/ov/pkg/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/articulation/articulation.py", line 146, in root_physx_view
    return self._root_physx_view
AttributeError: 'Articulation' object has no attribute '_root_physx_view'. Did you mean: 'root_physx_view'?
'''

I have encountered a similar problem with the Articulation.data attribute where the terminal throws 'Articulation' object has no attribute '_data' and the error is from code in the Articulation source code files. Is this a bug or am I using the Articulation object wrongly? How do I fix this?

SakethTecholution, 2024-09-02:
This usually happens when we work with custom usd , We could be able to solve this issue by converting the usd from URDF. 
STL -> URDF -> OBJ -> USD

nugglet, 2024-09-03:
I see. I am not using a custom usd here tho. I am importing from `from omni.isaac.lab_assets.unitree import UNITREE_GO2_CFG `
The unitree.py file in the source code shows the the import is from `usd_path=f"{ISAACLAB_NUCLEUS_DIR}/Robots/Unitree/Go2/go2.usd"`

I am using a custom DirectRLEnv environment and I did notice that the Articulation class methods and attributes can be called normally as long as I don't call it in `_setup_scene()`, which is the where the articulation is first instantiated. Is this just due to the sequence of initializing the classes?

nugglet, 2024-09-03:
Sorry accidentally closed the issue with my misclick :")

wzx16, 2024-09-08:
Same issue here, may I know if you have solved this problem?

MiladShafiee, 2024-09-08:
The issue arises because the physics simulation has not yet been initialized. The `_setup_scene()` method is called within the `__init__` method of the parent class for your customized environment, which is `direct_rl_env.py`. This can be seen in the following line from the code: [source](https://github.com/isaac-sim/IsaacLab/blob/af088f593216f7215310cf5e3a13a259d7320a63/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/direct_rl_env.py#L111):

```python
with Timer("[INFO]: Time taken for scene creation", "scene_creation"):
    self.scene = InteractiveScene(self.cfg.scene)
    self._setup_scene()
```

At this point, the scene is created, but the physics engine and simulation data have not yet been initialized. The initialization of the physics and simulation engine happens later in the code at [this line](https://github.com/isaac-sim/IsaacLab/blob/af088f593216f7215310cf5e3a13a259d7320a63/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/direct_rl_env.py#L129):

```python
if builtins.ISAAC_LAUNCHED_FROM_TERMINAL is False:
    print("[INFO]: Starting the simulation. This may take a few seconds. Please wait...")
    with Timer("[INFO]: Time taken for simulation start", "simulation_start"):
        self.sim.reset()
```

Only after this point, where the simulation is reset and initialized, can you safely access simulation-specific properties, such as the number of joints in the robot. For instance, you can check this information by running ```python
print(self.robot.num_joints)``` after the mentioned line.

Therefore, the issue occurs when you attempt to access simulation-related data before the parent class has fully initialized. For example, in the `anymal` environment, you must ensure that the parent class is initialized before interacting with the simulation. This is handled at [this line](https://github.com/isaac-sim/IsaacLab/blob/af088f593216f7215310cf5e3a13a259d7320a63/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/direct/anymal_c/anymal_c_env.py#L127). After this point, the code will run without any issues.

wzx16, 2024-09-08:
@MiladShafiee Thank you, that makes sense.

closed 2024-09-10 by nugglet