isaac-sim/IsaacLab #919: How to simulate Realsenese camera data in Isaaclab && Why Arm is Falling in TeleOperation code of Franka

SakethTecholution, 2024-09-02:
@ViktorM @danzeeeman @animesh-garg @kshivakumar-nvidia @adevalla-nvidia 
Hi IsaacLab Team, 
I have these questions :- 
1. I had read through the sensors tutorials of IsaacLab but as far i know that it is only supporting pinhole camera, But in isaac sim we can add sensors like relasense and get the depth data as well. How to do that with Isaaclab 
2. And Ik controller with even franka arm is falling while i am running the teleop code ..Could you guys assist what could be the issues and why arm is falling after some time ..?
Can you please help me with this guys and it would be great if you provide some sample scripts

SakethTecholution, 2024-09-04:
@animesh-garg 
Do you have any idea regarding this

Mayankm96, 2024-09-05:
1. A realsense camera follows a pinhole camera model. Am I missing something here?
2. You either need to have high PD gains or enable gravity compensation to not have the arm fall down. Please check the settings used here: https://github.com/isaac-sim/IsaacLab/blob/main/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/manager_based/manipulation/lift/config/franka/ik_abs_env_cfg.py#L26

SakethTecholution, 2024-09-06:
ok got it thanks @Mayankm96

closed 2024-09-06 by SakethTecholution