isaac-sim/IsaacLab #921: [Bug Report] "Windows fatal exception: int divide by zero"

CrossStyle, 2024-09-02:
### Describe the bug
when I try runing standalone demo, e.g., create_empty.py, it shows that "Windows fatal exception: int divide by zero". It ran successfully once, and then never again.

### Steps to reproduce

python source/standalone/tutorials/00_sim/create_empty.py

```
(isaaclab) C:\Users\user\Desktop\code\sim\IsaacLab>python source/standalone/tutorials/00_sim/create_empty.py
[WARN][AppLauncher]: There are no arguments attached to the ArgumentParser object. If you have your own arguments, please load your own arguments before calling the `AppLauncher.add_app_launcher_args` method. This allows the method to check the validity of the arguments and perform checks for argument names.
[INFO][AppLauncher]: Loading experience file: C:\Users\user\Desktop\code\sim\IsaacLab\source\extensions\omni.isaac.lab\omni\isaac\lab\app\..\..\..\..\..\..\apps\isaaclab.python.kit
Starting kit application with the following args:  ['C:\\Users\\user\\Desktop\\code\\sim\\IsaacLab\\_isaac_sim\\exts\\omni.isaac.kit\\omni\\isaac\\kit\\simulation_app.py', 'C:\\Users\\user\\Desktop\\code\\sim\\IsaacLab\\source\\extensions\\omni.isaac.lab\\omni\\isaac\\lab\\app\\..\\..\\..\\..\\..\\..\\apps\\isaaclab.python.kit', '--/app/tokens/exe-path=C:\\Users\\user\\Desktop\\code\\sim\\IsaacLab\\_isaac_sim\\kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True', '--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--/app/fastShutdown=True', '--ext-folder', 'C:\\Users\\user\\Desktop\\code\\sim\\IsaacLab\\_isaac_sim/exts', '--ext-folder', 'C:\\Users\\user\\Desktop\\code\\sim\\IsaacLab\\_isaac_sim/apps', '--/renderer/activeGpu=0', '--/physics/cudaDevice=0', '--portable', '--/app/window/hideUi=False']
Passing the following args to the base kit application:  []
Loading user config located at: 'c:/users/user/desktop/code/sim/isaaclab/_isaac_sim/kit/data/Kit/Isaac-Sim/4.1/user.config.json'
[Info] [carb] Logging to file: c:/users/user/desktop/code/sim/isaaclab/_isaac_sim/kit/logs/Kit/Isaac-Sim/4.1/kit_20240902_232547.log
2024-09-02 15:25:47 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[0.161s] [ext: omni.kit.async_engine-0.0.0] startup
[0.513s] [ext: omni.stats-1.0.1] startup
[0.515s] [ext: omni.client-1.1.0] startup
[0.535s] [ext: omni.datastore-0.0.0] startup
[0.536s] [ext: omni.blobkey-1.1.0] startup
[0.537s] [ext: omni.ujitso.default-1.0.0] startup
[0.539s] [ext: omni.rtx.shadercache.vulkan-1.0.0] startup
[0.541s] [ext: omni.assets.plugins-0.0.0] startup
[0.545s] [ext: omni.gpu_foundation-0.0.0] startup
[0.587s] [ext: carb.windowing.plugins-1.0.0] startup
[0.696s] [ext: omni.kit.renderer.init-0.0.0] startup
[0.730s] [ext: omni.kit.pipapi-0.0.0] startup
[0.742s] [ext: omni.kit.pip_archive-0.0.0] startup
[0.746s] [ext: omni.pip.compute-1.4.0] startup
[0.749s] [ext: omni.pip.cloud-1.1.3] startup
[0.754s] [ext: omni.isaac.core_archive-2.3.0] startup
[0.755s] [ext: omni.mtlx-0.1.0] startup
[0.756s] [ext: omni.usd.config-1.0.4] startup
[0.776s] [ext: omni.gpucompute.plugins-0.0.0] startup
[0.781s] [ext: omni.usd.libs-1.0.1] startup
[0.957s] [ext: omni.kit.loop-isaac-1.2.0] startup
[0.958s] [ext: omni.kit.test-0.0.0] startup
[1.047s] [ext: omni.isaac.ml_archive-2.0.1] startup
[1.047s] [ext: omni.kit.telemetry-0.5.0] startup
[1.058s] [ext: omni.appwindow-1.1.8] startup
[1.103s] [ext: omni.kit.renderer.core-1.0.1] startup
[1.111s] [ext: omni.kit.renderer.capture-0.0.0] startup
[1.116s] [ext: omni.kit.renderer.imgui-1.0.1] startup
[1.130s] [ext: omni.ui-2.23.11] startup
[1.155s] [ext: omni.kit.mainwindow-1.0.3] startup
[1.158s] [ext: carb.audio-0.1.0] startup
[1.162s] [ext: omni.uiaudio-1.0.0] startup
[1.165s] [ext: omni.kit.uiapp-0.0.0] startup
[1.165s] [ext: omni.usd.schema.semantics-0.0.0] startup
[1.412s] [ext: omni.usd.schema.omnigraph-1.0.0] startup
[1.424s] [ext: omni.usd.schema.audio-0.0.0] startup
[1.437s] [ext: omni.usd.schema.forcefield-106.0.20] startup
[1.473s] [ext: omni.usd.schema.scene.visualization-2.0.2] startup
[1.477s] [ext: omni.usd.schema.isaac-2.1.0] startup
[1.487s] [ext: omni.usd.schema.physx-106.0.20] startup
[1.526s] [ext: omni.usd.schema.geospatial-0.0.0] startup
[1.532s] [ext: omni.usd.schema.anim-0.0.0] startup
[1.557s] [ext: omni.kit.window.popup_dialog-2.0.24] startup
[1.565s] [ext: omni.usd_resolver-1.0.0] startup
[1.577s] [ext: omni.kit.widget.nucleus_connector-1.1.8] startup
[1.582s] [ext: omni.timeline-1.0.10] startup
[1.585s] [ext: omni.usd.core-1.2.11] startup
[1.588s] [ext: omni.kit.usd_undo-0.1.8] startup
[1.589s] [ext: omni.hydra.scene_delegate-0.3.3] startup
[1.602s] [ext: omni.activity.core-1.0.1] startup
[1.606s] [ext: omni.resourcemonitor-105.0.1] startup
[1.612s] [ext: omni.graph.exec-0.9.3] startup
[1.615s] [ext: omni.kit.actions.core-1.0.0] startup
[1.620s] [ext: usdrt.scenegraph-7.4.8] startup
[... 265 lines cut ...]
  File "C:\Users\user\Desktop\code\sim\IsaacLab\source\extensions\omni.isaac.lab\omni\isaac\lab\app\app_launcher.py", line 576 in _create_app
  File "C:\Users\user\Desktop\code\sim\IsaacLab\source\extensions\omni.isaac.lab\omni\isaac\lab\app\app_launcher.py", line 113 in __init__
  File "C:\Users\user\Desktop\code\sim\IsaacLab\source\standalone\tutorials\00_sim\create_empty.py", line 29 in <module>

```

### System Info

Describe the characteristic of your environment:

- Commit: [7452386]
- Isaac Sim Version: [e.g. 2022.2.0, this can be obtained by `cat ${ISAACSIM_PATH}/VERSION`]
- OS: [WIndow 11]
- GPU: [RTX 4090]
- CUDA: [11.8]
- GPU Driver: [560.81]

### Additional context

Add any other context about the problem here.

### Checklist

- [*] I have checked that there is no similar issue in the repo (**required**)
- [*] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

johnBuffer, 2024-09-09:
I also experienced this error and I solved it by disabling the data collection in the Omniverse launcher's settings.

johnBuffer, 2024-09-12:
Nevermind, I had the problem again yesterday after several days without any issue.

Trippasch, 2024-11-13:
I also have the same problem. For me, when I run the example in headless mode the standalone example scripts run and work as expected. I only experience this error when I set **headless** to **False**.
To be more specific, when I try to run standalone example of "object_based_sdg.py" for example with headless set to False in the config, I also get the above error. Specifically, the error happens when I try  to create the simulation app in this line: 
``` simulation_app = SimulationApp(launch_config=launch_config) ```.

SamuelAnsel, 2024-11-15:
Same problem here, any workaround ?

RandomOakForest, 2024-11-15:
Thanks for posting this. It is likely a different driver will help, as it has been suggested in [this Isaac Sim forum topic](https://forums.developer.nvidia.com/t/windows-fatal-exception-int-divide-by-zero/313166). Could you give this solution a try and let us know if the problem remains? Thanks!

Trippasch, 2024-11-16:
I've tried different driver versions and the problem is still there.

> Thanks for posting this. It is likely a different driver will help, as it has been suggested in [this Isaac Sim forum topic](https://forums.developer.nvidia.com/t/windows-fatal-exception-int-divide-by-zero/313166). Could you give this solution a try and let us know if the problem remains? Thanks!

RandomOakForest, 2025-01-05:
Thank you for trying. We will investigate.

Trippasch, 2025-01-08:
Hi,

A person in the [Isaac Sim forum topic](https://forums.developer.nvidia.com/t/windows-fatal-exception-int-divide-by-zero/313166) suggested that turning MSI Afterburner off, gets rid of the errors. I can confirm that too for my use case.

Hope this helps someone!

RandomOakForest, 2025-02-11:
Thank you for following up with [this workaround](https://forums.developer.nvidia.com/t/windows-fatal-exception-int-divide-by-zero/313166/12). If you still face this issue, please try upgrading to the latest Isaac Lab.