isaac-sim/IsaacLab #926: [Bug Report] Cannot access attribute "set_camera_view" for class "SimulationContext"

waitingyondripen, 2024-09-03:
[image]
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

There are some unexpected errors existing in the code, but the code can run normally, I think I don't configure the path completely, but I don't know the exact trouble position. Could you please help me figure out.

### Steps to reproduce

Please try to provide a minimal example to reproduce the bug. Error messages and stack traces are also helpful.

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: [e.g. 2022.2.0, this can be obtained by `cat ${ISAACSIM_PATH}/VERSION`]
- OS: [e.g. Ubuntu 20.04]
- GPU: [e.g. RTX 2060 Super]
- CUDA: [e.g. 11.4]
- GPU Driver: [e.g. 470.82.01, this can be seen by using `nvidia-smi` command.]

### Additional context

Add any other context about the problem here.

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

closed 2024-09-11 by pascal-roth

zxhuang97, 2024-09-25:
I think the issue here is that Pylance mistakenly locate `omni.isaac.core.simulation_context` instead of `omni.isaac.lab.sim.simulation_context`.  Would be great if there's a way to fix this.

[image]

ricard-inho, 2024-11-11:
@zxhuang97 Have you found a solution for this?

Marvelousp4, 2025-03-07:
You can add this to pyproject.toml

[tool.pyright]
extraPaths = [
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.app.about",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.app.selector",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.app.setup",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.asset.browser",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.asset.exporter.urdf",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.asset.exporter.urdf/pip_prebundle",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.asset.gen.conveyor",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.asset.gen.conveyor.ui",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.asset.gen.omap",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.asset.gen.omap.ui",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.asset.importer.heightmap",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.benchmark.examples",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.benchmark.services",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.code_editor.jupyter",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.code_editor.vscode",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.core.api",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.core.cloner",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.core.deprecation_manager",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.core.includes",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.core.nodes",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.core.prims",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.core.simulation_manager",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.core.throttling",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.core.utils",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.core.version",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.cortex.behaviors",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.cortex.framework",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.examples.browser",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.examples.extension",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.examples.interactive",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.examples.ui",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.gui.components",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.gui.menu",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.gui.property",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.replicator.behavior",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.replicator.behavior.ui",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.replicator.domain_randomization",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.replicator.examples",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.replicator.scene_blox",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.replicator.synthetic_recorder",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.replicator.writers",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot.manipulators",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot.manipulators.examples",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot.manipulators.ui",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot.policy.examples",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot.schema",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot.surface_gripper",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot.surface_gripper.ui",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot.wheeled_robots",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot.wheeled_robots.ui",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot_motion.lula",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot_motion.lula/pip_prebundle",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot_motion.lula_test_widget",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot_motion.motion_generation",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot_setup.assembler",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot_setup.gain_tuner",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot_setup.grasp_editor",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot_setup.import_wizard",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot_setup.xrdf_editor",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.ros1.bridge",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.ros2.bridge",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.ros2.tf_viewer",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.ros2.urdf",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.sensors.camera",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.sensors.camera.ui",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.sensors.physics",
[... 552 lines cut ...]
        "${workspaceFolder}/_isaac_sim/extscache/omni.usd.schema.semantics-0.0.0+d02c707b.lx64.r.cp310",
        "${workspaceFolder}/_isaac_sim/extscache/omni.usd.schema.sequence-2.3.1+106.5.0.lx64.r.cp310",
        "${workspaceFolder}/_isaac_sim/extscache/omni.usd_resolver-1.0.0+d02c707b.lx64.r.cp310",
        "${workspaceFolder}/_isaac_sim/extscache/omni.vdb_timesample_editor-0.1.11",
        "${workspaceFolder}/_isaac_sim/extscache/omni.videoencoding-0.1.1+d02c707b.lx64.r.cp310",
[... cut ...]
        "${workspaceFolder}/_isaac_sim/kit/kernel/py",
        "${workspaceFolder}/_isaac_sim/kit/plugins/bindings-python",
        "${workspaceFolder}/_isaac_sim/kit/python/lib/python3.10",
        "${workspaceFolder}/_isaac_sim/kit/python/lib/python3.10/site-packages",
        "${workspaceFolder}/source/isaaclab_mimic",
        "${workspaceFolder}/source/isaaclab",
        "${workspaceFolder}/source/isaaclab_rl",
        "${workspaceFolder}/source/isaaclab_assets",
        "${workspaceFolder}/source/isaaclab_tasks"
    ]