isaac-sim/IsaacLab #929: [Question] Help: Camera outputting same image to files despite moving in IsaacSim

jbt-cmu, 2024-09-03:
### Question

As my "rod" agent moves across the scene, so too does the affixed camera, meaning the saved image every 200 timesteps should change. However, it remains the same as the image provided upon reset. I'm not sure if this is a bug or something I've done, please help:

# Copyright (c) 2022-2024, The Isaac Lab Project Developers.
# All rights reserved.
#
# SPDX-License-Identifier: BSD-3-Clause

"""
This script demonstrates how to run IsaacSim via the AppLauncher

.. code-block:: bash

    # Usage
    ./isaaclab.sh -p source/standalone/tutorials/00_sim/launch_app.py

"""

"""Launch Isaac Sim Simulator first."""

import argparse

from omni.isaac.lab.app import AppLauncher

# create argparser
parser = argparse.ArgumentParser(description="Tutorial on running IsaacSim via the AppLauncher.")
parser.add_argument("--size", type=float, default=1.0, help="Side-length of cuboid")
# SimulationApp arguments https://docs.omniverse.nvidia.com/py/isaacsim/source/extensions/omni.isaac.kit/docs/index.html?highlight=simulationapp#omni.isaac.kit.SimulationApp
parser.add_argument(
    "--width", type=int, default=1280, help="Width of the viewport and generated images. Defaults to 1280"
)
parser.add_argument(
    "--height", type=int, default=720, help="Height of the viewport and generated images. Defaults to 720"
)

# append AppLauncher cli args
AppLauncher.add_app_launcher_args(parser)
# parse the arguments
args_cli = parser.parse_args()
# launch omniverse app
app_launcher = AppLauncher(args_cli)
simulation_app = app_launcher.app

"""Rest everything follows."""

import omni.isaac.lab.sim as sim_utils
from omni.isaac.lab.sensors import Camera, CameraCfg
import omni.isaac.core.utils.prims as prim_utils
import gym
import cv2
import numpy as np
import torch
import os
import time
import random
from omni.isaac.lab.scene import InteractiveScene, InteractiveSceneCfg
from omni.isaac.lab.assets import AssetBaseCfg
from omni.isaac.lab.utils import configclass
from PIL import Image
from omni.isaac.lab.sim.spawners import RigidObjectSpawnerCfg
from omni.isaac.lab.sim import schemas

```
@configclass
class SimpleSceneCfg(InteractiveSceneCfg):
    """Design the scene with a rod and camera."""

    def __init__(self, *args, **kwargs):
        super().__init__(*args, **kwargs)
[... 203 lines cut ...]
    # Play the simulator
    sim.play()  # Start the simulation
    # Now we are ready!
    print("[INFO]: Setup complete...")

    # Simulate physics
    while simulation_app.is_running():
        # Generate random action
        action = env.action_space.sample()  # Random action

        action[0] = 1
        action[1] = 0
        action[2] = 0 
        for i in range(5000):
            observations, _, _, _ = env.step(action)

            image = observations["image"]

if __name__ == "__main__":
    # run the main function
    main()
    # close sim app
    simulation_app.close()

```

pascal-roth, 2024-09-09:
Hi, 
I assume that you placed your camera under the wrong prim. "{ENV_REGEX_NS}/Rod" will be the Xform under which the robot is spawned, but the actual articulation will be something like "{ENV_REGEX_NS}/Rod/base". For the camera to move with the robot, the camera would have to be placed under base. 

As a reference, you can check the example [here](https://github.com/isaac-sim/IsaacLab/blob/0c3fb1e11e455db48343ece651c1bb629f1884fe/source/standalone/demos/cameras.py#L84C39-L84C46)

pascal-roth, 2024-09-19:
Did this fix your issue?

closed 2025-01-05 by RandomOakForest

pascal-roth, 2025-05-14:
What kind of a prim is the "{ENV_REGEX_NS}/Robot/robot/tooltip" ?