isaac-sim/IsaacLab #931: [Question] Is there a solution for the Patch buffer overflow error?

H-Hisamichi, 2024-09-04:
### Question

I'm looking for a solution to this error.

`
[INFO]: Base environment:
	Environment device    : cuda:0
	Physics step-size     : 0.005
	Rendering step-size   : 0.02
	Environment step-size : 0.02
	Physics GPU pipeline  : True
	Physics GPU simulation: True
[INFO]: Time taken for scene creation : 3.000548 seconds

[INFO]: Scene manager:  <class InteractiveScene>
	Number of environments: 4096
	Environment spacing   : 4.0
	Source prim name      : /World/envs/env_0
	Global prim paths     : ['/World/ground']
	Replicate physics     : True

[INFO]: Starting the simulation. This may take a few seconds. Please wait...
2024-09-04 07:26:10 [16,447ms] [Error] [omni.physx.plugin] PhysX error: Patch buffer overflow detected, please increase its size to at least 503808 in the scene desc!
, FILE /builds/omniverse/physics/physx/source/gpunarrowphase/src/PxgNarrowphaseCore.cpp, LINE 1272
2024-09-04 07:26:13 [19,646ms] [Error] [omni.physx.plugin] PhysX error: Patch buffer overflow detected, please increase its size to at least 494469 in the scene desc!
, FILE /builds/omniverse/physics/physx/source/gpunarrowphase/src/PxgNarrowphaseCore.cpp, LINE 1272
2024-09-04 07:26:14 [19,887ms] [Error] [omni.physx.plugin] PhysX error: Patch buffer overflow detected, please increase its size to at least 487277 in the scene desc!
, FILE /builds/omniverse/physics/physx/source/gpunarrowphase/src/PxgNarrowphaseCore.cpp, LINE 1272
2024-09-04 07:26:14 [20,106ms] [Error] [omni.physx.plugin] PhysX error: Patch buffer overflow detected, please increase its size to at least 486615 in the scene desc!
, FILE /builds/omniverse/physics/physx/source/gpunarrowphase/src/PxgNarrowphaseCore.cpp, LINE 1272
2024-09-04 07:26:14 [20,325ms] [Error] [omni.physx.plugin] PhysX error: Patch buffer overflow detected, please increase its size to at least 481943 in the scene desc!
, FILE /builds/omniverse/physics/physx/source/gpunarrowphase/src/PxgNarrowphaseCore.cpp, LINE 1272

[INFO]: Time taken for simulation start : 11.268937 seconds
[INFO]: Completed setting up the environment...

2024-09-04 07:26:15 [20,806ms] [Error] [omni.physx.plugin] PhysX error: PxPhysics::createMaterial: limit of 64K materials reached., FILE /builds/omniverse/physics/physx/source/physx/src/NpPhysics.cpp, LINE 598
2024-09-04 07:26:15 [20,806ms] [Error] [omni.physx.plugin] PhysX error: material pointer 0 is NULL!, FILE /builds/omniverse/physics/physx/source/physx/src/NpShape.h, LINE 232
...
`

I've checked the reference for omni.isaac.lab.sim.PhysxCfg, and it seems like the buffer is set to be significantly larger than what the error code suggests is needed. When I adjust the values in rl_games_ppo_cfg.yaml and limit the number of spawns to 100, the error doesn't occur. I've also tried increasing various buffer sizes using omni.isaac.lab.sim.PhysxCfg, but that didn't solve the issue. Additionally, I'm puzzled by the fact that after a few error messages, there's a log indicating the setup was completed (followed by another error...).
Any advice would be greatly ### ### appreciated.

H-Hisamichi, 2024-09-05:
The investigation into the error has made progress, so I will be closing this thread.

closed 2024-09-05 by H-Hisamichi

KKSTB, 2024-09-05:
I recently encouter the same PhysX patch buffer overflow erro, but during rsl-l training:
```
...
--------------------------------------------------------------------------------
                   Total timesteps: 8650752
                    Iteration time: 8.99s
                        Total time: 729.23s
                               ETA: 539.3s

2024-09-05 14:18:43 [759,765ms] [Error] [omni.physx.plugin] PhysX error: Patch buffer overflow detected, please increase its size to at least 178421 in the scene desc!
, FILE /builds/omniverse/physics/physx/source/gpunarrowphase/src/PxgNarrowphaseCore.cpp, LINE 1272
2024-09-05 14:18:43 [759,853ms] [Error] [omni.physx.plugin] PhysX error: Patch buffer overflow detected, please increase its size to at least 178421 in the scene desc!
, FILE /builds/omniverse/physics/physx/source/gpunarrowphase/src/PxgNarrowphaseCore.cpp, LINE 1272
2024-09-05 14:18:43 [759,946ms] [Error] [omni.physx.plugin] PhysX error: Patch buffer overflow detected, please increase its size to at least 178427 in the scene desc!
, FILE /builds/omniverse/physics/physx/source/gpunarrowphase/src/PxgNarrowphaseCore.cpp, LINE 1272
2024-09-05 14:18:43 [760,034ms] [Error] [omni.physx.plugin] PhysX error: Patch buffer overflow detected, please increase its size to at least 178475 in the scene desc!
, FILE /builds/omniverse/physics/physx/source/gpunarrowphase/src/PxgNarrowphaseCore.cpp, LINE 1272
2024-09-05 14:18:43 [760,133ms] [Error] [omni.physx.plugin] PhysX error: Patch buffer overflow detected, please increase its size to at least 177711 in the scene desc!
, FILE /builds/omniverse/physics/physx/source/gpunarrowphase/src/PxgNarrowphaseCore.cpp, LINE 1272
2024-09-05 14:18:43 [760,223ms] [Error] [omni.physx.plugin] PhysX error: Patch buffer overflow detected, please increase its size to at least 177915 in the scene desc!
, FILE /builds/omniverse/physics/physx/source/gpunarrowphase/src/PxgNarrowphaseCore.cpp, LINE 1272
...
```

Then the training crashes:
```
Error executing job with overrides: []
Traceback (most recent call last):
  File "/home/kk/IsaacLab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/hydra.py", line 91, in hydra_main
    func(env_cfg, agent_cfg, *args, **kwargs)
  File "/home/kk/IsaacLab/source/standalone/workflows/rsl_rl/train.py", line 127, in main
    runner.learn(num_learning_iterations=agent_cfg.max_iterations, init_at_random_ep_len=True)
  File "/home/kk/IsaacLab/_isaac_sim/kit/python/lib/python3.10/site-packages/rsl_rl/runners/on_policy_runner.py", line 111, in learn
    actions = self.alg.act(obs, critic_obs)
  File "/home/kk/IsaacLab/_isaac_sim/kit/python/lib/python3.10/site-packages/rsl_rl/algorithms/ppo.py", line 73, in act
    self.transition.actions = self.actor_critic.act(obs).detach()
  File "/home/kk/IsaacLab/_isaac_sim/kit/python/lib/python3.10/site-packages/rsl_rl/modules/actor_critic.py", line 105, in act
    return self.distribution.sample()
  File "/home/kk/.local/share/ov/pkg/isaac-sim-4.1.0/exts/omni.isaac.ml_archive/pip_prebundle/torch/distributions/normal.py", line 70, in sample
    return torch.normal(self.loc.expand(shape), self.scale.expand(shape))
RuntimeError: normal expects all elements of std >= 0.0

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.
2024-09-05 14:18:46 [763,187ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
2024-09-05 14:18:46 [763,191ms] [Warning] [omni.core.ITypeFactory] Module /home/kk/.local/share/ov/pkg/isaac-sim-4.1.0/kit/exts/omni.activity.core/bin/libomni.activity.core.plugin.so remained loaded after unload request.
There was an error running python
```

bd-fauna, 2024-11-07:
Seems like in your `class EnvCfg(ManagerBasedRLEnvCfg)`, you can updated the value by setting

```python
class EnvCfg(ManagerBasedRLEnvCfg):
    ...
    def __post_init__(self):
        self.sim.physx.gpu_max_rigid_patch_count = 4096 * 4096
```

SahasraKeerthi, 2025-09-16:
Even this [https://github.com/isaac-sim/IsaacLab/issues/931#issuecomment-2463352822](url) is not working for me, can anyone help? My training crashes like above

H-Hisamichi, 2025-09-17:
Hello @SahasraKeerthi,

In the latest IsaacLab v2.2.0, you can specify the buffer size in the [RL config](https://github.com/isaac-sim/IsaacLab/blob/75b67154b96ab839a8ee9352b19115719e8cfb84/source/isaaclab_tasks/isaaclab_tasks/manager_based/locomotion/velocity/velocity_env_cfg.py#L314).
Please refer to this [document](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.sim.html#isaaclab.sim.PhysxCfg) for the required buffer.