isaac-sim/IsaacLab #934: [Bug Report] Bug title

Vinci0007, 2024-09-04:
Describe Bug:
When I use conda env to run some isaac-sim DEMO, it cannot work.
System Info
WINDOWS 11 + Anaconda 3 + Python 3.10
Isaac Sim Version: 4.1.0
CUDA: 12.1
GPU Driver: 560.94 and 550
### LOGs

[Warning] [omni.isaac.kit.simulation_app] Modules: ['omni.kit_app'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Traceback (most recent call last):
File "E:\0001_Work_New\005_YangGu\0001_ROBOT\00004_RobotSim2Real\Isaacsim\work\env_test.py", line 70, in
env = SimulationApp()
File "D:\Documents\VirtualMachineOS\Anaconda\envs\isaac_linux\Lib\site-packages\isaacsim\exts\omni.isaac.kit\omni\isaac\kit\simulation_app.py", line 198, in init
self._app = omni.kit.app.get_app()
File "D:\Documents\VirtualMachineOS\Anaconda\envs\isaac_linux\Lib\site-packages\omni\kernel\py\omni\kit\app_impl\app_iface.py", line 19, in get_app
return get_app_interface()
File "D:\Documents\VirtualMachineOS\Anaconda\envs\isaac_linux\Lib\site-packages\omni\kernel\py\omni\kit\app_impl\app_iface.py", line 14, in get_app_interface
return acquire_app_interface()
RuntimeError: Failed to acquire interface: omni::kit::IApp (pluginName: nullptr)

Checklist

 I have checked that there is no similar issue in the repo (required)

 I have checked that the issue is not in running Isaac Sim itself and is related to the repo

jtigue-bdai, 2024-09-06:
Please update ticket with what GPU hardware you are using and please provide a reproduction script that produces this error.

bearpaw, 2024-09-24:
Same error here:
```bash
(isaaclab) $ ./isaaclab.sh -p source/standalone/tutorials/00_sim/create_empty.py
[INFO] Using python from: /home/weiy/Softwares/miniconda3/envs/isaaclab/bin/python                                                                                                                                                                                                                                                               
[WARN][AppLauncher]: There are no arguments attached to the ArgumentParser object. If you have your own arguments, please load your own arguments before calling the `AppLauncher.add_app_launcher_args` method. This allows the method to check the validity of the arguments and perform checks for argument names.
[INFO][AppLauncher]: Loading experience file: /home/weiy/Projects/code/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/../../../../../../apps/isaaclab.python.kit
[Warning] [omni.isaac.kit.simulation_app] Modules: ['omni.kit_app'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Traceback (most recent call last):
  File "/home/weiy/Projects/code/IsaacLab/source/standalone/tutorials/00_sim/create_empty.py", line 29, in <module>
    app_launcher = AppLauncher(args_cli)
  File "/home/weiy/Projects/code/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/app_launcher.py", line 113, in __init__
    self._create_app()
  File "/home/weiy/Projects/code/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/app_launcher.py", line 576, in _create_app
    self._app = SimulationApp(self._sim_app_config, experience=self._sim_experience_file)
  File "/home/weiy/Softwares/miniconda3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py", line 199, in __init__
    self._app = omni.kit.app.get_app()
  File "/home/weiy/Softwares/miniconda3/envs/isaaclab/lib/python3.10/site-packages/omni/kernel/py/omni/kit/app/_impl/app_iface.py", line 19, in get_app
    return get_app_interface()
  File "/home/weiy/Softwares/miniconda3/envs/isaaclab/lib/python3.10/site-packages/omni/kernel/py/omni/kit/app/_impl/app_iface.py", line 14, in get_app_interface
    return acquire_app_interface()
RuntimeError: Failed to acquire interface: omni::kit::IApp (pluginName: nullptr)
(isaaclab) weiy@lfp:~/Projects/code/IsaacLab$ uname -a
Linux lfp 6.8.0-44-generic #44-Ubuntu SMP PREEMPT_DYNAMIC Tue Aug 13 13:35:26 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

```
Ubuntu with conda environment:
```bash
OS: Ubuntu 24.04.1 LTS 
Driver Version: 560.35.03
CUDA Version: 12.6 
```

Vinci0007, 2024-09-25:
I try to use binary install the isaac and use conda to build isaac env, it can work. This time , I think it is the best way to resolve this problem, donot use conda/.env to install the whole environment to use the ISAAC . 
I hope this way will be helpful to all of you .
BTW:
I also try the way from "@jtigue-bdai", but it cannot work also.

jtigue-bdai, 2024-09-25:
@kellyguo11 you might know best with this issue.

Vinci0007, 2024-10-08:
There are no better solution. 
# Pending.............

closed 2024-10-08 by Vinci0007