isaac-sim/IsaacLab #937: Turn off error or warning

Entongsu, 2024-09-05:
Hi,

I wonder how to turn off the warning and error printing during the simulation. Could you provide some hints about that? Thank you.

Mayankm96, 2024-09-05:
You can check this here: https://docs.omniverse.nvidia.com/kit/docs/carbonite/latest/docs/omni.log/Logging.html

Mayankm96, 2024-09-16:
Closing this MR since there isn't any update in the activity and the above reply seems to answer the question.

closed 2024-09-16 by Mayankm96

bikcrum, 2025-08-31:
I also ran into this. The logging docs linked are the right place, but just to share what worked for me in IsaacLab:

When launching with `play.py`, you can forward Kit logging options via the `--kit_args` flag. Important: all the log options need to be passed as **one argument**. For example, to only show errors:

```bash
play.py --task My-Task-v0 \
 --kit_args "--/log/level=error --/log/outputStreamLevel=error --/log/fileLogLevel=error"