isaac-sim/IsaacLab #941: [Question] PhysX Error: Material Limit and Buffer Overflow Issues with High Agent Counts in IsaacLab

H-Hisamichi, 2024-09-05:
### Question

Hello everyone,
I’m encountering some issues while running reinforcement learning experiments in IsaacLab with large agent counts, specifically when using more than 107 agents. Here’s a summary of the problem:

- When the agent count exceeds 107, I start seeing the following error:
```
[Error] [omni.physx.plugin] PhysX error: PxPhysics::createMaterial: limit of 64K materials reached., FILE /builds/omniverse/physics/physx/source/physx/src/NpPhysics.cpp, LINE 598
[Error] [omni.physx.plugin] PhysX error: material pointer 0 is NULL!, FILE /builds/omniverse/physics/physx/source/physx/src/NpShape.h, LINE 232
```
- With more than 2048 agents, I get additional buffer overflow errors(The RL still runs despite the errors occurring.):
```
[Error] [omni.physx.plugin] PhysX error: Patch buffer overflow detected, please increase its size to at least 372736 in the scene desc!
```
- At 4096 agents, the errors increase, and eventually, the application crashes with warnings like:
```
[INFO] Recording videos during training.
        video_folder: /home/robot/IsaacLab/logs/rl_games/anymal_c_flat/2024-09-05_16-01-16/videos
        step_trigger: lambda step: step % args_cli.video_interval == 0
        video_length: 200
        disable_logger: True
/home/robot/anaconda3/envs/HexapodRobot/lib/python3.10/site-packages/gymnasium/core.py:297: UserWarning: WARN: env.is_vector_env to get variables from other wrappers is deprecated and will be removed in v1.0, to get this variable you can do `env.unwrapped.is_vector_env` for environment variables or `env.get_attr('is_vector_env')` that will search the reminding wrappers.
  logger.warn(
self.seed = 42
Setting seed: 42
Started to train
Exact experiment name requested from command line: 2024-09-05_16-01-16
seq_length: 4
current training device: cuda:0
build mlp: 33
RunningMeanStd:  (1,)
2024-09-05 07:14:45 [820,319ms] [Warning] [omni.fabric.plugin] No source has valid data array=0x1529a8ca0 usdValid=0 cpuValid=0
2024-09-05 07:14:45 [820,322ms] [Warning] [omni.fabric.plugin] No source has valid data array=0x3e4eb680 usdValid=0 cpuValid=0
2024-09-05 07:14:46 [821,306ms] [Error] [omni.physx.plugin] PhysX error: Patch buffer overflow detected, please increase its size to at least 172032 in the scene desc!
, FILE /builds/omniverse/physics/physx/source/gpunarrowphase/src/PxgNarrowphaseCore.cpp, LINE 1272
2024-09-05 07:14:55 [830,028ms] [Warning] [omni.usd] Unexpected reference count of 2 for UsdStage 'anon:0x115bbd90:World0.usd' while being closed in UsdContext (this may indicate it is still resident in memory).
[830.053s] Simulation is stopped. Shutting down the app...
```
When using fewer than 107 agents, everything works fine, and I can run the reinforcement learning without any issues. It seems the errors are related to PhysX material limits and buffer overflows.
Has anyone experienced similar issues with high agent counts in IsaacLab or any other PhysX-based environments? Are there any recommendations for increasing the material or buffer limits to handle larger simulations? I’ve already tried adjusting `gpu_max_rigid_patch_count`, but the errors persist.

Thanks in advance for your help!

Mayankm96, 2024-09-06:
The error is correct for what is says. You're registering a lot more rigid body materials than what PhysX can handle. You'd need to simplify the randomization probably (i.e. reduce number of physics materials made).

Can you share more information on your setup?

H-Hisamichi, 2024-09-07:
Hi, @Mayankm96 

Thank you for the advice. I hadn't considered that randomization might be the cause.

- In my RL code, I’m spawning a Hexapod Robot composed of 19 parts:
```
# Copyright (c) 2022-2024, The Isaac Lab Project Developers.
# All rights reserved.
#
# SPDX-License-Identifier: BSD-3-Clause

from __future__ import annotations

import torch

import omni.isaac.lab.sim as sim_utils
from omni.isaac.lab.assets import Articulation, ArticulationCfg
from omni.isaac.lab.envs import DirectRLEnv, DirectRLEnvCfg
from omni.isaac.lab.scene import InteractiveSceneCfg
from omni.isaac.lab.sensors import ContactSensor, ContactSensorCfg, RayCaster, RayCasterCfg, patterns
from omni.isaac.lab.sim import SimulationCfg, PhysxCfg
from omni.isaac.lab.terrains import TerrainImporterCfg
from omni.isaac.lab.utils import configclass, math
from pxr import PhysxSchema
#from omni.isaac.lab.sim.SimulationCfg import physics_prim_path

##
# Pre-defined configs
##
from omni.isaac.lab_assets.hr_v2 import HR_CFG # isort: skip
from omni.isaac.lab.terrains.config.rough import ROUGH_TERRAINS_CFG  # isort: skip
import omni.isaac.core.utils.transformations as transformations_utils
from torch.utils.tensorboard import SummaryWriter
from omni.isaac.lab.sim.spawners import RigidObjectSpawnerCfg

@configclass
class HexapodFlatEnvCfg(DirectRLEnvCfg):
    # env
    episode_length_s = 20.0
    decimation = 4
    action_scale = 1.0
    servo1_action_scale = 0.3
    num_actions = 18
    num_observations = 33
    num_states = 0

    # simulation
    sim: SimulationCfg = SimulationCfg(
        dt=1 / 200,
        render_interval=decimation,
        disable_contact_processing=True,
        physics_material=sim_utils.RigidBodyMaterialCfg(
            friction_combine_mode="multiply",
            restitution_combine_mode="multiply",
            static_friction=1.0,
            dynamic_friction=1.0,
            restitution=0.0,
        ),
    )
    terrain = TerrainImporterCfg(
        prim_path="/World/ground",
        terrain_type="plane",
        collision_group=-1,
        physics_material=sim_utils.RigidBodyMaterialCfg(
            friction_combine_mode="multiply",
            restitution_combine_mode="multiply",
            static_friction=1.0,
            dynamic_friction=1.0,
            restitution=0.0,
        ),
[... 395 lines cut ...]
      scale_value: 0.6
    normalize_advantage: True
    gamma: 0.99
    tau: 0.95
    learning_rate: 1e-3
    lr_schedule: adaptive
    schedule_type: legacy
    kl_threshold: 0.01
    score_to_win: 20000
    max_epochs: 5000
    save_best_after: 100
    save_frequency: 50
    grad_norm: 1.0
    entropy_coef: 0.005
    truncate_grads: True
    e_clip: 0.2
    horizon_length: 24
    #minibatch_size: 24576
    minibatch_size: 49152
    mini_epochs: 5
    critic_coef: 2.0
    clip_value: True
    seq_length: 4
    bounds_loss_coef: 0.0
```

H-Hisamichi, 2024-09-09:
Additional information.

- Robot model configuration:
```
# Copyright (c) 2022-2024, The Isaac Lab Project Developers.
# All rights reserved.
#
# SPDX-License-Identifier: BSD-3-Clause

"""Configuration for the Mujoco Ant robot."""

from __future__ import annotations

import omni.isaac.lab.sim as sim_utils
from omni.isaac.lab.actuators import ImplicitActuatorCfg
from omni.isaac.lab.assets import ArticulationCfg
from omni.isaac.lab.utils.assets import ISAAC_NUCLEUS_DIR
import torch
import numpy as np
from omni.isaac.lab.utils.math import quat_from_angle_axis, quat_mul

##
# Configuration
##

angles_y = torch.tensor([1.5708], dtype=torch.float32)
axes_y = torch.tensor([[0, 1, 0]], dtype=torch.float32)
angles_z = torch.tensor([1.5708], dtype=torch.float32)
axes_z = torch.tensor([[0, 0, 1]], dtype=torch.float32)
quaternions_y = quat_from_angle_axis(angles_y, axes_y)
quaternions_z = quat_from_angle_axis(angles_z, axes_z)
quaternions = quat_mul(quaternions_y, quaternions_z)
numpy_array = quaternions.numpy()
#numpy_array = quaternions_y.numpy()
formatted_tuple = tuple(float(f"{val:.4f}") for val in numpy_array[0])
print(f"formatted_tuple: {formatted_tuple}")

HR_CFG = ArticulationCfg(
    prim_path="{ENV_REGEX_NS}/Robot",
    spawn=sim_utils.UsdFileCfg(
        #usd_path=f"{ISAAC_NUCLEUS_DIR}/Robots/Ant/ant_instanceable.usd",
        #usd_path="/home/robot/hisamichi/IsaacSim/IsaacSim_HexapodBody_simplification_v1.usd",
        usd_path="/home/robot/hisamichi/IsaacSim/IsaacSim_HexapodBody_simple_v2.usd",
        rigid_props=sim_utils.RigidBodyPropertiesCfg(
            disable_gravity=False,
            retain_accelerations=False, # add
            linear_damping=0.0, # add
            angular_damping=0.0, # add
            max_linear_velocity=1000.0, # add
            max_angular_velocity=1000.0, # add
            max_depenetration_velocity=1.0,
            #enable_gyroscopic_forces=True,
        ),
        articulation_props=sim_utils.ArticulationRootPropertiesCfg(
            enabled_self_collisions=True,
            solver_position_iteration_count=4,
            solver_velocity_iteration_count=0,
            #fix_root_link=True,
            #sleep_threshold=0.005,
            #stabilization_threshold=0.001,
        ),
        copy_from_source=False,
        activate_contact_sensors=True, # add
    ),
    init_state=ArticulationCfg.InitialStateCfg(
        pos=(0.0, 0.0, 0.36),
        rot=formatted_tuple,
        joint_pos={
            "Leg1RevJoint1":0.0, "Leg2RevJoint1":0.0, "Leg3RevJoint1":0.0, "Leg4RevJoint1":0.0, "Leg5RevJoint1":0.0, "Leg6RevJoint1":0.0,
            "Leg1RevJoint2":1.71, "Leg2RevJoint2":1.71, "Leg3RevJoint2":1.71, "Leg4RevJoint2":1.71, "Leg5RevJoint2":1.71, "Leg6RevJoint2":1.71,
            "Leg1RevJoint3":-2.195, "Leg2RevJoint3":-2.195, "Leg3RevJoint3":-2.195, "Leg4RevJoint3":-2.195, "Leg5RevJoint3":-2.195, "Leg6RevJoint3":-2.195
        },
    ),
    actuators={
        "Leg": ImplicitActuatorCfg(
            joint_names_expr=["Leg.*RevJoint.*"],
            stiffness=None,
            damping=None,
        ),
    },
)

```

MiladShafiee, 2024-09-09:
The issue arises in the following section of your code:

```python
#      Randomize robot friction
        env_ids = self._robot._ALL_INDICES
        mat_props = self._robot.root_physx_view.get_material_properties()
        mat_props[:, :, :2].uniform_(0.6, 0.8)
        self._robot.root_physx_view.set_material_properties(mat_props, env_ids.cpu())

        # Randomize base mass
        base_id, _ = self._robot.find_bodies(".*Arm3")
        masses = self._robot.root_physx_view.get_masses()
        masses[:, base_id] += torch.zeros_like(masses[:, base_id]).uniform_(1.0, 5.0)
        self._robot.root_physx_view.set_masses(masses, env_ids.cpu())
```

It seems like you're using a direct environment, and this problem is not fully handled in this method. However, if you look at the manager-based randomization in the [this line in events](https://github.com/isaac-sim/IsaacLab/blob/af088f593216f7215310cf5e3a13a259d7320a63/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/mdp/events.py#L93), you’ll notice that the randomization process is handled differently there. So as error said the original method of randomizing material properties using PhysX might hit the 64k material limit. To handle this, we need to use a bucketing mechanism. Material properties are divided into 64 buckets, and the values are assigned by binning the randomized values into these buckets.

### Quick Solution
To address this issue, you can modify the code as follows:

```python
        # Get body ids
        for _, actuator_cfg in self.cfg.robot.actuators.items():
                                self._joint_ids, self._joint_names = self._robot.find_joints(actuator_cfg.joint_names_expr)
                                self._body_ids, _ = self._robot.find_bodies(self._robot.body_names)

        # Randomize robot friction
        env_ids = self._robot._ALL_INDICES.cpu()
        mat_props = self._robot.root_physx_view.get_material_properties()
        mat_props[:, :, :2].uniform_(0.6, 0.8)
        lo = torch.tensor([0.6, 0.8, 0.0], device="cpu")
        hi = torch.tensor([0.6, 0.8, 0.0], device="cpu")

        material_samples = torch.zeros(mat_props[env_ids].shape)
        # to avoid 64k material limit in physx, we bucket materials by binning randomized material properties
            # into buckets based on the number of buckets specified
        for d in range(3):
            buckets = torch.tensor([(hi[d] - lo[d]) * i / 64 + lo[d] for i in range(64)], device="cpu")
            material_samples[..., d] = buckets[torch.searchsorted(buckets, material_samples[..., d].contiguous()) - 1]
    
        # update material buffer with new samples
            # obtain number of shapes per body (needed for indexing the material properties correctly)
            # note: this is a workaround since the Articulation does not provide a direct way to obtain the number of shapes
            #  per body. We use the physics simulation view to obtain the number of shapes per body.
        num_shapes_per_body = []
        materials = self._robot.root_physx_view.get_material_properties()
        for link_path in self._robot.root_physx_view.link_paths[0]:
            link_physx_view = self._robot._physics_sim_view.create_rigid_body_view(link_path)  # type: ignore
            num_shapes_per_body.append(link_physx_view.max_shapes)
        # sample material properties from the given ranges
        for body_id in self._body_ids:
            # start index of shape
            start_idx = sum(num_shapes_per_body[:body_id])
            # end index of shape
            end_idx = start_idx + num_shapes_per_body[body_id]
            # assign the new materials
            # material ids are of shape: num_env_ids x num_shapes
            # material_buckets are of shape: num_buckets x 3
            materials[env_ids, start_idx:end_idx] = material_samples[:, start_idx:end_idx]
        self._robot.root_physx_view.set_material_properties(mat_props, env_ids.cpu())

        # Randomize base mass
        base_id, _ = self._robot.find_bodies("base")
        masses = self._robot.root_physx_view.get_masses()
        masses[:, base_id] += torch.zeros_like(masses[:, base_id]).uniform_(-5.0, 5.0)
        self._robot.root_physx_view.set_masses(masses, env_ids.cpu())
```

We used the same solution that used in `events.py`. For simplicity of sharing code here, the number `64` is hardcoded as the number of buckets. You can make this a configurable parameter in your code.

H-Hisamichi, 2024-09-11:
Hi, @Mayankm96 , @MiladShafiee 

Thank you everyone for your help. I managed to resolve the error, and I really appreciate your support.

On a slightly different note, I came across something in the [documentation](https://isaac-sim.github.io/IsaacLab/source/migration/migrating_from_isaacgymenvs.html) under the `Simulation Config` section. The code snippet ```physx: PhysxCfg = PhysxCfg()``` seems like it might be a syntax error, so I thought I’d mention it.

MiladShafiee, 2024-09-11:
@H-Hisamichi  Great to hear that, so the problem was ```64k material limit in physx,```? and solved with bucketing materials? If yes I may open a MR to add it to the Anymal Direct Demo

H-Hisamichi, 2024-09-12:
@MiladShafiee 

Yes, that's correct. After modifying the randomization part of my code following your advice, the '64k material limit in PhysX' error was resolved.
As for the buffer overflow error, I was able to fix it by implementing the following code:
(`physx = sim_utils.PhysxCfg(gpu_max_rigid_patch_count=2**24`, which resolved the issue by adjusting the GPU's rigid patch count.)
```
class HexapodFlatEnvCfg(DirectRLEnvCfg):
    # env
    episode_length_s = 20.0
    decimation = 4
    action_scale = 1.0
    servo1_action_scale = 0.3
    num_actions = 18
    num_observations = 33
    num_states = 0

    # simulation
    sim: SimulationCfg = SimulationCfg(
        dt=1 / 200,
        render_interval=decimation,
        disable_contact_processing=True,
        physics_material=sim_utils.RigidBodyMaterialCfg(
            friction_combine_mode="multiply",
            restitution_combine_mode="multiply",
            static_friction=1.0,
            dynamic_friction=1.0,
            restitution=0.0,
        ),
        physx = sim_utils.PhysxCfg(
            gpu_max_rigid_patch_count=2**24
        )
    )
    terrain = TerrainImporterCfg(
        prim_path="/World/ground",
        terrain_type="plane",
        collision_group=-1,
        physics_material=sim_utils.RigidBodyMaterialCfg(
            friction_combine_mode="multiply",
            restitution_combine_mode="multiply",
            static_friction=1.0,
            dynamic_friction=1.0,
            restitution=0.0,
        ),
        debug_vis=False,
    )
```

MiladShafiee, 2024-09-12:
Ok great thanks @H-Hisamichi .

Hi @Mayankm96, What do you think about adding these corrections to Anymal-C Direct Demo? It seems other direct environments not having randomization and not using `set_material_properties`.

Mayankm96, 2024-09-16:
Yes definitely. @kellyguo11 for vis.

MiladShafiee, 2024-09-16:
I was just about to open a pull request that I saw you had already submitted one. Thank you so much.

closed 2024-09-19 by Dhoeller19