isaac-sim/IsaacLab #943: [Proposal] Modular config for RL (initially for Skrl lib)

lopatovsky, 2024-09-05:
### Proposal

Implement Reinforcement Learning config class, which configures all parameters related to RL ( e.g. RL agent, RL model. ) in modular fashion using architecture identical to ManagerBasedRLEnvCfg.

I already have it partially implemented for Skrl lib. This is an example how this could be used :
(This example is just showing the main config with some simple overwrites. Subconfigs are declared elsewhere with default values taken from agents/skrl_ppo_cfg.yaml)

```
@configclass
class HumanoidRLCfg(RLCfg):

    seed = 42
    rl_library = "skrl"

    model: ModelCfg = SkrlModelCfg(
        separate=True,
        policy=GaussianMLPModelCfg(
            hiddens=[256, 128, 64],
        ),
        value=DeterministicMLPModelCfg(
            hiddens=[256, 128, 64],
        ),
    )

    agent: AgentCfg = PPOAgentCfg(learning_rate=1e-4)

    experiment: ExperimentCfg = ExperimentCfg(
        directory="humanoid", 
        experiment_name="example", 
        write_interval=128, 
        checkpoint_interval=5000
    )

    trainer: TrainerCfg = TrainerCfg(timesteps=160000)
```

This config is later processed in skrl/train.py 

### Motivation

Architecture of ManagerBasedRLEnvCfg has modular architecture using @configclass, this is very nice design because it enables flexible manipulation with submodules, what we are using when running experiments across multiple RL tasks.

However RL config (for skrl lib) exists as a single yaml file. Unifying it to modular architecture identical to ManagerBasedRLEnvCfg would make experiment in Isaac-lab smoother and cleaner. 

In the future all RL libs could be using this type of configuration. It can make reusing the modules across libraries possible. Eventually it would be possible to start all libs from one train script and change your task to other library just by toggling the rl_library parameter.

In addition this architecture allows easy extention of current functionality by external libraries: 

(e.g. I am developing a library for model instantiation, that should be modular and powerful enough to instantiate any model's architecture easily. If the Isaac lab RL configuration is modular, one can simply switch the model config for this new lib model config.) 

### Alternatives

Proposed change could live directly in ManagerBasedRLEnvCfg.  (however this only makes sense in the future, if this was wildly adopted by all other RL libs in Isaac-lab)

Please let me know if this contribution is welcomed (and not conflicting with some of your plans) and I will issue a PR for it.

RandomOakForest, 2025-02-14:
Following up, @Toni-SM for vis. per @kellyguo11.

StafaH, 2026-09-16:
After discussing with the maintainer of skrl @Toni-SM, we have decided not to support configclasses for skrl. yaml is the preferred config method for skrl.

We recommend using rsl-rl if you would like a modular/typed configclass rl system. You can also implement locally a configclass system for skrl, se #7606

closed 2026-09-16 by StafaH