isaac-sim/IsaacLab #947: [Bug Report] AttributeError: 'RigidObject' object has no attribute '_external_force_b' when spawning Rigid Object on DirectRL Environment.

SantiDiazC, 2024-09-06:
Hi, I am trying to create an DirectRL environment in which I spawn some robot, gripper and objects for a grasping task. When spawning the robot and gripper I don't have any problem, however, when I tried to add the objects I have the following error:

``` error
Traceback (most recent call last):
  File "/home/hri-david/Omniverse/Isaaclab/IsaacLab/Santi_tests/DirectEnv_examples/test_env.py", line 760, in <module>
    main()
  File "/home/hri-david/Omniverse/Isaaclab/IsaacLab/Santi_tests/DirectEnv_examples/test_env.py", line 716, in main
    env.reset()
  File "/home/hri-david/Omniverse/Isaaclab/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/direct_rl_env.py", line 258, in reset
    self._reset_idx(indices)
  File "/home/hri-david/Omniverse/Isaaclab/IsaacLab/Santi_tests/DirectEnv_examples/test_env.py", line 518, in _reset_idx
    super()._reset_idx(env_ids)
  File "/home/hri-david/Omniverse/Isaaclab/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/direct_rl_env.py", line 516, in _reset_idx
    self.scene.reset(env_ids)
  File "/home/hri-david/Omniverse/Isaaclab/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/scene/interactive_scene.py", line 330, in reset
    rigid_object.reset(env_ids)
  File "/home/hri-david/Omniverse/Isaaclab/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/rigid_object/rigid_object.py", line 100, in reset
    self._external_force_b[env_ids] = 0.0
AttributeError: 'RigidObject' object has no attribute '_external_force_b'
```

For more context, I set the object configs within the environment config class like this:

``` python
@configclass
class MyEnvCfg(DirectRLEnvCfg):
          #         
          #
          #

          cylinder_cfg = RigidObjectCfg(
                      prim_path="/World/envs/env_.*/Cylinder",
                      init_state=RigidObjectCfg.InitialStateCfg(pos=[0.5, 0, 0.055], rot=[1, 0, 0, 0]),
                      spawn=UsdFileCfg(
                          usd_path=f"{ISAAC_NUCLEUS_DIR}/Props/Blocks/DexCube/dex_cube_instanceable.usd",
                          scale=(0.8, 0.8, 0.8),
                          rigid_props=RigidBodyPropertiesCfg(
                              solver_position_iteration_count=16,
                              solver_velocity_iteration_count=1,
                              max_angular_velocity=1000.0,
                              max_linear_velocity=1000.0,
                              max_depenetration_velocity=5.0,
                              disable_gravity=False,
                          ),
                      ),
                  )
```

and its Spawned  inside the env class setup:

``` python

class MyEnv(DirectRLEnv):
    # pre-physics step calls
    # 

    cfg: MyEnvCfg

    def _setup_scene(self):
            self._robot = Articulation(self.cfg.robot)
            self._cylinder = RigidObject(self.cfg.cylinder_cfg)
            self.scene.rigid_objects["Cylinder"] = self._cylinder
            self.scene.articulations["Robot"] = self._robot
```
I don't know if I have to modify the `_reset_idx` method since is mentioned in the error when the `env.reset()` is called.
Here it is how I define it (similar to the `FrankaCabinetEnv` example):

```python
def _reset_idx(self, env_ids: torch.Tensor | None):
        super()._reset_idx(env_ids)
        # robot state
        print("default joint pos: ", self._robot.data.default_joint_pos)
        joint_pos = self._robot.data.default_joint_pos[env_ids] + sample_uniform(
            -0.125,
            0.125,
            (len(env_ids), self._robot.num_joints),
            self.device,
        )
        joint_pos = torch.clamp(joint_pos, self.robot_dof_lower_limits, self.robot_dof_upper_limits)
        joint_vel = torch.zeros_like(joint_pos)
        self._robot.set_joint_position_target(joint_pos, env_ids=env_ids)
        self._robot.write_joint_state_to_sim(joint_pos, joint_vel, env_ids=env_ids)
```
If there is any solution for this please tell me, thank you!

elle-miller, 2024-09-06:
I am getting the same issue with latest IsaacLab release. 

Before getting the `_external_force_b` error, it leads with this error for me. My object config is similar to above.

```
Traceback (most recent call last):
  File "/home/elle/code/external/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/asset_base.py", line 92, in <lambda>
    lambda event, obj=weakref.proxy(self): obj._initialize_callback(event),
  File "/home/elle/code/external/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/asset_base.py", line 263, in _initialize_callback
    self._initialize_impl()
  File "/home/elle/code/external/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/rigid_object/rigid_object.py", line 280, in _initialize_impl
    raise RuntimeError(
RuntimeError: Failed to find a rigid body when resolving '/World/envs/env_.*/Clutter'. Please ensure that the prim has 'USD RigidBodyAPI' applied.
```
I also modified the reset function to reset the object position but that doesn't make a difference. 
```
box_default_state = self.box.data.default_root_state.clone()[env_ids]
self.box.write_root_state_to_sim(box_default_state, env_ids)
```

jtigue-bdai, 2024-09-06:
@SantiDiazC can you please provide a reproduction script of this error

jtigue-bdai, 2024-09-06:
Just comparing your configuration of the RigidObject to the [ShadoHandEnvCfg](https://github.com/isaac-sim/IsaacLab/blob/main/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/direct/shadow_hand/shadow_hand_env_cfg.py) 

 I see that you are not setting mass properties in the spawner. This may be preventing the rigid object from being created completly.
 
 **mass_props=sim_utils.MassPropertiesCfg(density=567.0),**
```
 @configclass
class MyEnvCfg(DirectRLEnvCfg):
          #         
          #
          #

          cylinder_cfg = RigidObjectCfg(
                      prim_path="/World/envs/env_.*/Cylinder",
                      init_state=RigidObjectCfg.InitialStateCfg(pos=[0.5, 0, 0.055], rot=[1, 0, 0, 0]),
                      spawn=UsdFileCfg(
                          usd_path=f"{ISAAC_NUCLEUS_DIR}/Props/Blocks/DexCube/dex_cube_instanceable.usd",
                          scale=(0.8, 0.8, 0.8),
                          rigid_props=RigidBodyPropertiesCfg(
                              solver_position_iteration_count=16,
                              solver_velocity_iteration_count=1,
                              max_angular_velocity=1000.0,
                              max_linear_velocity=1000.0,
                              max_depenetration_velocity=5.0,
                              disable_gravity=False,
                          ),
                          mass_props=sim_utils.MassPropertiesCfg(density=567.0),
                      ),
                  )

```

elle-miller, 2024-09-09:
Hi @jtigue-bdai ,

For me, the error comes trying to add the table of the manager-based Franka Lift environment to my own DirectRL version.

https://github.com/isaac-sim/IsaacLab/blob/af088f593216f7215310cf5e3a13a259d7320a63/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/manager_based/manipulation/lift/lift_env_cfg.py#L49

My table config is 
```
table_cfg: RigidObjectCfg = RigidObjectCfg(
            prim_path="/World/envs/env_.*/Table",
            init_state=RigidObjectCfg.InitialStateCfg(pos=[0.5, 0, 0], rot=[0.707, 0, 0, 0.707]),
            spawn=UsdFileCfg(usd_path=f"{ISAAC_NUCLEUS_DIR}/Props/Mounts/SeattleLabTable/table_instanceable.usd",
            rigid_props=RigidBodyPropertiesCfg(
                    solver_position_iteration_count=16,
                    solver_velocity_iteration_count=1,
                    max_angular_velocity=1000.0,
                    max_linear_velocity=1000.0,
                    max_depenetration_velocity=5.0,
                    disable_gravity=False,
                ),
        # mass_props=sim_utils.MassPropertiesCfg(density=567.0)
        )
    )
```
and I instantiate is as:
```
def _setup_scene(self):
         .....
        self.table = RigidObject(self.cfg.table_cfg)
        self.scene.rigid_objects["table"] = self.table
```
I've just noticed an additional error with this setup which probably indicates the problem - can you help decipher this message?
```
2024-09-09 09:41:27 [11,814ms] [Warning] [omni.isaac.lab.sim.utils] Could not perform 'modify_rigid_body_properties' on any prims under: '/World/envs/env_0/Table'. This might be because of the following reasons:
    (1) The desired attribute does not exist on any of the prims.
    (2) The desired attribute exists on an instanced prim.
        Discovered list of instanced prim paths: ['/World/envs/env_0/Table/Visuals', '/World/envs/env_0/Table/Collisions']
```
When I add the `mass_props` you suggested, the above error prints out with the below appended
```
2024-09-09 09:43:13 [11,591ms] [Warning] [omni.isaac.lab.sim.utils] Could not perform 'modify_mass_properties' on any prims under: '/World/envs/env_0/Table'. This might be because of the following reasons:
    (1) The desired attribute does not exist on any of the prims.
    (2) The desired attribute exists on an instanced prim.
        Discovered list of instanced prim paths: ['/World/envs/env_0/Table/Visuals', '/World/envs/env_0/Table/Collisions']
```

jtigue-bdai, 2024-09-19:
@SantiDiazC and @elle-miller because you are using the DirectRLEnv workflow I cannot recreate your issue without your whole DirectRLEnv and Cfg. Please provide a your DirectRLEnv and DirectRLEnvCfg. So that we can recreate your issue.

Dhoeller19, 2024-10-04:
Hi @SantiDiazC and @elle-miller,
The reason you are seeing this problem is that your USD file you try to load doesn't contain any Physics Rigid body.
As an example for the table, you have to create a new USD that contains a rigid body:

1. Open Isaac Sim and open the table USD (http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.2/Isaac/Props/Mounts/SeattleLabTable/table_instanceable.usd)
2. Save a local copy (File -> Save as)
3. Right click on Collisions and do Add -> Physics -> Rigid Body
4. Save the file and load this one instead in your script.