isaac-sim/IsaacLab #949: [bug] Isaac lab streaming not working with client 103.1.1

ben-kaye, 2024-09-06:
### Describe the bug
omni.isaac.lab.app.AppLauncher does not work with livestream=1 using container

### Steps to reproduce
run any Isaac lab script eg.
./isaaclab.sh -p source/standalone/tutorials/00_sim/create_empty.py --livestream 1
attempt to connect via streaming client
fails unknown error 0x100

```
...TBD...

See fix
```

### System Info

Describe the characteristic of your environment:
<HOST MACHINE>
- Commit: tag:v1.1.0
- Isaac Sim Version: [e.g. 2022.2.0, this can be obtained by `cat ${ISAACSIM_PATH}/VERSION`]
- OS: Ubuntu 20.04 <isaac-lab-base>
- GPU: 2 RTX4090
- CUDA: 12.2
- GPU Driver: 535.183.01 

<CLIENT>
GPU: TITAN X 
OS: ubuntu 20.04
CUDA: 12.2
Driver: 535.183.01
Omniverse launcher: Streaming client 103.1.1

### FIX

omni.isaac.lab.app.AppLauncher 
Line 564: replacing the plugin versions with versions in the working livestream.py example (_isaac_sim/standalone_examples/api/omni.isaac.kit/livestream.py) allows connections.

```
# Default App: Streaming Client from the Omniverse Launcher
# enable_extension("omni.kit.streamsdk.plugins-4.5.1")
# enable_extension("omni.kit.livestream.core-4.3.6")
# enable_extension("omni.kit.livestream.native-4.1.0")

enable_extension("omni.kit.streamsdk.plugins-3.2.1")
enable_extension("omni.kit.livestream.core-3.2.0")
enable_extension("omni.kit.livestream.native")
```

### Checklist

- [X] I have checked that there is no similar issue in the repo (**required**)
- [X] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Isaac lab AppLauncher uses a working streaming configuration
- [ ]

closed 2024-09-10 by Dhoeller19

alberthli, 2024-09-13:
I get this error when trying to stream the headless native app in a docker container.
```
# start the container using the following command
# see: https://docs.omniverse.nvidia.com/isaacsim/latest/installation/install_container.html#isaac-sim-app-install-container
docker run --name isaac-sim --entrypoint bash -it --runtime=nvidia --gpus all -e "ACCEPT_EULA=Y" --rm --network=host \
    -e "PRIVACY_CONSENT=Y" \
    -v ~/docker/isaac-sim/cache/kit:/isaac-sim/kit/cache:rw \
    -v ~/docker/isaac-sim/cache/ov:/root/.cache/ov:rw \
    -v ~/docker/isaac-sim/cache/pip:/root/.cache/pip:rw \
    -v ~/docker/isaac-sim/cache/glcache:/root/.cache/nvidia/GLCache:rw \
    -v ~/docker/isaac-sim/cache/computecache:/root/.nv/ComputeCache:rw \
    -v ~/docker/isaac-sim/logs:/root/.nvidia-omniverse/logs:rw \
    -v ~/docker/isaac-sim/data:/root/.local/share/ov/data:rw \
    -v ~/docker/isaac-sim/documents:/root/Documents:rw \
    nvcr.io/nvidia/isaac-sim:4.1.0

# within the container, start the headless native app
./runheadless.native.sh
```
The app starts correctly with output:
```
Loading user config located at: '/root/.local/share/ov/data/Kit/Isaac-Sim/4.1/user.config.json'
[Info] [carb] Logging to file: /root/.nvidia-omniverse/logs/Kit/Isaac-Sim/4.1/kit_20240913_220521.log
2024-09-13 22:05:21 [3ms] [Warning] [carb.crashreporter-breakpad.plugin] [previous crash] preventing upload of minidump due to user opt-out: '/root/.local/share/ov/data/Kit/Isaac-Sim/4.1/6a8ac5f2-49e5-4c31-c93e0f88-8c741cc6.dmp.zip'
2024-09-13 22:05:21 [7ms] [Warning] [carb.crashreporter-breakpad.plugin] [previous crash] preventing upload of minidump due to user opt-out: '/root/.local/share/ov/data/Kit/Isaac-Sim/4.1/f8fd6a3b-c40d-4629-74075a8f-cf5f7d7e.dmp.zip'
2024-09-13 22:05:21 [9ms] [Warning] [carb.crashreporter-breakpad.plugin] [previous crash] preventing upload of minidump due to user opt-out: '/root/.local/share/ov/data/Kit/Isaac-Sim/4.1/be7be3d8-412f-4529-db1aaba4-f66bb6b5.dmp.zip'
2024-09-13 22:05:21 [10ms] [Warning] [carb.crashreporter-breakpad.plugin] [previous crash] preventing upload of minidump due to user opt-out: '/root/.local/share/ov/data/Kit/Isaac-Sim/4.1/2ce0020e-02fd-4192-0aee5d8c-eaf1c07f.dmp.zip'
2024-09-13 22:05:21 [25ms] [Warning] [carb.crashreporter-breakpad.plugin] [previous crash] preventing upload of minidump due to user opt-out: '/root/.local/share/ov/data/Kit/Isaac-Sim/4.1/8af0efb9-d788-4e4f-4a9b48a3-68d519c9.dmp.zip'
2024-09-13 22:05:21 [34ms] [Warning] [carb.crashreporter-breakpad.plugin] [previous crash] preventing upload of minidump due to user opt-out: '/root/.local/share/ov/data/Kit/Isaac-Sim/4.1/30d540e3-2a0c-4bd5-084f639f-e345cb8c.dmp.zip'
[0.534s] [ext: omni.kit.async_engine-0.0.0] startup
[1.342s] [ext: omni.stats-1.0.1] startup
[1.343s] [ext: omni.client-1.1.0] startup
[1.377s] [ext: omni.datastore-0.0.0] startup
[1.377s] [ext: omni.blobkey-1.1.0] startup
[1.378s] [ext: omni.ujitso.default-1.0.0] startup
[1.379s] [ext: omni.hsscclient-0.0.0] startup
[1.380s] [ext: omni.rtx.shadercache.vulkan-1.0.0] startup
[1.381s] [ext: omni.assets.plugins-0.0.0] startup
[1.383s] [ext: omni.gpu_foundation-0.0.0] startup
[1.401s] [ext: carb.windowing.plugins-1.0.0] startup
2024-09-13 22:05:23 [1,372ms] [Warning] [carb.windowing-glfw.plugin] GLFW initialization failed.
2024-09-13 22:05:23 [1,372ms] [Warning] [carb] Failed to startup plugin carb.windowing-glfw.plugin (interfaces: [carb::windowing::IGLContext v1.0],[carb::windowing::IWindowing v1.4]) (impl: carb.windowing-glfw.plugin)
[1.402s] [ext: omni.kit.renderer.init-0.0.0] startup
2024-09-13 22:05:23 [1,462ms] [Warning] [omni.platforminfo.plugin] failed to open the default display.  Can't verify X Server version.
2024-09-13 22:05:24 [2,767ms] [Warning] [carb.windowing-glfw.plugin] GLFW initialization failed.
2024-09-13 22:05:24 [2,767ms] [Warning] [carb] Failed to startup plugin carb.windowing-glfw.plugin (interfaces: [carb::windowing::IGLContext v1.0],[carb::windowing::IWindowing v1.4]) (impl: carb.windowing-glfw.plugin)
2024-09-13 22:05:24 [2,767ms] [Error] [carb.glinterop.plugin] GLInteropContext::init: carb::windowing is not available

|---------------------------------------------------------------------------------------------|
| Driver Version: 550.107.02    | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA H100 NVL                  | Yes: 0 |     | 95830   MB | 10de      | 0          |
|     |                                  |        |     |            | 2321      | 82969629.. |
|     |                                  |        |     |            | 1         |            |
|---------------------------------------------------------------------------------------------|
| 1   | NVIDIA H100 NVL                  | Yes: 1 |     | 95830   MB | 10de      | 0          |
|     |                                  |        |     |            | 2321      | 2a5af47c.. |
|     |                                  |        |     |            | 21        |            |
[... cut ...]
2024-09-13 22:06:58 [96,700ms] [Warning] [omni.kit.imgui_renderer.plugin] _createExtendCursor: No windowing.
2024-09-13 22:06:58 [96,700ms] [Warning] [omni.kit.imgui_renderer.plugin] _createExtendCursor: No windowing.
[96.909s] Isaac Sim Headless Native App is loaded.
```
When I try to stream to a client computer, a window very briefly pops up and disappears, then on the server, I see
```
main: thread_init: already added for thread
main: thread_init: already added for thread
```
On the client, I'm using Omniverse Streaming Client v103.1.1.

Are there any workarounds to this issue?

jtigue-bdai, 2024-09-16:
it looks like you are using the fix that came with this issue so I am supprised its not working. 

Can you try: setting`enable_extension("omni.kit.livestream.native-4.1.0")` to `# enable_extension("omni.kit.livestream.native")` in your `source/extensions/omni.isaac.lab/omni/isaac/lab/app/app_launcher.py` file?

Let me know if that works. If that doesn't work please open another ticket.

alberthli, 2024-09-17:
@jtigue-bdai, thanks for the quick response. Sorry if I'm missing something obvious (I'm new to Isaac Lab), but since I'm starting up the server using the script in the Isaac Docker container, I'm not sure I have or am using the `app_launcher.py` file. I `grep`d the phrase
```
omni.kit.livestream.native
```
in the `isaac-sim` directory of the container, and it seems like the version for this is just set in a bunch of `.kit` files. In the file `apps/omni.isaac.sim.headless.native.kit`, I changed
```
"omni.kit.livestream.native" = {version = "4.1.0", exact = true}
```
to 
```
"omni.kit.livestream.native" = {version = "*"}
```
and ran into the same issue. If I'm doing something wrong, please let me know. Otherwise, I'll open a new issue.

jtigue-bdai, 2024-10-18:
@alberthli and @josyulavt are you both using NVidia H100 GPUs? If so they cannot run Isaac Sim.
[requirements](https://docs.omniverse.nvidia.com/isaacsim/latest/installation/requirements.html)

qZhang88, 2025-02-26:
```
[INFO]: Starting the simulation. This may take a few seconds. Please wait...
main: thread_init: already added for thread
main: thread_init: already added for thread
2025-02-26 03:50:35 [52,989ms] [Warning] [isaaclab.sim.simulation_context] Simulation is stopped. The app will keep running with physics disabled. Press Ctrl+C or close the window to exit the app.
Warning: Possible version incompatibility. Attempting to load carb::cudainterop::CudaInterop with version v0.15 against v0.11.
^X^C[120.514s] Simulation App Shutting Down
```

cannot connect to stream client 103.1, any help?

commit 710f64d4 (2024-09-11): Sets native livestream extensions to Isaac Sim 4.1-4.0 defaults (#954)

# Description

Streaming does not work for `isaaclab --livestream 1` using streaming
client 103.1.1.

This change sets the streaming client extension versions to the defaults
in isaac-sim-4.1.0/apps/omni.isaac.sim.headless.native.kit

Fixes #949 

## Type of change

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with
`./isaaclab.sh --format`
- [ ] I have made corresponding changes to the documentation
- [x] My changes generate no new warnings
- [ ] I have added tests that prove my fix is  [... cut]