isaac-sim/IsaacLab #950: [Question] Timeline for full rendering for TiledCamera

elle-miller, 2024-09-06:
### Question

In issue #493, Kelly mentioned that you are working on full rendering with the tiled camera:

> Currently, the RGB tiled rendering API only provides ambient RGB, which does not capture any lighting and shadow information. We are working on a full tiled rendering API that will provide the same data as what we would achieve with the renderer.

Can we get an approximate estimate for when this feature may be available? 

I am doing research with pixel-based agents where lighting/shadows can be very important, so would really appreciate any info you can provide. Thank you! :)

jtigue-bdai, 2024-09-06:
@Dhoeller19 can we provide the estimated release of the full rendering TiledCamera

elle-miller, 2024-09-23:
[image]

Thank you! :100:

closed 2024-09-23 by elle-miller