isaac-sim/IsaacLab #959: error: Please ensure that the prim has 'USD ArticulationRootAPI' applied

kaiseong, 2024-09-07:
I need a robot with Robotiq 2F140 attached to UR5e, so I configured it in Isaac Sim as shown in the picture. However, when I edited UR10-Reach appropriately and ran the code, the learning that worked fine when I only used ur5e occurred an error saying **AttributeError: 'Articulation' object has no attribute '_root_physx_view'.** So, to check, I simply loaded the Robot using the test.py code below, which only launches the environment, and I got two errors: **RuntimeError: Failed to find an articulation when resolving '/World/envs/env_.*/Robot'. Please ensure that the prim has 'USD ArticulationRootAPI' applied.** and **AttributeError: 'Articulation' object has no attribute '_data'. Did you mean: 'data'?** Does anyone know how to solve this or how to attach a gripper to a manipulator through code?
Thank you
[image]

test.py
```import argparse
from omni.isaac.lab.app import AppLauncher

parser = argparse.ArgumentParser(description="custom test to Isaac-Reach-UR5-v0")
parser.add_argument("--num_envs", type=int, default=64, help="number of environments to spawn")

AppLauncher.add_app_launcher_args(parser)
args_cli = parser.parse_args()

app_launcher = AppLauncher(args_cli)
simulation_app = app_launcher.app

import torch
from omni.isaac.lab.envs import ManagerBasedRLEnv
from omni.isaac.lab_assets.ur5 import UR5_CFG
import omni.isaac.lab.sim as sim_utils
from omni.isaac.lab.assets import ArticulationCfg, AssetBaseCfg
from omni.isaac.lab.scene import InteractiveScene, InteractiveSceneCfg
from omni.isaac.lab.sim import SimulationContext
from omni.isaac.lab.utils import configclass
from omni.isaac.lab.utils.assets import ISAAC_NUCLEUS_DIR

@configclass
class URSceneCfg(InteractiveSceneCfg):
    """Configuration for the scene with a robotic arm."""

    # world
    ground = AssetBaseCfg(
        prim_path="/World/ground",
        spawn=sim_utils.GroundPlaneCfg(),
        init_state=AssetBaseCfg.InitialStateCfg(pos=(0.0, 0.0, -1.05)),
    )

    table = AssetBaseCfg(
        prim_path="{ENV_REGEX_NS}/Table",
        spawn=sim_utils.UsdFileCfg(
            usd_path=f"{ISAAC_NUCLEUS_DIR}/Props/Mounts/SeattleLabTable/table_instanceable.usd",
        ),
        init_state=AssetBaseCfg.InitialStateCfg(pos=(0.55, 0.0, 0.0), rot=(0.70711, 0.0, 0.0, 0.70711)),
    )

    # robots
    robot: ArticulationCfg = UR5_CFG.replace(prim_path="/World/envs/env_.*/Robot")

    # lights
    light = AssetBaseCfg(
        prim_path="/World/light",
        spawn=sim_utils.DomeLightCfg(color=(0.75, 0.75, 0.75), intensity=2500.0),
    )
def run_simulator(sim: sim_utils.SimulationContext, scene: InteractiveScene):
    """Runs the simulation loop."""
    # Extract scene entities
    # note: we only do this here for readability.
    robot = scene["robot"]
    # Define simulation stepping
    sim_dt = sim.get_physics_dt()
    count = 0
    # Simulation loop
    while simulation_app.is_running():
        # Reset
        if count % 500 == 0:
            # reset counter
            count = 0
            # reset the scene entities
            # root state
            # we offset the root state by the origin since the states are written in simulation world frame
            # if this is not done, then the robots will be spawned at the (0, 0, 0) of the simulation world
            root_state = robot.data.default_root_state.clone()
            root_state[:, :3] += scene.env_origins
            robot.write_root_state_to_sim(root_state)
            # set joint positions with some noise
            joint_pos, joint_vel = robot.data.default_joint_pos.clone(), robot.data.default_joint_vel.clone()
            joint_pos += torch.rand_like(joint_pos) * 0.1
            robot.write_joint_state_to_sim(joint_pos, joint_vel)
            # clear internal buffers
            scene.reset()
            print("[INFO]: Resetting robot state...")
        # Apply random action
        # -- generate random joint efforts
        efforts = torch.randn_like(robot.data.joint_pos) * 5.0
        # -- apply action to the robot
        robot.set_joint_effort_target(efforts)
        # -- write data to sim
        scene.write_data_to_sim()
        # Perform step
        sim.step()
        # Increment counter
        count += 1
        # Update buffers
        scene.update(sim_dt)

def main():
    """Main function."""
    # Load kit helper
    sim_cfg = sim_utils.SimulationCfg()
    sim = SimulationContext(sim_cfg)
    # Set main camera
    sim.set_camera_view([2.5, 0.0, 4.0], [0.0, 0.0, 2.0])
    # Design scene
    scene_cfg = URSceneCfg(num_envs=args_cli.num_envs, env_spacing=2.0)
    scene = InteractiveScene(scene_cfg)
    # Play the simulator
    sim.reset()
    # Now we are ready!
    print("[INFO]: Setup complete...")
    # Run the simulator
    run_simulator(sim, scene)

if __name__ == "__main__":
    # run the main function
    main()
    # close sim app
    simulation_app.close()

```

closed 2024-09-08 by kaiseong

TerenceXue-tech, 2024-11-07:
I encountered the same issue. Could you please share how you resolved it

LittlePotatoChip, 2024-12-25:
same questions，hope to know how to solve，thanks

kellyguo11, 2024-12-25:
Do you have an ArticulationRootAPI schema applied to the robot? You can right click on the root prim of the robot or the root fixed joint from the UI to add an ArticulationRootAPI.

itsmehere, 2025-02-02:
@kellyguo11 Is there any way to add the `ArticulationRootAPI` schema in the `robot = ArticulationCfg(...)` or in the `spawn=sim_utils.UrdfFileCfg(...)`? If not, would I need to modify the URDF itself so that `UrdfFileCfg` can appropriately set this up?

kellyguo11, 2025-02-03:
We don't have a way to apply the schema in Isaac Lab right now, mostly because we don't actually know which prim the ArticulationRootAPI should be applied to. One way to debug is to first do a separate workflow of importing the URDF to USD, and then inspecting the USD file to see which prim has the ArticulationRootAPI. It could be some things have changed with the latest URDF importer as it went through a pretty major makeover. If you could share your URDF and repro script, we can also look into it.

itsmehere, 2025-02-03:
@kellyguo11 Ah ok, I've added the urdf and my `ArticulationCfg` for the robot into [this gist](https://gist.github.com/itsmehere/e60c0d1ea72d1be380e9d9341a380693).

There should be two files: `dual_ur5e.urdf` and `universal_robots.py`. `universal_robots.py` is the same config file within `isaaclab_assets/robots` to which I've added my new ur5 based robot. For context, the exact `BIMANUAL_UR5_CFG` I have worked prior to the isaac sim update, except without the need to specify `joint_drive`.

itsmehere, 2025-02-03:
@kellyguo11 I've basically uploaded my entire repo [here](https://github.com/itsmehere/bimanual_ur5). The commit where I've added all my files is [here](https://github.com/itsmehere/bimanual_ur5/commit/38ed2b3fc65fd00907b2560145fa77ea8d28477c). I run the environment using this command: `./isaaclab.sh -p scripts/reinforcement_learning/rsl_rl/train.py --task=Isaac-Lift-Cube-Bimanual-UR5-v0 --headless`. For context, the current `BIMANUAL_UR5_CFG` was working prior to the isaac sim update, except without the need to specify `joint_drive`.

Junb0Dong, 2025-03-03:
Hi @kellyguo11 Thank you so much for your solution! But I could not find the `add ArticulationRootAPI` option in UI. Is there any other way to add ArticulationRootAPI？Thanks!

System Info
Commit:
IsaacSim Version: 4.5
IsaacLab Version: 2.0
OS: Ubuntu 20.04.1

kellyguo11, 2025-03-03:
You can do it through the UI by right clicking on the root prim of the articulation in the Stage tree and adding a physics schema for ArticulationRootAPI.

Junb0Dong, 2025-03-04:
@kellyguo11  Thank you so much for your detailed reply! I added `Articulation Root` as the below image, yet it still didn't work. The error message remains exactly the same as before. Did I do that step wrong, I didn't find  `ArticulationRootAPI` in UI, so I added the `Articulation Root`.

[image]

pumfish, 2025-05-18:
> [@kellyguo11](https://github.com/kellyguo11) I've basically uploaded my entire repo [here](https://github.com/itsmehere/bimanual_ur5). The commit where I've added all my files is [...]

Hello? Have you solved the problem of ArticulationRootAPI, I've encountered the same question either.

pumfish, 2025-05-18:
> Do you have an ArticulationRootAPI schema applied to the robot? You can right click on the root prim of the robot or the root fixed joint from the UI to add an ArticulationRootAPI.

Hello @kellyguo11 , I'm working in an IsaacLab environment without a GUI. Is there a way to set up the ArticulationRootAPI schema just through code?