isaac-sim/IsaacLab #979: [Question] Implementing a trained robot in a reinforcement learning environment.

Hokite, 2024-09-12:
I'm working on reinforcement learning tasks using Isaac Lab.

Is it possible to bring a pre-trained robot into the training environment and carry out the specific learning tasks I want?

For example, when training a mobile manipulator, the manipulator is the robot I want to train, while the mobile robot is an already pre-trained robot.

Could you let me know if this is possible?

Mayankm96, 2024-09-12:
The example here shows how to use pre-trained policies for defining action terms: https://github.com/isaac-sim/IsaacLab/tree/main/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/manager_based/navigation

That should probably help you get started :)

Hokite, 2024-09-19:
@Mayankm96 Thank you so much. The method you suggested is the one I found. However, there was a problem, and regarding the policy.pt used in the method you suggested, if I change it to policy.pt of the robot I learned and proceed with the learning, I get an error that there is no constants.pkl, can you tell me if the pt file you used is a learned weight with rsl_rl or something else?