isaac-sim/IsaacLab #985: [Question] How to create aggregates?

locoxsoco, 2024-09-13:
### Question

In IsaacGymEnvs it was possible to aggregate actors, useful when having multiple actors or a complex single actor in a scene (faster computation, saving GPU memory and minimizing collision errors). Is it possible to do this in Isaac Lab?

[Here's](https://github.com/isaac-sim/IsaacGymEnvs/blob/aeed298638a1f7b5421b38f5f3cc2d1079b6d9c3/isaacgymenvs/tasks/shadow_hand.py#L356) an example of how to apply this in IssacGymEnvs.

And [here's](https://docs.nvidia.com/gameworks/content/gameworkslibrary/physx/guide/Manual/RigidBodyCollision.html#aggregates) a deeper documentation about Physx aggregates.

Mayankm96, 2024-09-16:
Checking with the PhysX team on this, but I think this is now handled automatically by the PhysX parser of the USD stage.

locoxsoco, 2024-09-16:
I appreciate your support! The reason behind this question is that I'm currently facing an issue while attempting to train a humanoid model with hands in IsaacLab. This model was successfully trained in IsaacGymEnvs, but when I try to run the training in IsaacLab, some agents crash after a few steps, causing the training process to stop entirely (this issue occurs even without applying any actions to the actuators). This crash only occurs once I set `enabled_self_collisions=True` in `ArticulationRootPropertiesCfg`

I encountered a similar problem in IsaacGymEnvs, and I managed to resolve it by aggregating the actor using PhysX aggregates. If the info seems not related to this issue, I'll create another one explaining this behaviour in more detail.

I'm commited to help the sim to handle these complex models, so if there is anything I can add to this or other ticket I'll be very happy to contribute.

Mayankm96, 2024-09-17:
We are benchmarking for robots with a large number of links and joints, so doesn't seem like an issue with that specifically. Internally, the stage parser does make aggregates for articulations so that part should be similar to what you did in Isaac Gym.

From the description above, it seems the issue is more with the colliders in the scene than the asset itself. Also, do you have mimic joints and other more complex mechanisms (like tendons)?

RandomOakForest, 2025-02-11:
Will close this for now, but feel free to start a new _discussion_ may we be of further help. Thanks.

closed 2025-02-11 by RandomOakForest