isaac-sim/IsaacLab #987: [Question] OmniGraphSettings::getCudaDeviceOrdinal: unable to get a valid CUDA device id from the renderer. Defaulting to GPU0.

LiangZhisama, 2024-09-14:
### Question

Hello everyone,
I'm running Isaaclab on an ubuntu 20.04 remote server and installing via pip. 
Previously I was able to run and record videos and save them normally through command like `./isaaclab.sh -p source/standalone/workflows/rsl_rl/train.py --task Isaac-Reach-Franka-v0 --headless --video`
but after I removed and re-updated the Nvidia drivers, the following problem occurred when recording video:
```
[INFO] Using python from: /home/guardx/miniconda3/envs/isaaclab/bin/python                                                                                                        
[INFO][AppLauncher]: Loading experience file: /home/guardx/feihan_dev/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/../../../../../../apps/isaaclab.python.headless.rendering.kit
[Warning] [omni.isaac.kit.simulation_app] Modules: ['omni.kit_app'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Starting kit application with the following args:  ['/home/guardx/miniconda3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/home/guardx/feihan_dev/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/../../../../../../apps/isaaclab.python.headless.rendering.kit', '--/app/tokens/exe-path=/home/guardx/miniconda3/envs/isaaclab/lib/python3.10/site-packages/omni', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True', '--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--/app/fastShutdown=True', '--ext-folder', '/home/guardx/miniconda3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/exts', '--ext-folder', '/home/guardx/miniconda3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/apps', '--/renderer/activeGpu=0', '--/physics/cudaDevice=0', '--portable', '--no-window', '--/app/window/hideUi=True']
Passing the following args to the base kit application:  ['--task', 'Isaac-Reach-Franka-v0', '--headless', '--video']
Loading user config located at: '/home/guardx/miniconda3/envs/isaaclab/lib/python3.10/site-packages/omni/data/Kit/Isaac-Sim/4.0/user.config.json'
[Info] [carb] Logging to file: /home/guardx/miniconda3/envs/isaaclab/lib/python3.10/site-packages/omni/logs/Kit/Isaac-Sim/4.0/kit_20240914_040852.log
2024-09-14 08:08:52 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[0.061s] [ext: omni.kit.async_engine-0.0.0] startup
[0.156s] [ext: omni.stats-1.0.1] startup
[0.157s] [ext: omni.client-1.1.0] startup
[0.189s] [ext: omni.datastore-0.0.0] startup
[0.190s] [ext: omni.blobkey-1.1.0] startup
[0.190s] [ext: omni.ujitso.default-1.0.0] startup
[0.191s] [ext: omni.hsscclient-0.0.0] startup
[0.191s] [ext: omni.rtx.shadercache.vulkan-1.0.0] startup
[0.192s] [ext: omni.assets.plugins-0.0.0] startup
[0.194s] [ext: omni.gpu_foundation-0.0.0] startup
[0.205s] [ext: carb.windowing.plugins-1.0.0] startup
2024-09-14 08:08:52 [178ms] [Warning] [carb.windowing-glfw.plugin] GLFW initialization failed.
2024-09-14 08:08:52 [178ms] [Warning] [carb] Failed to startup plugin carb.windowing-glfw.plugin (interfaces: [carb::windowing::IGLContext v1.0],[carb::windowing::IWindowing v1.4]) (impl: carb.windowing-glfw.plugin)
[0.206s] [ext: omni.kit.renderer.init-0.0.0] startup
2024-09-14 08:08:52 [232ms] [Warning] [omni.platforminfo.plugin] failed to open the default display.  Can't verify X Server version.
2024-09-14 08:08:52 [285ms] [Error] [carb.graphics-vulkan.plugin] VkResult: ERROR_INITIALIZATION_FAILED
2024-09-14 08:08:52 [285ms] [Error] [carb.graphics-vulkan.plugin] vkEnumeratePhysicalDevices failed. No physical device is found.
2024-09-14 08:08:52 [285ms] [Error] [carb.graphics-vulkan.plugin] No physical device is found.
2024-09-14 08:08:52 [286ms] [Error] [gpu.foundation.plugin] carb::graphics::createInstance failed.
2024-09-14 08:08:54 [2,070ms] [Error] [carb.graphics-vulkan.plugin] VkResult: ERROR_INITIALIZATION_FAILED
2024-09-14 08:08:54 [2,070ms] [Error] [carb.graphics-vulkan.plugin] vkEnumeratePhysicalDevices failed. No physical device is found.
2024-09-14 08:08:54 [2,070ms] [Error] [carb.graphics-vulkan.plugin] No physical device is found.
2024-09-14 08:08:54 [2,071ms] [Error] [gpu.foundation.plugin] carb::graphics::createInstance failed.
2024-09-14 08:08:55 [3,372ms] [Error] [omni.gpu_foundation_factory.plugin] Failed to create any GPU devices, including an attempt with compatibility mode.

(zenity:2450233): Gtk-WARNING **: 04:08:55.669: cannot open display: 
[3.447s] [ext: omni.kit.pipapi-0.0.0] startup
[3.449s] [ext: omni.kit.pip_archive-0.0.0] startup
[3.450s] [ext: omni.kit.telemetry-0.5.0] startup
[3.467s] [ext: omni.mtlx-0.1.0] startup
[3.468s] [ext: omni.usd.config-1.0.4] startup
[3.473s] [ext: omni.gpucompute.plugins-0.0.0] startup
[3.474s] [ext: omni.usd.libs-1.0.1] startup
[3.601s] [ext: omni.kit.loop-isaac-1.2.0] startup
[3.602s] [ext: omni.kit.test-0.0.0] startup
[3.644s] [ext: omni.appwindow-1.1.8] startup
2024-09-14 08:08:55 [3,617ms] [Warning] [carb.windowing-glfw.plugin] GLFW initialization failed.
2024-09-14 08:08:55 [3,617ms] [Warning] [carb] Failed to startup plugin carb.windowing-glfw.plugin (interfaces: [carb::windowing::IGLContext v1.0],[carb::windowing::IWindowing v1.4]) (impl: carb.windowing-glfw.plugin)
[3.646s] [ext: omni.kit.renderer.core-1.0.1] startup
2024-09-14 08:08:55 [3,619ms] [Error] [omni.kit.renderer.plugin] GPU Foundation is not initialized!
2024-09-14 08:08:55 [3,622ms] [Warning] [carb.windowing-glfw.plugin] GLFW initialization failed.
2024-09-14 08:08:55 [3,622ms] [Warning] [carb] Failed to startup plugin carb.windowing-glfw.plugin (interfaces: [carb::windowing::IGLContext v1.0],[carb::windowing::IWindowing v1.4]) (impl: carb.windowing-glfw.plugin)
[3.651s] [ext: omni.kit.renderer.capture-0.0.0] startup
[3.653s] [ext: omni.kit.renderer.imgui-1.0.1] startup
2024-09-14 08:08:55 [3,627ms] [Warning] [carb.windowing-glfw.plugin] GLFW initialization failed.
2024-09-14 08:08:55 [3,627ms] [Warning] [carb] Failed to startup plugin carb.windowing-glfw.plugin (interfaces: [carb::windowing::IGLContext v1.0],[carb::windowing::IWindowing v1.4]) (impl: carb.windowing-glfw.plugin)
2024-09-14 08:08:55 [3,628ms] [Error] [omni.kit.imgui_renderer.plugin] _prepareDefaultWindow: invalid device attached to the default window!
[3.658s] [ext: omni.ui-2.23.11] startup
[3.668s] [ext: omni.kit.mainwindow-1.0.3] startup
[3.669s] [ext: omni.usd.schema.geospatial-0.0.0] startup
[3.764s] [ext: omni.usd.schema.forcefield-106.0.20] startup
[3.798s] [ext: omni.usd.schema.anim-0.0.0] startup
[3.819s] [ext: omni.usd.schema.semantics-0.0.0] startup
[3.823s] [ext: omni.usd.schema.audio-0.0.0] startup
[... 174 lines cut ...]
2024-09-14 08:09:00 [8,304ms] [Warning] [omni.graph.core.plugin] OmniGraphSettings::getCudaDeviceOrdinal: unable to get a valid CUDA device id from the renderer. Defaulting to GPU0.
2024-09-14 08:09:00 [8,312ms] [Warning] [omni.graph.core.plugin] OmniGraphSettings::getCudaDeviceOrdinal: unable to get a valid CUDA device id from the renderer. Defaulting to GPU0.
2024-09-14 08:09:00 [8,321ms] [Warning] [omni.graph.core.plugin] OmniGraphSettings::getCudaDeviceOrdinal: unable to get a valid CUDA device id from the renderer. Defaulting to GPU0.
2024-09-14 08:09:00 [8,329ms] [Warning] [omni.graph.core.plugin] OmniGraphSettings::getCudaDeviceOrdinal: unable to get a valid CUDA device id from the renderer. Defaulting to GPU0.
2024-09-14 08:09:00 [8,338ms] [Warning] [omni.graph.core.plugin] OmniGraphSettings::getCudaDeviceOrdinal: unable to get a valid CUDA device id from the renderer. Defaulting to GPU0.
2024-09-14 08:09:00 [8,346ms] [Warning] [omni.graph.core.plugin] OmniGraphSettings::getCudaDeviceOrdinal: unable to get a valid CUDA device id from the renderer. Defaulting to GPU0.
2024-09-14 08:09:00 [8,354ms] [Warning] [omni.graph.core.plugin] OmniGraphSettings::getCudaDeviceOrdinal: unable to get a valid CUDA device id from the renderer. Defaulting to GPU0.
```
Then the program falls into an infinite loop showing this.
Here is the nvidia drivers information for typing `dpkg -l | grep nvidia`
```
[... cut ...]
ii  nvidia-utils-550-server                    550.90.07-0lambda0.22.04.4                        amd64        NVIDIA driver support binaries
ii  screen-resolution-extra                    0.18.2                                            all          Extension for the nvidia-settings control panel
ii  xserver-xorg-video-nvidia-550-server       550.90.07-0lambda0.22.04.4                        amd64        NVIDIA binary Xorg driver
```
I am so frustrated with this issue and would appreciate any help you can give me! Also let me know if you guys need any additional information.

Toni-SM, 2024-09-14:
Hi @LiangZhisama 

It seems to be GPU driver is not working.
Use `nvidia-smi` to show GPU info

LiangZhisama, 2024-09-14:
> Hi @LiangZhisama
> 
> It seems to be GPU driver is not working. Use `nvidia-smi` to show GPU info

Thanks so much for your quick reply! Actually I can run the code and get normal output without turning on the video recording function(which may actually correspond to env.render()). And here is my GPU info:
[image]
And here is the info by running `nvcc-V`
```
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2024 NVIDIA Corporation
Built on Thu_Mar_28_02:18:24_PDT_2024
Cuda compilation tools, release 12.4, V12.4.131
Build cuda_12.4.r12.4/compiler.34097967_0
```

LiangZhisama, 2024-09-14:
> Hi @LiangZhisama
> 
> It seems to be GPU driver is not working. Use `nvidia-smi` to show GPU info

Thanks for your answer! I notice that the updated NVIDIA driver version on remote server is incorrect, which is `Xserver-xorg-video-nvidia-550-server`. After I re-install the version `nvidia-driver-550-server`, problem solved.

closed 2024-10-02 by kellyguo11